openbmc/openbmc #526: bootloader not updated when using all.tar in REST update interface

geissonator, 2016-08-25:
I had a barreleye that was at v1.0.4 stable.  I had built an image based off of master that I patched in the GPIO fix for.  After updating to it, I noticed that all of the boot console output was garbled until we got to userspace apps and then everything was readable (at the faster master baud rate) in the serial console.

This implies to me that the boot loader was not updated using this flash image (it's still at the stable branch with the slower baud rate which is why the serial output is garbled when we go through the boot code).

I'm just using this script to do the update - https://github.com/causten/tools/blob/master/obmc/codeupdate.py

It copies all.tar onto the BMC and then uses the REST api's to trigger the update.  I verified the erase and write process all went as expected in the serial console

@mdmillerii  thoughts?

amboar, 2016-11-25:
@geissonator I was just looking at this with respect to @mine260309's patch https://gerrit.openbmc-project.xyz/#/c/544/

As of right now, if the code is the documentation then this is expected behaviour. If you'd like to update the bootloader then you *must* supply the `<platform>.tar` file (not `<platform>.all.tar`) and you *must not* supply the `update_kernel_and_apps` option.

mine260309, 2016-11-25:
@amboar By flashing `<platform>.all.tar` the bootloader should be updated as well.
If you try to update `<platform>.tar` and `update_kernel_and_apps` option is 0, you get error: `ERROR: File not found in update archive: image-bmc`

amboar, 2016-11-25:
@mine260309 agh! I got the tarball names backwards.

@geissonator ignore the above :)

williamspatrick, 2016-11-27:
Garbled console doesn't prove anything about what was updated. There is also a uboot environment variable with the console speed. If this isn't cleared, it will override any default that might be in your image.

johnhcwang, 2016-12-06:
@williamspatrick , may I know how to clear the uboot environment variable? Should I call another method to clear it before update?
I tried to update via REST using image (.all.tar) with the slower speed (9600), but the baudrate in uboot environment still keep as 38400 after the update.
If I update the image by socflash, the baud rate 9600 is working.
I'm working on stable branch by the way.

johnhcwang, 2016-12-06:
Seems that I have to change the preserve_network_settings as 0 to clear uboot environment variable. I'll try it.

geissonator, 2017-09-18:
We seem to be all set here now.

closed 2017-09-18 by geissonator