openbmc/openbmc #527: Second boot does not work with master

geissonator, 2016-08-25:
I've seen this on both Palmetto and Barreleye in recent days with the GPIO pin testing.  Basically, the 2nd boot (after an initial successful boot) never works when using master.

On Palmetto, I would just not get anything to the console.  The boot watchdog would keep hitting and we'd just keep retrying to power on.  

On Barreleye, I got some data to the console.  It did have a special PNOR image from Elizabeth but the fact it booted fine the first time leads me to believe it's the unrelated to the PNOR image.

```
1.69629|ERRL|Dumping errors reported prior to registration
  1.73291|================================================
  1.73291|Error reported by fsi (0x0A00)
  1.75122|  FsiDD::checkForErrors> Error discovered in MAEB
  1.75122|  ModuleId   0x0a FSI::MOD_FSIDD_CHECKFORERRORS
  1.75123|  ReasonCode 0x0a11 FSI::RC_ERROR_IN_MAEB
  1.75123|  UserData1  Physical Node of FSI Master processor : 0x0000000000000103
  1.75124|  UserData2  Physical Position of FSI Master processor : 0x0000000080000000
  1.75124|User Data Section 0, type UD
  1.75124|  Subsection type 0x06
  1.75125|  ComponentId errl (0x0100)
  1.75125|  CALLOUT
  1.75125|  PROCEDURE ERROR
  1.75125|  Procedure: 45
  1.75125|User Data Section 1, type UD
  1.75126|  Subsection type 0x06
  1.75126|  ComponentId errl (0x0100)
  1.75126|  CALLOUT
  1.75126|  HW CALLOUT
  1.75127|  Reporting CPU ID: 1061
  1.75127|  Called out entity:
  1.75127|User Data Section 2, type UD
  1.75127|  Subsection type 0x04
  1.75128|  ComponentId errl (0x0100)
  1.75128|User Data Section 3, type UD
  1.75128|  Subsection type 0x02
  1.75128|  ComponentId errl (0x0100)
  1.75129|  TARGET
  1.75129|  HUID: 00050000
  1.75130|  Phys path: Physical:/Sys0/Node0/Proc0
  1.75130|  Affinity path: Logical:/Sys0/Node0/Proc0
  1.75130|User Data Section 4, type UD
  1.75131|  Subsection type 0x05
  1.75131|  ComponentId errl (0x0100)
  1.75131|User Data Section 5, type UD
  1.75131|  Subsection type 0x02
  1.75132|  ComponentId errl (0x0100)
  1.75132|  TARGET
  1.75132|  HUID: 00040000
  1.75133|  Phys path: Physical:/Sys0/Node0/Membuf0
  1.75134|  Affinity path: Logical:/Sys0/Node0/Proc0/MCS0/Membuf0
  1.75134|User Data Section 6, type UD
  1.75135|  Subsection type 0x05
  1.75135|  ComponentId errl (0x0100)
  1.75135|User Data Section 7, type UD
  1.75135|  Subsection type 0x03
  1.75136|  ComponentId errl (0x0100)
  1.75136|User Data Section 8, type UD
  1.75136|  Subsection type 0x01
  1.75136|  ComponentId errl (0x0100)
  1.75137|  STRING
  1.75137|  Hostboot Build ID: hostboot-1f6784d-opdirty-2751695/hbicore.bin/eliner
  1.75137|User Data Section 9, type UD
  1.75138|  Subsection type 0x04
  1.75138|  ComponentId errl (0x0100)
  1.75138|User Data Section 10, type UD
  1.75139|  Subsection type 0x15
  1.75139|  ComponentId hb-trace (0x3100)
  1.75139|User Data Section 11, type UD
  1.75139|  Subsection type 0x15
  1.75140|  ComponentId hb-trace (0x3100)
  1.75140|================================================
  2.65105|================================================
[... 196 lines cut ...]
  3.67259|  Subsection type 0x05
  3.67259|  ComponentId errl (0x0100)
  3.67259|User Data Section 5, type UD
  3.67260|  Subsection type 0x04
  3.67260|  ComponentId errl (0x0100)
  3.67260|User Data Section 6, type UD
  3.67260|  Subsection type 0x03
  3.67260|  ComponentId errl (0x0100)
  3.67261|User Data Section 7, type UD
  3.67261|  Subsection type 0x01
  3.67261|  ComponentId errl (0x0100)
  3.67261|  STRING
  3.67262|  Hostboot Build ID: hostboot-1f6784d-opdirty-2751695/hbicore.bin/eliner
  3.67262|User Data Section 8, type UD
  3.67262|  Subsection type 0x04
  3.67262|  ComponentId errl (0x0100)
  3.67263|User Data Section 9, type UD
  3.67263|  Subsection type 0x15
  3.67263|  ComponentId hb-trace (0x3100)
  3.67263|User Data Section 10, type UD
  3.67264|  Subsection type 0x15
  3.67264|  ComponentId hb-trace (0x3100)
  3.67264|================================================
  3.72551|ISTEP  6. 3
```

geissonator, 2016-08-25:
Also, ever if you power cycle the system, your chances of getting an IPL to work after the initial flash of the image are only about 20% or so.

textaddict, 2016-09-01:
After doing some experimentation on a palmetto system netbooting into v1.0-stable build and then booting from flash into a recent master build I found some significant differences between the GPIO direction registers at 0x1e780004 and 0x1e780024.  

After I boot the system from this version in flash:
cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.0-174"
VERSION_ID="v1.0-174-ga81b2a7-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.0-174"
BUILD_ID="v1.0"

The GPIO direction registers look like this:
0x1e780000 : 0x130F8CE3 **_0x01706074_** 0x00000000 0x00000000
0x1e780010 : 0x00000000 0x00000000 0x00000000 0x00000000
0x1e780020 : 0x9F48F7FF **_0x00004002_** 0x00000000 0x00000000

And the system won't IPL it just loops AC power cycles every 30 seconds or so.  Then I net booted this 1.0-stable build:
cat /etc/os-release 
ID=openbmc-phosphor
NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)
VERSION=0.1.0 (jethro)
VERSION_ID=v1.0.3-4-g63288c0-dirty
PRETTY_NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 (jethro)
BUILD_ID=v1.0.3

Which left the GPIOS looking like this:
0x1e780000 : 0x130F8CE3 **_0x0370E677_** 0x00000000 0x00000000
0x1e780010 : 0x00000000 0x00000000 0x00000000 0x00000000
0x1e780020 : 0xDF48F7FF **_0xC738F202_** 0x00000000 0x00000000

In addition to the difference at offset 0x4 and 0x24 for the gpio direction there is a difference at offset 0x20 on a few bits.  
And the system IPLs fine with that code.  It doesn't look like master is touching those bits because after I reboot into the master version from flash the system still IPLs fine.  

As a final check I AC power cycled, confirmed the system wouldn't IPL then used these devmem commands to make master IPL the system, it worked.

root@palmetto:~# devmem 0x1e780000 32 0x130f8ce3
root@palmetto:~# devmem 0x1e780004 32 0x0370e677
root@palmetto:~# devmem 0x1e780020 32 0xdf48f7ff
root@palmetto:~# devmem 0x1e780024 32 0xc738f202

I will review the schematics tomorrow to see what is implicated by these differences.

geissonator, 2017-09-18:
We seem to be good now with this issue.

closed 2017-09-18 by geissonator