openbmc/openbmc #528: Specifying VUART(s) to userspace.

williamspatrick, 2016-08-25:
When we moved from 4.4 to 4.7, the VUART changed from ttyS5 to ttyS0.  We believe this to be an artifact of the driver probing.  I fixed the obmc-console application to be hardcoded to ttyS0 instead of ttyS5 but this is not the best long-term solution.

Need to come up with the appropriate way to identify this to userspace.  dts entries and udev symlinks were mentioned in various venues as possible solutions.  Once we come up with a mechanism we need to update userspace to utilize.

legoater, 2016-08-25:
we could use : 

` +		serial5 = &vuart;`

to fix the vuart to ttyS5.

On a related topic, the  `obmc-console` systemd service needs an update to take into account the recent changes of the daemon which now requires a `<tty>` parameter. This was introduced by commit [Supply Serial port as an optional argument to obmc-console-server](https://github.com/openbmc/obmc-console/commit/6221ce9483f9431061223f635a4dbd82e44dc75b)

legoater, 2016-08-26:
a udev rule is also quite simple to put in place : 

```
root@palmetto:~# cat /etc/udev/rules.d/vuart.rules
SUBSYSTEM=="tty", ATTR{iomem_base}=="0x1E787000", SYMLINK+="tty-host"

root@palmetto:~# ls -la /dev/tty-host
lrwxrwxrwx    1 root     root             5 Aug 25 10:25 /dev/tty-host -> ttyS5
```

and then change `/lib/systemd/system/obmc-console.service` to use `tty-host`

shenki, 2016-08-29:
I think we should introduce both fixes. This way we remain compatible with old userspaces that assume the vuart is at ttyS5, as well as doing the right thing for if the device tree/kernel changes in the future.

shenki, 2016-08-30:
Device tree patch posted to the list:

 https://lists.ozlabs.org/pipermail/openbmc/2016-August/004466.html

Need an owner for the udev change.

williamspatrick, 2016-08-30:
Userspace is already fixed.  Do we need this alias?  I think we need an explanation of what to look for in order to make the udev rule.

shenki, 2016-08-30:
No, given we've re-broken userspace we should not merge the dt change. Cedric has posted the udev rule we need above my comment.

williamspatrick, 2016-08-30:
Oh, I missed that, sorry.

His udev rule requires us to know the iomem address of the vuart?  That seems BMC specific, so I guess we should place that into the AST2400/AST2500 specific recipes?

shenki, 2016-08-30:
I've not seen that kind of matching done in udev before. I assume it's the best way to match.

Yeah, it is Aspeed specific so that makes sense to me.

legoater, 2016-08-30:
@williamspatrick : however the changes to ```lib/systemd/system/obmc-console.service ```  to use ```tty-host``` or whatever we choose for a device name is not aspeed specific.

williamspatrick, 2016-08-30:
Any objections to this?

```
SUBSYSTEM=="tty", DRIVERS=="aspeed-vuart",ATTRS{lpc_address}=="0x3f8", SYMLINK+="ttyVUART0"
```

legoater, 2016-08-30:
looks good to me.

closed 2016-09-21 by williamspatrick