openbmc/openbmc #529: boot_flags documentation missing on barreleye at customer site

basuv, 2016-08-25:
Customer was unable to set the boot order due to missing documentation. 

[image]

anoo1, 2016-08-26:
Expanding on the issue reported, the 'docs/host-management.md' is out of date, the 'BIOS boot options' describes the original supported values which were the hex values described in the ipmi spec, but since then the ipmi code changed to accept strings from dbus:

    {"Network"},
    {"Disk"},
    {"Safe"},
    {"CDROM"},
    {"Setup"},
    {"Default"},

basuv, 2016-09-09:
I

basuv, 2016-09-09:
I would like the customer to be able to see the list of boot flags to set from, on the openbmc documentation, and not to go to IPMI spec

rfrandse, 2018-06-08:
<https://gerrit.openbmc-project.xyz/11039> Add links to boot interfaces
Resolves  #529 boot_flags documentation missing on barreleye at customer site

closed 2018-06-08 by geissonator