openbmc/openbmc #530: Host Soft-reboot hang with BMC Release 1.0.4

agangidi53, 2016-08-25:
Hi I am hitting a 1-1 Host hang on Barreleye running BMC release 1.0.4
CPLD version1.0.0
BIOS firmware version 1.0.1
BMC Release Foxconn release based on based on 1.0.4 (v10.40.256) https://github.com/ken1029/openbmc/releases) 
OS Tested: Ubuntu 15.04 Live Image

Console Log is below:

Journal Logs don't show much info but are below:

Console logs:
root@ubuntu:~# reboot
[ 1080.799108] INFO: task loop0:954 blocked for more than 120 seconds.
[ 1080.799316]       Tainted: G           OE  3.19.0-51-generic #58-Ubuntu
[ 1080.799516] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[ 1080.799854] INFO: task casper-stop:2289 blocked for more than 120 seconds.
[ 1080.800075]       Tainted: G           OE  3.19.0-51-generic #58-Ubuntu
[ 1080.800277] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[ 1200.797615] INFO: task loop0:954 blocked for more than 120 seconds.
[ 1200.797827]       Tainted: G           OE  3.19.0-51-generic #58-Ubuntu
[ 1200.798039] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[ 1200.798378] INFO: task casper-stop:2289 blocked for more than 120 seconds.
[ 1200.798587]       Tainted: G           OE  3.19.0-51-generic #58-Ubuntu

Journal Logs

Aug 25 20:54:59 barreleye systemd[1]: systemd-hwdb-update.service: Cannot add dependency job, ignoring: Unit systemd-hwdb-update.service is masked.
Aug 25 20:54:59 barreleye systemd[1]: Started SSH Per-Connection Server (10.127.91.183:35180).
Aug 25 20:54:59 barreleye dropbear[1238]: Failed loading /etc/dropbear/dropbear_dss_host_key
Aug 25 20:54:59 barreleye dropbear[1238]: Failed loading /etc/dropbear/dropbear_ecdsa_host_key
Aug 25 20:54:59 barreleye dropbear[1238]: Child connection from 10.127.91.183:35180
Aug 25 20:55:09 barreleye dropbear[1238]: Bad password attempt for 'root' from 10.127.91.183:35180
Aug 25 20:55:09 barreleye dropbear[1238]: Bad password attempt for 'root' from 10.127.91.183:35180
Aug 25 20:55:12 barreleye dropbear[1238]: Password auth succeeded for 'root' from 10.127.91.183:35180
Aug 25 20:55:12 barreleye dropbear[1239]: lastlog_perform_login: Couldn't stat /var/log/lastlog: No such file or directory
Aug 25 20:55:12 barreleye dropbear[1239]: lastlog_openseek: /var/log/lastlog is not a file or directory!
Aug 25 20:57:09 barreleye system_manager.py[987]: Cannot read attributes: /sys/class/hwmon/hwmon1/fan4_input
Aug 25 20:59:59 barreleye systemd[1]: Time has been changed
Aug 25 20:59:58 barreleye systemd[1]: Time has been changed

mdmillerii, 2016-08-25:
@agangidi53 I see the task that is blocked is `loop0:954`.  Was this system in the "prepare for update" mode?

Unfortunately debugging stuck tasks usually involves finding the cause of the blockage, which is often not the stuck task.

Is the journal log from the next boot or were you able to log in while it was failed?

agangidi53, 2016-08-25:
Hey @mdmillerii  System is not in "prepare for update" mode.  I want to clarify again that this is a HOST soft-reboot hang that only happens on new BMC release. Journal log was for as it failed.

mdmillerii, 2016-08-25:
Ok if the host is logging hung tasks then it needs to be debuged from that side first to find what resource the tasks are waiting on.  First step is collecting stack traces (sysrq-t) and finding the blocked tasks.  Then analyze to find the lock or waitq ...

agangidi53, 2016-08-25:
@mdmillerii  Sure. Should I file a hostboot issue ? I field it here since I didn't see it on some openBMC releases vs newer ones, as I originally indicated.

mdmillerii, 2016-08-25:
its not any more likely to be hostboot than BMC  ... how recreatable is this?

The message says the task has not been scheduled for 2 minutes , but was put to sleep in an UNINTERPRETABLE state and should have been woken up.  

The loop0 task handles /dev/loop0 and on a live image I would expect that to be the mount of the main file system image on the boot or live media device.

I have seen hung tasks before when we told firmware to reboot but there was nothing to start the process ... but I would expect a message from opal about communication failing.  Can you watch if the ipmi reboot is being sent down to the bmc ?   @causten can you help how to trace that?

agangidi53, 2016-08-30:
Hey @mdmillerii   @causten   This is recreatable 1-1 ,  I am not sure on how to see if ipmi reboot is being sent down to bmc .  Journal logs ?

Don't see much on journal logs except for remote IO errors:
 Aug 30 22:20:34 barreleye system_manager.py[943]: i2c_io:134 ERROR Failed to do raw io: Remote I/O error
Aug 30 22:20:35 barreleye system_manager.py[943]: i2c_io:134 ERROR Failed to do raw io: Remote I/O error
Aug 30 22:20:37 barreleye system_manager.py[943]: i2c_io:134 ERROR Failed to do raw io: Remote I/O error
Aug 30 22:20:38 barreleye system_manager.py[943]: i2c_io:134 ERROR Failed to do raw io: Remote I/O error
Aug 30 22:20:40 barreleye system_manager.py[943]: i2c_io:134 ERROR Failed to do raw io: Remote I/O error

johnhcwang, 2016-08-31:
Hi @agangidi53 , the messages was printed from [here](https://github.com/ken1029/skeleton/blob/master/info/info.c#L132).
This process was created by FXN and try to get the HDD presence status from expander.
When the system power is disappeared and this process is trying to get the status from expander, we will get this message due to the i2c access error.

agangidi53, 2016-09-27:
@johnhcwang  @causten  

I believe I am now pretty sure of how to reproduce this. Reboot the BMC while host is up and any consecutive soft-reboots will hang

geissonator, 2017-02-23:
@agangidi53 still an issue with v1.0.5?  Does this need some priority or just an annoyance?

gtmills, 2018-05-23:
@agangidi53 @geissonator Still an issue or can we close? This is year and half old.

geissonator, 2018-05-23:
Please give our latest tag, v2.1 a try (with latest released op-build as well).  There have been a lot of fixes in this area between BMC and Host.

closed 2018-05-23 by geissonator

gtmills, 2018-05-23:
Our latest tag is actually v2.2, but either should work. https://github.com/openbmc/openbmc/releases/tag/v2.2