openbmc/openbmc #533: Firestone fails to boot with BMC Release 1.0.5

madscientist159, 2016-08-28:
Release 1.0.5 is not able to start the Firestone server.  Chassis power turns on but the main fans never ramp up, and no hostboot messages are printed on serial console.  BMC journal contains entries similar to following, indicating the OCC I2C devices may not be detected:

```
Aug 28 18:07:19 firestone system_manager.py[937]: New object: /org/openbmc/sensors (:1.12)
Aug 28 18:07:19 firestone system_manager.py[937]: All required objects started for BASE_APPS
Aug 28 18:07:19 firestone system_manager.py[937]: SystemManager Goto System State: BMC_STARTING
Aug 28 18:07:19 firestone system_manager.py[937]: Running System State: BMC_STARTING
Aug 28 18:07:19 firestone system_manager.py[937]: Starting process: /usr/sbin/power_control.exe 3000 10: power_control
Aug 28 18:07:19 firestone system_manager.py[937]: Deleting OCC device
Aug 28 18:07:19 firestone system_manager.py[937]: Starting process: /usr/sbin/control_host.exe: host_control
Aug 28 18:07:19 firestone system_manager.py[937]: sh: /sys/class/i2c-adapter/i2c-3/3-0050/online: No such file or directory
Aug 28 18:07:19 firestone system_manager.py[937]: Deleting OCC device
Aug 28 18:07:19 firestone system_manager.py[937]: Starting process: /usr/sbin/hwmon.py Firestone: hwmon
Aug 28 18:07:19 firestone system_manager.py[937]: sh: /sys/class/i2c-adapter/i2c-3/3-0050/online: No such file or directory
Aug 28 18:07:20 firestone system_manager.py[937]: Starting sensor manager
Aug 28 18:07:20 firestone system_manager.py[937]: Starting process: /usr/sbin/led_controller.exe: led_control
Aug 28 18:07:20 firestone system_manager.py[937]: Starting process: /usr/sbin/download_manager.py Firestone: download_manager
Aug 28 18:07:20 firestone system_manager.py[937]: Starting process: /usr/sbin/bmc_update.py: bmc_flash_control
Aug 28 18:07:20 firestone system_manager.py[937]: No LEDs present in the system
Aug 28 18:07:20 firestone system_manager.py[937]: Starting process: /usr/sbin/button_reset.exe: reset_button
Aug 28 18:07:21 firestone system_manager.py[937]: Starting process: /usr/sbin/button_power.exe: power_button
Aug 28 18:07:21 firestone system_manager.py[937]: Starting process: /usr/sbin/chassis_control.py: chassis_control
Aug 28 18:07:21 firestone system_manager.py[937]: Starting process: /usr/sbin/host_watchdog.exe: host_watchdog
Aug 28 18:07:21 firestone system_manager.py[937]: Starting process: /usr/sbin/control_bmc.exe: bmc_control
Aug 28 18:07:22 firestone system_manager.py[937]: Starting process: /usr/sbin/flash_bios.exe: flash_control
Aug 28 18:07:22 firestone systemd[1]: systemd-hwdb-update.service: Cannot add dependency job, ignoring: Unit systemd-hwdb-update.service is masked.
Aug 28 18:07:22 firestone systemd[1]: Started SSH Per-Connection Server (192.168.2.7:51272).
Aug 28 18:07:22 firestone dropbear[987]: Failed loading /etc/dropbear/dropbear_dss_host_key
Aug 28 18:07:22 firestone dropbear[987]: Failed loading /etc/dropbear/dropbear_ecdsa_host_key
Aug 28 18:07:22 firestone dropbear[987]: Child connection from 192.168.2.7:51272
Aug 28 18:07:23 firestone system_manager.py[937]: Starting process: /usr/sbin/inventory_items.py Firestone: inventory
Aug 28 18:07:23 firestone system_manager.py[937]: New object: /org/openbmc/sensors/host/BootProgress (:1.12)
Aug 28 18:07:23 firestone system_manager.py[937]: New object: /org/openbmc/sensors/host/cpu0/OccStatus (:1.12)
Aug 28 18:07:23 firestone system_manager.py[937]: New object: /org/openbmc/sensors/host/cpu1/OccStatus (:1.12)
Aug 28 18:07:23 firestone system_manager.py[937]: New object: /org/openbmc/sensors/host/powercap (:1.12)
Aug 28 18:07:23 firestone system_manager.py[937]: New object: /org/openbmc/sensors/host/OperatingSystemStatus (:1.12)
Aug 28 18:07:23 firestone system_manager.py[937]: New object: /org/openbmc/sensors/host/BootCount (:1.12)
Aug 28 18:07:23 firestone system_manager.py[937]: New object: /org/openbmc/buttons/reset0 (:1.13)
Aug 28 18:07:24 firestone system_manager.py[937]: New object: /org/openbmc/buttons/power0 (:1.14)
Aug 28 18:07:24 firestone system_manager.py[937]: New object: /org/openbmc/watchdog/host0 (:1.15)
Aug 28 18:07:24 firestone system_manager.py[937]: New object: /org/openbmc/control/power0 (:1.16)
Aug 28 18:07:24 firestone system_manager.py[937]: New object: /org/openbmc/control/host0 (:1.17)
Aug 28 18:07:24 firestone system_manager.py[937]: GPIO Lookup:  FSI_DATA = 325,out
Aug 28 18:07:24 firestone system_manager.py[937]: GPIO Lookup:  RESET_BUTTON = 356,both
Aug 28 18:07:24 firestone system_manager.py[937]: GPIO Lookup:  BMC_POWER_UP = 345,out
Aug 28 18:07:24 firestone system_manager.py[937]: New object: /org/openbmc/control/flash/bios (:1.19)
Aug 28 18:07:25 firestone system_manager.py[937]: GPIO Lookup:  POWER_BUTTON = 352,both
Aug 28 18:07:25 firestone system_manager.py[937]: GPIO Lookup:  FSI_CLK = 324,out
Aug 28 18:07:25 firestone system_manager.py[937]: GPIO Lookup:  SYS_PWROK_BUFF = 346,in
Aug 28 18:07:25 firestone system_manager.py[937]: GPIO Lookup:  FSI_ENABLE = 344,out
Aug 28 18:07:25 firestone system_manager.py[937]: GPIO Lookup:  BMC_WD_CLEAR_PULSE_N = 428,out
Aug 28 18:07:26 firestone system_manager.py[937]: GPIO Lookup:  CRONUS_SEL = 326,out
Aug 28 18:07:26 firestone system_manager.py[937]: GPIO Lookup:  CM1_OE_R_N = 454,out
Aug 28 18:07:26 firestone system_manager.py[937]: GPIO Lookup:  BMC_THROTTLE = 395,out
Aug 28 18:07:26 firestone system_manager.py[937]: GPIO Lookup:  BMC_CP0_RESET_N = 434,out
Aug 28 18:07:27 firestone system_manager.py[937]: GPIO Lookup:  BMC_CFAM_RESET_N_R = 394,out
Aug 28 18:07:27 firestone system_manager.py[937]: GPIO Lookup:  IDBTN = 455,out
Aug 28 18:07:27 firestone system_manager.py[937]: GPIO Lookup:  PEX8718_DEVICES_RESET_N = 334,out
Aug 28 18:07:28 firestone system_manager.py[937]: GPIO Lookup:  CP0_DEVICES_RESET_N = 427,out
Aug 28 18:07:28 firestone system_manager.py[937]: GPIO Lookup:  CP1_DEVICES_RESET_N = 429,out
Aug 28 18:07:36 firestone dropbear[987]: Password auth succeeded for 'root' from 192.168.2.7:51272
Aug 28 18:07:36 firestone dropbear[1032]: lastlog_perform_login: Couldn't stat /var/log/lastlog: No such file or directory
Aug 28 18:07:36 firestone dropbear[1032]: lastlog_openseek: /var/log/lastlog is not a file or directory!
Aug 28 18:07:45 firestone system_manager.py[937]: Running Download Manager
Aug 28 18:07:46 firestone system_manager.py[937]: New object: /org/openbmc/control/chassis0 (:1.21)
Aug 28 18:07:46 firestone system_manager.py[937]: All required objects started for BMC_STARTING
Aug 28 18:07:46 firestone system_manager.py[937]: SystemManager Goto System State: BMC_READY
Aug 28 18:07:46 firestone system_manager.py[937]: Running System State: BMC_READY
Aug 28 18:07:46 firestone system_manager.py[937]: Starting process: /usr/sbin/discover_system_state.py: restore
Aug 28 18:07:46 firestone system_manager.py[937]: Running ChassisControlService
[... 173 lines cut ...]
Aug 28 18:10:26 firestone system_manager.py[937]: Running System State: HOST_POWERED_ON
Aug 28 18:10:26 firestone system_manager.py[937]: Booting host
Aug 28 18:10:26 firestone system_manager.py[937]: Using primary side of the bios flash
Aug 28 18:10:26 firestone system_manager.py[937]: Running System State: HOST_BOOTING
Aug 28 18:10:26 firestone system_manager.py[937]: Setting watchdog poll interval to: 30000
Aug 28 18:10:26 firestone system_manager.py[937]: Starting watchdog with poll interval: 30000
Aug 28 18:10:56 firestone system_manager.py[937]: Polling watchdog: 0
Aug 28 18:10:56 firestone system_manager.py[937]: Watchdog Error, Hard Rebooting
Aug 28 18:10:56 firestone system_manager.py[937]: Turn off power
Aug 28 18:10:56 firestone system_manager.py[937]: Stopping watchdog
Aug 28 18:10:56 firestone system_manager.py[937]: Running System State: HOST_POWERING_OFF
Aug 28 18:10:59 firestone system_manager.py[937]: Deleting OCC device
Aug 28 18:10:59 firestone system_manager.py[937]: Running System State: HOST_POWERED_OFF
Aug 28 18:10:59 firestone system_manager.py[937]: Turn on power and boot
Aug 28 18:10:59 firestone system_manager.py[937]: sh: /sys/class/i2c-adapter/i2c-3/3-0050/online: No such file or directory
Aug 28 18:10:59 firestone system_manager.py[937]: Deleting OCC device
Aug 28 18:10:59 firestone system_manager.py[937]: Running System State: HOST_POWERING_ON
Aug 28 18:10:59 firestone system_manager.py[937]: sh: /sys/class/i2c-adapter/i2c-3/3-0051/online: No such file or directory
Aug 28 18:11:02 firestone system_manager.py[937]: Running System State: HOST_POWERED_ON
Aug 28 18:11:02 firestone system_manager.py[937]: Booting host
Aug 28 18:11:02 firestone system_manager.py[937]: Using primary side of the bios flash
Aug 28 18:11:02 firestone system_manager.py[937]: Running System State: HOST_BOOTING
Aug 28 18:11:02 firestone system_manager.py[937]: Setting watchdog poll interval to: 30000
Aug 28 18:11:02 firestone system_manager.py[937]: Starting watchdog with poll interval: 30000
```

williamspatrick, 2016-08-29:
@anoo1 / @geissonator  - Have we tested stable on a Firestone lately?  I know we have one Firestone that was recently given to our team that we are trying to get working.

anoo1, 2016-08-29:
With stable we were not able to power it on either, like this issue describes. I'm not sure if when support was added for Firestone, if the system powered on then.

madscientist159, 2016-08-29:
I have a report here that the Palmetto codebase was able to power the machine on at least to an OS, and this might have worked (at least in terms of powering the machine on) at time of publication:
https://github.com/sannerd/openbmc/tree/firestone

madscientist159, 2016-08-31:
@williamspatrick Is anyone currently working on this or is it "up for grabs" when we have time to analyze the failure in more detail?

williamspatrick, 2016-08-31:
@madscientist159 - We have a few issues going on with regard to various machines.  As part of some of the kernel work that was being done the way GPIOs are configured changed.  Some of the critical GPIOs needed to boot various machines aren't in the correct state and we're trying to unravel it all.  See #513 for more details.

It appears that Firestone is included in the list that is currently messed up.  Our main focus right now is on restoring all functionality on Barreleye and Palmetto on master and then Firestone.  (The whole team isn't working on this one problem.)

Based on your comments above from @sannerd's repository, Firestone must look a lot like Palmetto from this perspective, so hopefully it will be pretty easy going after we figure out the Palmetto.  Keep tabs on that other issue and I'll have @anoo1 / @geissonator keep you updated on our Firestone progress.

If you are on IRC (freenode #openbmc), you can chat with them as well to see where things are at and if you want to help Firestone along more.

madscientist159, 2016-08-31:
@williamspatrick Thanks for the pointers / update.  I just didn't want to duplicate effort, that happened with Flashrom earlier this month so I've adopted an "ask first" policy before diving in to active projects.

madscientist159, 2016-09-08:
Finally got around to testing the sannerd repository.  It was unable to IPL the Firestone, and worse it left one of the CPUs running with no fans.  Currently comparing GPIOs from a successfully IPLed Firestone to the stable OpenBMC branch since OpenBMC master won't even boot the BMC kernel.

amboar, 2016-09-09:
@madscientist159 ping myself and @shenki if you find issues with the GPIOs.

madscientist159, 2016-09-09:
@amboar Will do.  Unfortunately my prior test of sannerd's code was invalid, the wrong branch was checked out.  Will post new results as I have them.  Note to self: don't try to download OpenBMC while on the phone! :warning:

madscientist159, 2016-09-09:
Confirming that the firestone branch of sannerd's repository does not IPL the Firestone.  Twiddling the GPIOs to match running status of the proprietary BMC does not work either.

madscientist159, 2016-09-12:
We're working on a fix for this and the various other issues with Firestone and OpenBMC master.  The serial BMC console works again but a number of other devices are miswired / misconfigured and it is taking time to sort this out.

shenki, 2016-09-12:
The are a few items to sort out here on master, due to no one owning the Firestone support:

 * device tree updates for 4.7 kernel branch
 * GPIOs and pinmux configuration
 * direct-attched PHY testing (aka non-NSCI networking)

madscientist159, 2016-09-12:
@shenki Yep, we pretty much hit every one of those!  Grinding through the updates for 4.7 now.

madscientist159, 2016-09-20:
Firestone is now successfully booting and has been stable for a couple of days uptime using my tree here:
https://github.com/madscientist159/linux-openbmc/commits/dev-4.7-firestone-fixes

That tree also adds a fan control driver.

madscientist159, 2016-10-17:
Be advised that hostboot still fails due to hostboot commit https://github.com/open-power/hostboot/commit/1fbe5e7bf5042ee9e7353a169d13eec54c270f04 as noted in https://github.com/open-power/hostboot/issues/53 .  This issue is still being worked on at IBM.

adamliyi, 2016-11-02:
Hi @madscientist159 Timothy, an obmc user is trying to boot Firestone Host CPU. I suggest to use your patch: https://github.com/madscientist159/linux-openbmc/commits/dev-4.7-firestone-fixes with a IBM GA Hostboot image. Are you pushing your patch to upstream openbmc/linux?

madscientist159, 2016-11-02:
@adamliyi The patches should all be merged at this point.  They were submitted as soon as Firestone was stabilized enough to find the hostboot problem.

adamliyi, 2016-11-08:
@madscientist159 , @mine260309 mentioned that if we want to IPL Firestone, we need:
```
Clone (or add remote) to https://github.com/madscientist159/openbmc-openbmc
Checkout branch firestone-test-last-working, verify its HEAD is 175c297c

When host is powered off, use below command to redirect the host output to VUART:
devmem 0x1e789174 32 0x00008000

```
I am wondering - if all your patch has been merged to openbmc, then the master head of openbmc should be able to IPL firestone? (plus the `redirect host output to VUART` step)?

madscientist159, 2016-11-08:
@adamliyi The redirect step is a hack -- Firestone has a dedicated physical serial port that is active under the proprietary BMC firmware.  I would greatly prefer that this dedicated serial port remain active under OpenBMC as well, as it does function with older hostboot versions and OpenBMC.

I think there is still a pinmux patch that is being worked on and is not yet merged.  I am not able to allocate further time to test / fix OpenBMC master until the hostboot problem is resolved.

amboar, 2016-11-09:
> I think there is still a pinmux patch that is being worked on by other OpenBMC developers and is not yet merged. I am not able to allocate further time to test / fix OpenBMC master until the hostboot problem is resolved.

@madscientist159 As a note, all required pinmux functionality is in openbmc/linux, the patches that I Cc'ed you on are for upstream.

madscientist159, 2016-11-09:
@amboar Thank you for the clarification!  I had tried to compile master a week or two ago but it wouldn't IPL Firestone and I shelved that project waiting for the hostboot fixes in case the same root cause is in play.  I'll pay more attention to the Email headers next time! :wink:

gtmills, 2018-05-23:
@madscientist159 Can this be closed? or is this still valid? Firestone is no longer a supported machine.

closed 2018-05-24 by shenki