openbmc/openbmc #534: OBMC failed to build

adamliyi, 2016-08-31:
Using latest openbmc code from github:

Latest commit:

```
commit a81b2a743ce76bc08df6114dce681301dd4c7fd7
Author: Patrick Williams <patrick@stwcx.xyz>
Date:   Tue Aug 30 11:50:02 2016 -0500

    host-ipmid-fru.bb: bump version
    
    Change-Id: I4487343724e460010411c633e59e4b23efb3f46a
    Signed-off-by: Patrick Williams <patrick@stwcx.xyz>
```
Build error message:
```
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: make -j 4
| arm-openbmc-linux-gnueabi-gcc  -march=armv5e -marm --sysroot=/home/adam/workdir/openbmc_palmetto/build/tmp/sysroots/palmetto -c -pthread -I/home/adam/workdir/openbmc_palmetto/build/tmp/sysroots/palmetto/usr/include/gio-unix-2.0/ -I/home/adam/workdir/openbmc_palmetto/build/tmp/sysroots/palmetto/usr/include/glib-2.0 -I/home/adam/workdir/openbmc_palmetto/build/tmp/sysroots/palmetto/usr/lib/glib-2.0/include -fPIC -Werror  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/adam/workdir/openbmc_palmetto/build/tmp/work/armv5e-openbmc-linux-gnueabi/obmc-op-control-power/1.0-r1=/usr/src/debug/obmc-op-control-power/1.0-r1 -fdebug-prefix-map=/home/adam/workdir/openbmc_palmetto/build/tmp/sysroots/x86_64-linux= -fdebug-prefix-map=/home/adam/workdir/openbmc_palmetto/build/tmp/sysroots/palmetto=  -o power_control.o power_control_obj.c
| make -C pgood_wait
| Package libsystemd was not found in the pkg-config search path.
| Perhaps you should add the directory containing `libsystemd.pc'
| to the PKG_CONFIG_PATH environment variable
| No package 'libsystemd' found
| make[1]: Entering directory `/home/adam/workdir/openbmc_palmetto/build/tmp/work/armv5e-openbmc-linux-gnueabi/obmc-op-control-power/1.0-r1/git/op-pwrctl/pgood_wait'
| arm-openbmc-linux-gnueabi-gcc  -march=armv5e -marm --sysroot=/home/adam/workdir/openbmc_palmetto/build/tmp/sysroots/palmetto -c  -fPIC -Werror  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/adam/workdir/openbmc_palmetto/build/tmp/work/armv5e-openbmc-linux-gnueabi/obmc-op-control-power/1.0-r1=/usr/src/debug/obmc-op-control-power/1.0-r1 -fdebug-prefix-map=/home/adam/workdir/openbmc_palmetto/build/tmp/sysroots/x86_64-linux= -fdebug-prefix-map=/home/adam/workdir/openbmc_palmetto/build/tmp/sysroots/palmetto=  -o pgood_wait.o pgood_wait.c
| pgood_wait.c:19:28: fatal error: systemd/sd-bus.h: No such file or directory
| compilation terminated.
| make[1]: *** [pgood_wait.o] Error 1
| make[1]: Leaving directory `/home/adam/workdir/openbmc_palmetto/build/tmp/work/armv5e-openbmc-linux-gnueabi/obmc-op-control-power/1.0-r1/git/op-pwrctl/pgood_wait'
| make: *** [pgood_wait] Error 2
| make: *** Waiting for unfinished jobs....
| ERROR: oe_runmake failed
| ERROR: Function failed: do_compile (log file is located at /home/adam/workdir/openbmc_palmetto/build/tmp/work/armv5e-openbmc-linux-gnueabi/obmc-op-control-power/1.0-r1/temp/log.do_compile.20302)
ERROR: Task 2027 (/home/adam/workdir/openbmc_palmetto/meta-phosphor/common/recipes-phosphor/obmc-phosphor-chassis/obmc-op-control-power.bb, do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 1328 tasks of which 1021 didn't need to be rerun and 1 failed.
Waiting for 0 running tasks to finish:
```

rednewguy, 2016-09-01:
If you retry build, works the third time.

bradbishop, 2016-09-01:
https://gerrit.openbmc-project.xyz/#/c/358

closed 2016-09-01 by williamspatrick