openbmc/openbmc #538: Unable to flash BIOS on Witherspoon:  LIBFLASH: Flash in writing state ! Waiting...

jazurin, 2016-09-01:
We verified that the PNOR chip was properly installed with correct orientation according to the BMC board schematics.  We removed the PNOR chip and using a Dediprog we were able to successfully erase, flash and verify a firestone.pnor image.

The issue can be reproduced on Witherspoon system w1 which is running open-bmc VERSION_ID="v1.0-177-g3e3ee2d-dirty".   We tried using pflash, REST API and flasher.exe utility, and the result is the same.  

```
root@witherspoon:~# ps |grep flasher
 1031 root      2972 S    grep flasher
root@witherspoon:~# stat /tmp/firestone.pnor
  File: /tmp/firestone.pnor
  Size: 67108864        Blocks: 131072     IO Block: 4096   regular file
Device: 14h/20d Inode: 8779        Links: 1
Access: (0644/-rw-r--r--)  Uid: (    0/    root)   Gid: (    0/    root)
Access: 2016-08-30 19:58:28.000000000
Modify: 2016-08-30 19:58:53.000000000
Change: 2016-08-30 19:58:53.000000000

root@witherspoon:~# flasher.exe bios /tmp/firestone.pnor /org/openbmc/control/flash/bios
Starting flasher: bios,/tmp/firestone.pnor,/org/openbmc/control/flash/bios,
flasher: /tmp/firestone.pnor, BMC = 0, address = 0x0
Setting up BIOS flash
LIBFLASH: Flash in writing state ! Waiting...
<<<<<<<< No more output after this >>>>>>>>>>>
```
```
root@witherspoon:~# pflash -E -f -p /tmp/firestone.pnor
LIBFLASH: Flash in writing state ! Waiting...
<<<<<<<< No more output after this >>>>>>>>>>>
```

williamspatrick, 2016-09-02:
@shenki - Can you take a first look at this?

shenki, 2016-09-02:
The error message is due to the status register in the flash chip having the Write In Progress (WIP) bit set. I have seen this before, but I don't recall how to fix it. Does running `pflash -i` and then trying again fix it?

I don't know anything about `flasher.exe`. I see that you then tried to use `pflash`. Did you power cycle the BMC in between trying flasher and pflash?

I believe that pflash, when run the way you are using it, will access the flash MMIO space through `/dev/mem`. We have a kernel driver for the flash, so now you have a userspace program and a kernel driver owning the same bit of hardware.

I suggest updating the flash through the /dev/mtd interface instead.

In the case where you're writing the entire PNOR, you can use flashcp. I think it goes like this

```
# flashcp firestone.pnor /dev/mtd7
```

williamspatrick, 2016-09-02:
The documented way to do a PNOR flash update is here: 
    https://github.com/openbmc/docs/blob/master/code-update.md#host-code-update
That method uses a REST API to call this code:
    https://github.com/openbmc/skeleton/blob/f834525a05d4f420ba90a28f4309149369f547bf/op-flasher/flasher_obj.c#L294

I believe 'flasher.exe' is a wrapper around a dbus method that invokes this code.  You can see some of the same print statements as "Setting up BIOS flash" in the above link.

This is the method that has worked for Barreleye and I believe still does.  Witherspoon appears to not work.  We shouldn't be recommending different tools when this is the documented path right now.  The 'flasher' path utilizes libflash from skiboot, which is the same as what pflash itself uses.

It appears that this code removes a write-lock:
    https://github.com/openbmc/skeleton/blob/f834525a05d4f420ba90a28f4309149369f547bf/op-flasher/flasher_obj.c#L260
Is this related to the "write in progress" bit?

Backtrace of the "LIBFLASH" printf is this:
    https://github.com/open-power/skiboot/blob/master/libflash/libflash.c#L599
    https://github.com/open-power/skiboot/blob/master/libflash/libflash.c#L629
    https://github.com/open-power/skiboot/blob/master/libflash/libflash.c#L826
    https://github.com/openbmc/skeleton/blob/f834525a05d4f420ba90a28f4309149369f547bf/op-flasher/flasher_obj.c#L252

Based on this backtrace we haven't even gotten to the write-protect unlock so that must not be related.

shenki, 2016-09-02:
Ok, will let you resolve this

williamspatrick, 2016-09-02:
@shenki -

You mentioned in a private message that you thought maybe the Linux driver was interfering based on device tree entries causing the driver to be loaded and cause contention.  The interesting thing is that Palmetto *has* the host_pnor entry and Witherspoon does not:
    https://github.com/openbmc/linux/blob/dev-4.7/arch/arm/boot/dts/aspeed-bmc-opp-palmetto.dts#L53
    https://github.com/openbmc/linux/blob/dev-4.7/arch/arm/boot/dts/aspeed-bmc-opp-witherspoon.dts

I believe we should get the host_pnor added to Witherspoon to resolve this.

williamspatrick, 2016-09-02:
We have done some more investigation and there are three [potential] issues:
    1. Witherspoon dts does not specify the host-pnor device.
    2. libflash (from open-power/skiboot does not support the AST2500 chip.
    3. pflash installed in openbmc uses upstream libflash but flasher from openbmc/skeleton does not.

I am going to convert this into an 'epic' and create sub-issues to address each of these items.

mine260309, 2016-10-20:
I'm not quite sure, but it looks like commit 0df9bcf9 removes the patch `0001-Add-flash-and-layout-to-Witherspoon-devtree.patch`, but the changes are not committed to linux-obmc, and the `host-pnor` device is missing.

shenki, 2016-10-21:
Lets close this bug. The pflash that was being used in the original report did not have support for the ast2500, which I missed at the time.

The work to enable the ast2500 PNOR controller is tracked in https://github.com/openbmc/linux/issues/100

williamspatrick, 2016-10-21:
This bug was converted into an epic to track all the pieces we need to update and control PNOR for Witherspoon power on.  We need to keep it open until we are certain PNOR "works".

Outstanding work items I see:

- [] kernel support for SPI controller.
- [] corresponding device tree updates
- [] update to pflash to utilize mtd interface.
- [] openbmc/skeleton : op-flasher/flasher_obj.c can utilize mtd

shenki, 2016-11-15:
All the items on your list are now complete.

There is an outstanding issue #785 that needs to be resolved before PNOR is ready for Witherspoon.

closed 2016-11-15 by williamspatrick