openbmc/openbmc #539: dbus-ipmi command fails with Insufficient privilege level while sending OEM partial add command

rahulmah, 2016-09-02:
dbus-ipmi command fails with Insufficient privilege level while sending OEM partial add command with fake content.

Sent below command OEM partial add command with correct Reservation ID but got Insufficient privilege level. This command should have been accepted.

16:27:23.179	INFO	Executing command '/tmp/ipmitool -I dbus raw 0x32 0xf0 0x07 0x00 0 0 0 0 0 1 2 3 4 5 6 7 8 9 0xa 0xb 0xc 0xd 0xe 0xf'.	
16:27:23.470	INFO	Command exited with return code 1.	
16:27:23.470	INFO	${stdout} = 	
16:27:23.471	INFO	${stderr} = Unable to send RAW command (channel=0x0 netfn=0x32 lun=0x0 cmd=0xf0 rsp=0xd4): Insufficient privilege level	
16:27:23.471	INFO	${output_3} = 1

Same "Insufficient privilege level" is seen while sending command with Test Wrong Reservation_ID also.

[openbmc_ffdc_report_Wrong_Reservation_ID.txt](https://github.com/openbmc/openbmc/files/451219/openbmc_ffdc_report_Wrong_Reservation_ID.txt)
[openbmc_ffdc_report_correct Reservation_ID.txt](https://github.com/openbmc/openbmc/files/451220/openbmc_ffdc_report_correct.Reservation_ID.txt)

System details: 
------------------------------------------------------------------------ 
DRIVER INFO : Build Info	Executed : cat /etc/version 
------------------------------------------------------------------------ 
20160831020951 
------------------------------------------------------------------------ 

------------------------------------------------------------------------ 
DRIVER INFO : FW Level	Executed : cat /etc/os-release 
------------------------------------------------------------------------ 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.0-174"
VERSION_ID="v1.0-174-ga81b2a7-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.0-174"
BUILD_ID="v1.0" 
------------------------------------------------------------------------

rahulmah, 2016-09-02:
@causten : Do you have any background info on this? It is suppose to be working.

vishwabmc, 2016-09-06:
@rahulmah Looks like you are trying to execute a command which is blacklisted in customer mode. Check what your modes are and then try to run the ones that are whitelisted.

Also, if you want to run all commands, change the mode to lab.

gkeishin, 2016-09-08:
@vishwabmc  and @rahulmah  

If its a customer black listed mode, then we should key off from the error message and expect the privilege level message unless we want to add method to enable lab mode.  

I dont think we would want to do changing the mode to lab.. 

probably we should key off from the error message if it is going to throw this "Insufficient privilege level"  when requested

${stderr} = Unable to send RAW command (channel=0x0 netfn=0x32 lun=0x0 cmd=0xf0 rsp=0xd4): Insufficient privilege level 

From the test documentation, here is what it is trying to do.

Test Wrong Reservation_ID
   [Documentation]   This testcase is to test BMC can handle multi-requestor's
   ...               oem partial add command with incorrect reservation id.
   ...               It simulates sending partial add command with fake content
   ...               and wrong Reservation ID. This command will be rejected.

Test Correct Reservation_ID
   [Documentation]   This testcase is to test BMC can handle multi-requestor's
   ...               oem partial add command with correct reservation id. It
   ...                simulates sending partial add command with fake content
   ...               and correct Reservation ID. This command will be accepted.

williamspatrick, 2016-09-09:
It is wrong to call this "lab" mode. Restricted mode is a field mode setting only to be used when the host is leased to another client and you do not want that client to modify BMC settings. Unrestricted mode is just as much a field mode and should be the typical setting for everyone except leased systems.

williamli80, 2016-09-13:
Could you tell me how to change the mode?

rahulmah, 2016-09-20:
@vishwabmc : I found that the system was having restricted_mode set to 0. Later I tried to play around "restricted_mode" (under interface "org.openbmc.settings.Host") by setting it to 1 and 0. But I don't see it makes any difference and I get the same "Insufficient privilege level" error. What value we need to provide for unrestricted mode and restricted mode?

sivassrr, 2016-09-28:
@tomjoseph83 any update - In another issue #572 - you were updating the whitelisted command. Will it be included? Any update on mode switching. @rahulmah not seeing the differences when he switches the mode.

tomjoseph83, 2017-05-08:
@rahulmah The OEM command(Add Partial SELL) is added to the whitelist. This issue can be closed.

gkeishin, 2017-05-10:
@tomjoseph83 @rahulmah 

the test cases is now working good https://gerrit.openbmc-project.xyz/#/c/4157/

rahulmah, 2017-05-11:
Issue is now fix.

closed 2017-05-11 by rahulmah