openbmc/openbmc #543: REST API to add user giving DBusException: org.freedesktop.DBus.Error.NoReply error

rahulmah, 2016-09-05:
REST API to add user giving DBusException: org.freedesktop.DBus.Error.NoReply error. Also this error is seen after 25 sec of sending REST. Cross checked and found that user is getting created and even able to login to the system. So this issue is only with respect to error seen.

11:57:12 Create and delete user without groupname :: ***GOOD PATH***           URI:https://<BMC_IP>/org/openbmc/UserManager/Users/action/UserAdd, method:Post, args:{u'data': {u'data': ['1RheZCLj', 'mzkguuMi', '', 'excQcOzS']}, u'headers': {u'Content-Type': u'application/json'}}
11:57:38 Response code:500, Content:{
11:57:38   "data": {
11:57:38     "description": "Internal Server Error", 
11:57:38     "exception": "DBusException('Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.',)", 
11:57:38     "traceback": [
11:57:38       "Traceback (most recent call last):", 
11:57:38       "  File \"/usr/lib/python2.7/site-packages/bottle.py\", line 862, in _handle", 
11:57:38       "    return route.call(**args)", 
11:57:38       "  File \"/usr/sbin/obmc-rest\", line 524, in __call__", 
11:57:38       "    return self.callback(*a, **kw)", 
11:57:38       "  File \"/usr/sbin/obmc-rest\", line 680, in wrap", 
11:57:38       "    return self.to_jsonp(callback(*a, **kw))", 
11:57:38       "  File \"/usr/lib/python2.7/site-packages/bottle.py\", line 1734, in wrapper", 
11:57:38       "    rv = callback(*a, **ka)", 
11:57:38       "  File \"/usr/sbin/obmc-rest\", line 615, in wrap", 
11:57:38       "    resp = {'data': callback(*a, **kw)}", 
11:57:38       "  File \"/usr/sbin/obmc-rest\", line 578, in wrap", 
11:57:38       "    return callback(*a, **kw)", 
11:57:38       "  File \"/usr/sbin/obmc-rest\", line 603, in wrap", 
11:57:38       "    return callback(*a, **kw)", 
11:57:38       "  File \"/usr/sbin/obmc-rest\", line 93, in __call__", 
11:57:38       "    return getattr(self, 'do_' + request.method.lower())(**kw)", 
11:57:38       "  File \"/usr/sbin/obmc-rest\", line 214, in do_post", 
11:57:38       "    return request.route_data['method'](*request.parameter_list)", 
11:57:38       "  File \"/usr/lib/python2.7/site-packages/dbus/proxies.py\", line 145, in __call__", 
11:57:38       "    **keywords)", 
11:57:38       "  File \"/usr/lib/python2.7/site-packages/dbus/connection.py\", line 651, in call_blocking", 
11:57:38       "    message, timeout)", 
11:57:38       "DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken."
11:57:38     ]
11:57:38   }, 
11:57:38   "message": "500 Internal Server Error", 
11:57:38   "status": "error"
11:57:38 }
11:57:38 
11:57:38 FFDC Collection Started 	: 2016-09-05 11:57:38,730421
11:57:45 FFDC Collection Completed 	: 2016-09-05 11:57:45,641957
11:57:45 | FAIL |
11:57:45 error != ok

Logged into BMC and found that use is created successfully.

root@barreleye:~# cat /etc/passwd | grep mzkguuMi
mzkguuMi:x:1051:1051:1RheZCLj:/home/mzkguuMi:go
root@barreleye:~#

Also found that the new use id able to login using REST.

rahulmah@rackspace:~$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST https://<BMC_IP>/login -d "{\"data\": [ \"mzkguuMi\", \"excQcOzS\" ] }"
{
  "data": "User 'root' logged in",
  "message": "200 OK",
  "status": "ok"
}rahulmah@rackspace:~$

FFDC from system attached below:
[FFDC_UserAdd.txt](https://github.com/openbmc/openbmc/files/455436/FFDC_UserAdd.txt)

rahulmah, 2016-09-08:
@bradbishop : We also see the the same problem with below GET operations.  Seem to be the same problem.  Here also we see the response time is around 25 secs. This response time is very long and needs to be fixed also. 

curl -c cjar -b cjar -k https://<BMC_IP>/enumerate
curl -c cjar -b cjar -k https://<BMC_IP>/org/enumerate

bradbishop, 2016-09-08:
@rahulmah this is nice and quick for me.  can you open a new issue ( I don't think they are the same ) and provide the journal ?

rahulmah, 2016-09-09:
@bradbishop : Okay. I have opened a separate issue(#558) for the error seen with GET operations. FFDC/Journal logs are attached with it. 
https://github.com/openbmc/openbmc/issues/558

closed 2016-09-12 by williamspatrick