openbmc/openbmc #546: Add SLP server.

williamspatrick, 2016-09-07:
Find an open source SLP server and add it to our images.

ratagupt, 2016-09-13:
why do we need both Avahi and slp sever, We can discover the network services with the use of AVAHI.

With the use of Avahi, system can publish and discover the services...Do we want the slp server as we want to discover the system VIA SLP protocol.

williamspatrick, 2016-09-15:
It was requested that we support SLP protocol by IBM manufacturing tools team.

williamspatrick, 2016-10-06:
This is blocked by the FSP manufacturing tools team confirming that using Avahi only is workable.

ratagupt, 2016-10-25:
Manufacturing teaam has a use case which can not be done by avahi..They use the slp to serach the services in unicast manner,hence we need to give the support for slp.
1) We are evaluating openslp
2) We are writing our own daemon(poc) to cater the need of manufacturing.

geissonator, 2016-10-31:
Initial thought was this was just a recipe update but turns out this is going to be fairly more complicated.  Moved from 1 to 3 points to reflect this.

ratagupt, 2016-11-04:
- [ ] Create a server which is listening on 427 for unicast udp packets

- [ ] Write the message parser for the SLPmessage

- [ ] Message Handler for the  message "findsrvtypes"

- [ ] Message Handler for the  message "findsrvs"

ratagupt, 2016-11-09:
- [ ] Make the changes in the BBclass to generate the conf file for slp.

- [ ] Write a function to fetch the all the interfaces on the system which would be required in the findsrvs.

ratagupt, 2016-11-09:
- [ ] Make file creation for the new repo.

ratagupt, 2016-11-14:
- [ ] write the new bb file for the new repo.

geissonator, 2016-11-23:
Not merged, moving sprint.  https://gerrit.openbmc-project.xyz/#/c/1110/

williamspatrick, 2017-01-06:
The recipe for this isn't merged yet.

geissonator, 2017-01-11:
https://gerrit.openbmc-project.xyz/#/c/1211/

closed 2017-01-16 by geissonator