openbmc/openbmc #550: Enhance LED manager to support grouping.

williamspatrick, 2016-09-07:
* Come up with a YAML format to define LEDs and groups.
    * Eventually this will be created from the MRW.
* Create LED and group objects based on YAML content.

williamspatrick, 2016-10-10:
Minutes from 2016-10-10 discussion we had on this topic.

LED management:
    * Key points of the design:
        - Abstract physical LED management from "virtual" LED causes.  Applications should be interested in causes / actions, not how the LED policy applies those onto physical LEDs.
        - Abstract LED policy from code into the MRW.  If two systems have different LED policies, we should not need to change any code for it.  That should be described in the MRW.
    * Low level "led controller" application will provide a dbus object for [physical] LEDs that it controls.  We may end up with different implementations of this in the future, but initially this is an abstraction from /sys/class/led kernel interfaces into dbus.  (In future, can imagine an LED-over-mailbox controller for PHYP controlled LEDs or LED-over-SPCN for those ones.)
    * LED actions are triggered by an "LED group" associated with a high-level activity and containing 0-to-N physical LEDs.  [Potential] Examples:
        - `Fan Present Group : { "Fan Present LED" : on }`
        - `Fan Fault Group : { "Fan Present LED" : blink , "System Fault LED" : on }`
    * Higher level "led manager" is responsible translating group state to physical LED state.  It provides dbus objects for each group to allow them to be asserted / deasserted.  As a result, it will ensure all of the LEDs in that group are in the correct state.
        - For the most part, this is a simple set-union operation.  Each group is a set.  All asserted groups are union'd together.  The union'd state is applied to the physical LED objects.
        - There are some policies to apply such as "blink is higher priority than on", which can be handled in the 'union' operation.
        - The final set apply is two set-intersection operations.  (calculated set vs current set).

vishwabmc, 2016-10-12:
Breakdown of this story into tasks.

- [ ] Create a .yaml file that contains groups and associated LEDs.

closed 2016-11-21 by williamspatrick