openbmc/openbmc #552: settingsd - settings verification.

williamspatrick, 2016-09-07:
* Need regular-expression (and/or plugins or similar) in YAML to document acceptable attribute values.
* Implement handling to reject attribute writes.

These are minutes from an internal design discussion on the concept:

* Data validation.
    * In general the owner of a dbus object is where the validation should be
      done.  May need to change this for 'manager' type objects (settingsd,
      inventory, etc) because they are generic, but still desire to satisfy.
    * Decided to extend settingsd to allow some sort of validation on
      attributes that can provided by regular expressions and/or python
      extension functions.
        * Expect the networkd code will provide those utilities for ip address,
          maybe locate in pyphosphor repository.
    * Need to understand how to intercept org.freedesktop.Object interfaces so
      we can provide validation there and give error back to the user (via REST)
      if done incorrectly.

williamspatrick, 2016-09-08:
Possible validation operators:
  * range (+ type)
  * enum-set
  * regex
  * plugin

vishwabmc, 2016-09-21:
Cod changes at:
https://gerrit.openbmc-project.xyz/#/c/550/
https://gerrit.openbmc-project.xyz/#/c/547/

closed 2016-09-30 by williamspatrick