openbmc/openbmc #553: Replace pflash with direct MTD access.

williamspatrick, 2016-09-08:
In #542, @shenki suggested it may be better to use direct MTD access provided by the kernel instead of pflash's direct register manipulation.  We need to better understand what the design for that would be and then an appropriate refactoring of the host flash code.

My immediate thoughts and questions would be:

1. How does the kernel know when it is acceptable to create the MTD?  Is this a probe / removal done by userspace?  We need to make sure we are not having dual access between the BMC and the Host.

2. Where does the parsing of the FFS TOC happen?  I _think_ there is kernel level support for that on Power, but perhaps that is provided as an abstraction by skiboot.  Do we still need pflash for TOC parsing or can we get support for creating independent MTD partitions?  The FSP driver team also has an FFS implementation, so there may be something we can leverage from them also.

3. For security reasons there was a potential P9 design that would prevent the Host from accessing the SFC entirely and the BMC would present a 'memboot' emulation of the PNOR.  Where are we with that?  If that is the intended Host firmware design, I would propose that we don't even use FFS natively but simply map the PNOR device as a second flash chip for the BMC.  We can be more efficient than the FFS layout is and we can xz compress many of the sections.  We can then utilize the extra space for our own purposes.

shenki, 2016-09-08:
There is not support for parsing the FFS TOC in power. We did a prototype it but didn't stick.

Point 1 and 3 are mutually exclusive. I think we should design for 3, which means the mtd device is present in the device tree and is always owned by the BMC.

williamspatrick, 2016-09-08:
If we want to design for 3 we need to ensure the host firmware team is on board and ready to go for that. We also need a plan for what to do with userspace on P8 machines. It is also a pain to have two different implementations.

rfrandse, 2017-05-31:
remove pflash as part of cleanup

closed 2017-05-31 by rfrandse

williamspatrick, 2017-05-31:
We will be removing pflash in favor of UBI-managed volumes.

ozbenh, 2017-06-01:
Don't remove it until the alternative exists. AFAIK the current pflash *will* use /dev/mtd when available and has the whole FFS stuff we need when updating skiboot etc... 

So until you have the new UBI stuff, please keep pflash around