openbmc/openbmc #555: REST clear event log action is not purging the events on witherspoon & palmetto

SrideviRamesh, 2016-09-08:
REST clear event log action is not purging the events on witherspoon 

Failure captured through automation
-------------------------------------------
Length is 14 	
15:30:51.665 	FAIL 	'[u'/org/openbmc/records/events/24', u'/org/openbmc/records/events/25', u'/org/openbmc/records/events/14', u'/org/openbmc/records/events/17', u'/org/openbmc/records/events/16', u'/org/openbmc/records/events/31', u'/org/openbmc/records/events/22', u'/org/openbmc/records/events/29', u'/org/openbmc/records/events/1', u'/org/openbmc/records/events/9', u'/org/openbmc/records/events/2', u'/org/openbmc/records/events/7', u'/org/openbmc/records/events/11', u'/org/openbmc/records/events/8']' should be empty.

Same issue is seen when tried through manually on witherspoon & palmetto using CURL. 

}bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type:application/json" -X GET https://9.3.185.29/org/openbmc/records/events/
{
  "data": [
    "/org/openbmc/records/events/11",
    "/org/openbmc/records/events/24",
    "/org/openbmc/records/events/14",
    "/org/openbmc/records/events/17",
    "/org/openbmc/records/events/16",
    "/org/openbmc/records/events/32",
    "/org/openbmc/records/events/31",
    "/org/openbmc/records/events/30",
    "/org/openbmc/records/events/37",
    "/org/openbmc/records/events/29",
    "/org/openbmc/records/events/35",
    "/org/openbmc/records/events/1",
    "/org/openbmc/records/events/2",
    "/org/openbmc/records/events/25",
    "/org/openbmc/records/events/7",
    "/org/openbmc/records/events/22",
    "/org/openbmc/records/events/9",
    "/org/openbmc/records/events/8",
    "/org/openbmc/records/events/28"
  ],
  "message": "200 OK",
  "status": "ok"
}

bash-4.1$ curl -c cjar -b cjar -k -H "Content-Typeapplication/json" -X POST -d "{\"data\": []}" https://9.3.164.153/org/openbmc/records/events/action/clear
{
  "data": 0,
  "message": "200 OK",
  "status": "ok"
}bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type:application/json" -XGET https://9.3.164.153/org/openbmc/records/events/r
{
  "data": [
    "/org/openbmc/records/events/300",
    "/org/openbmc/records/events/301"
  ],
  "message": "200 OK",
  "status": "ok"
}

FFDC  Logs is attached

[witherspoon_openbmc_ffdc_report.txt](https://github.com/openbmc/openbmc/files/461635/witherspoon_openbmc_ffdc_report.txt)

gkeishin, 2016-09-08:
@bradbishop  is this something we need to worry about ?   I see this in barreleye working good..

bradbishop, 2016-09-08:
It just seems like some more events were created after the log was purged?

gkeishin, 2016-09-08:
Will cross check that one from the log if new are created post clearing.. Thanks Brad...

gkeishin, 2016-09-08:
@bradbishop  Nope.. the one she post here seems she did trial latter manually.. but from the log

Before the clear : 
 	Response code:200, Content:{
  "data": [
    "/org/openbmc/records/events/24", 
    "/org/openbmc/records/events/25", 
    "/org/openbmc/records/events/14", 
    "/org/openbmc/records/events/17", 
    "/org/openbmc/records/events/16", 
    "/org/openbmc/records/events/31", 
    "/org/openbmc/records/events/22", 
    "/org/openbmc/records/events/28", 
    "/org/openbmc/records/events/29", 
    "/org/openbmc/records/events/1", 
    "/org/openbmc/records/events/9", 
    "/org/openbmc/records/events/2", 
    "/org/openbmc/records/events/7", 
    "/org/openbmc/records/events/11", 
    "/org/openbmc/records/events/8"
  ], 
  "message": "200 OK", 
  "status": "ok"
}

After Clear 
   It matches what was before and after trying clear... 

There is FFDC attached the the post.. probably if that helps ?

sivassrr, 2016-12-06:
Any update?
@SrideviRamesh could you check it on latest level. I think we are not seeing it, I guess.

gkeishin, 2016-12-06:
We haven't seen this so far in many builds we tested since this.. Closing this ..

closed 2016-12-06 by gkeishin