openbmc/openbmc #556: ipmi trace shows an error on a mac property

causten, 2016-09-08:
On each boot I see Opal asking this command and BMC returning a FF back (seen on a Barreleye not sure about others)).  Can you investigate to see if this is an issue and how to resolve it...

IPMI Incoming: Seq 0x0e, NetFn 0x00, CMD: 0x09
0x000000: 61 00 00                                        a..             
IPMI GET_SYS_BOOT_OPTIONS
IPMID boot option property get: {ipaddress=,prefix=,gateway=,mac=,addr_type=}.
Configuration String[ipaddress=,prefix=,gateway=,mac=,addr_type=]
 host_config.macaddress.c_str()=[]
sscanf Failed in extracting mac address.
getHostNetworkData failed for get_sys_boot_options.
ERROR:[0xFF] handling NetFn:[0x0], Cmd:[0x9]
IPMI Response:
0x000000: ff 01 61 80 21 70 62 21 00 01 06 04 00 00 00 00 ..a.!pb!........
0x000010: 00 00 00 00 00 00 00 00 00 00 00                ...........    

root@barreleye:~# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-7"
VERSION_ID="v1.99.0-7-g167aaae-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-7"

williamspatrick, 2016-09-08:
@ratagupt - I know you have code to parse the boot options and we left in a default which is purposefully "invalid".  It seems like this is still reporting an error.  Is that intended?  We shouldn't have errors on the default good-path.

Are we suppose to send an error for a "there are no boot overrides" case or are we suppose to send a good status?

ratagupt, 2016-10-13:
@williamspatrick :- Yes it was intended that we should not give false info to the petiboot,thats why during parsing of the data it got failed as nothing has been set.

rfrandse, 2017-07-18:
<https://gerrit.openbmc-project.xyz/5179> Make changes as per the new settings infrastructure
Resolves  #556 ipmi trace shows an error on a mac property

closed 2017-07-19 by williamspatrick