openbmc/openbmc #557: PreTag  openbmc/skeleton#43  - Set Less Than Minimum PowerCAP

williamspatrick, 2016-09-08:
@gkeishin commented on [Tue May 31 2016](https://github.com/openbmc/skeleton/issues/97)

Please refer to https://github.com/mkumatag/openbmc-automation/issues/61
for detail description...

Our concern was if this is some sort of REST server code bug returning Error even when that PowerCap property existing on the system..

If by design ( openbmc/skeleton#43 ) if itis not going to return HTTP OK ... then, why is this return HTTP OK here on Palmetto and throwing error on Barrelleye when PowerCap Property exist on both the systems..

Example:

On Palmetto :
openbmc-automation$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST -d "{\"data\": [ 1900 ]}" http://9.3.164.177/org/openbmc/sensors/host/PowerCap/action/setValue
{
"data": null,
"message": "200 OK",
"status": "ok"
}

On Barrelleye:
openbmc-automation$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST-d "{\"data\": [ 1900 ]}" https://9.3.23.24/org/openbmc/sensors/host/PowerCap/action/setValue
{
"data": {
"description": "org.openbmc.objectmapper.Error.NotFound: path or object not found: /org/openbmc/sensors/host/PowerCap"
},
"message": "404 Not Found",
"status": "error"
}

---

@gkeishin commented on [Thu Sep 08 2016](https://github.com/openbmc/skeleton/issues/97#issuecomment-245676332)

We are seeing the powercap setting errors now frequently.
Tests . Test Generic Conf
7 Test  4 pass   3  3 failed

out of 7 , 3 are failing

/org/openbmc/settings/host0, power_cap
1. Set the power with string of characters (abcdefg)
2. Set the power with greater then MAX_POWER_VALUE (1010)

/org/openbmc/settings/host0, boot_flags
1. Set the boot flags with string ( 3ab56f )

We would like to know what are the valid policies and what are not from those attributes..

williamspatrick, 2016-09-08:
I don't really understand what the issue is asking.  Is this about PowerCap sensors not being present?  Or is this about settings not being set properly?  The latest comment makes it sound like you are expecting some sort of verification of invalid settings, which is not currently implemented.  #552 will give us the ability to do that.

causten, 2016-09-08:
the test case was written to check if invalid settings could be set.  The max for pcap is 1000 so teh test was to set it above that level.  It allowed a value of 1010.  Hence the failure.  Same with setting the max pcap to abcdefg.  So there is no validation code in the pcap code.  If you have input validation handled via issue 552 then issue 557 should wait until 552 is ready

gkeishin, 2016-09-08:
Thanks will keep an eye on #552

anoo1, 2016-09-12:
The power cap value was eventually implemented in the sensors path after the placeholder in the settings file was created (with its arbitrary default value of 100).
We should determine if the power cap should be in 2 places and synced (sensors and settings), or just one place and remove the redundant one.

gkeishin, 2016-09-15:
another for powercap

Not able to set here /org/openbmc/sensors/host/powercap/action/setValue
The interfaces under sensors/powercap are
{
"data": [
"/org/openbmc/sensors/powercap/user_cap",
"/org/openbmc/sensors/powercap/system_power",
"/org/openbmc/sensors/powercap/curr_cap",
"/org/openbmc/sensors/powercap/n_cap",
"/org/openbmc/sensors/powercap/max_cap",
"/org/openbmc/sensors/powercap/min_cap"
], 

and we are able to Set here /org/openbmc/settings/host0,power_cap

So there seems not to be clear until this is streamlined.. 

@adamliyi  any thoughts ?

gkeishin, 2018-05-28:
Closing this as the implementation are working

closed 2018-05-28 by gkeishin