openbmc/openbmc #562: [Barreleye VERSION_ID="v1.99.0-42-gc7419c6-dirty" ]  BMC not booted READY nor REST service up

gkeishin, 2016-09-10:
FFDC attached here :  
[openbmc_ffdc_report.txt](https://github.com/openbmc/openbmc/files/465221/openbmc_ffdc_report.txt)

root@barreleye:~# obmcutil state
 = BMC_STARTING
root@barreleye:~# obmcutil state
 = BMC_STARTING
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-42"
VERSION_ID="v1.99.0-42-gc7419c6-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-42"
BUILD_ID="v1.99.0"
root@barreleye:~# uptime
 04:19:37 up 7 min,  load average: 5.21, 4.55, 2.42
root@barreleye:~# cat /etc/version
20160909203850
root@barreleye:~# obmcutil state
 = BMC_STARTING
root@barreleye:~#

Sep 10 04:22:44 barreleye dropbear[1073]: Password auth succeeded for 'root' from ::ffff:9.84.230.24:50001
Sep 10 04:22:46 barreleye phosphor-gevent[1070]: Traceback (most recent call last):
Sep 10 04:22:46 barreleye phosphor-gevent[1070]:   File "/usr/sbin/phosphor-gevent", line 30, in <module>
Sep 10 04:22:46 barreleye phosphor-gevent[1070]:     exec 'from obmc.wsgi.apps.%s import App' % sys.argv[1]
Sep 10 04:22:46 barreleye phosphor-gevent[1070]:   File "<string>", line 1, in <module>
Sep 10 04:22:46 barreleye phosphor-gevent[1070]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 22, in <module>
Sep 10 04:22:46 barreleye phosphor-gevent[1070]:     from bottle import Bottle, abort, request, response, JSONPlugin, HTTPError
Sep 10 04:22:46 barreleye phosphor-gevent[1070]:   File "/usr/lib/python2.7/site-packages/bottle.py", line 38, in <module>
Sep 10 04:22:46 barreleye phosphor-gevent[1070]:     import base64, cgi, email.utils, functools, hmac, imp, itertools, mimetypes,\
Sep 10 04:22:46 barreleye phosphor-gevent[1070]: ImportError: No module named cgi
Sep 10 04:22:46 barreleye systemd[1]: phosphor-gevent.service: Main process exited, code=exited, status=1/FAILURE
Sep 10 04:22:46 barreleye systemd[1]: phosphor-gevent.service: Unit entered failed state.
Sep 10 04:22:46 barreleye systemd[1]: phosphor-gevent.service: Failed with result 'exit-code'.
Sep 10 04:22:46 barreleye systemd[1]: phosphor-gevent.service: Service hold-off time over, scheduling restart.
Sep 10 04:22:46 barreleye systemd[1]: Stopped Phosphor Webserver.
Sep 10 04:22:46 barreleye systemd[1]: Started Phosphor Webserver.
Sep 10 04:22:51 barreleye dropbear[1076]: lastlog_perform_login: Couldn't stat /var/log/lastlog: No such file or directory
Sep 10 04:22:51 barreleye dropbear[1076]: lastlog_openseek: /var/log/lastlog is not a file or directory!
Sep 10 04:22:52 barreleye inventory_items.py[1075]: Traceback (most recent call last):
Sep 10 04:22:52 barreleye inventory_items.py[1075]:   File "/usr/sbin/inventory_items.py", line 87, in <module>
Sep 10 04:22:52 barreleye inventory_items.py[1075]:     INVENTORY_FILE = os.path.join(sys.prefix, 'share',
Sep 10 04:22:52 barreleye inventory_items.py[1075]: NameError: name 'os' is not defined 
------------------------------------------------------------------------ 

--- Others build information ---
Opening URL: https://openpower.xyz/job/openbmc-build/

Build Name: #221-c7419c67e80b26 	 Build Number: 221

---------------------------------------------------------
-----------------
Build Info:
-----------------
Build #221-c7419c67e80b26 (Sep 9, 2016 8:35:30 PM) 

-----------------
Build Change info:
-----------------

Changes
Git (git https://github.com/openbmc/openbmc.git)
systemd - Shrink systemd package contents (commit: c7419c6) (detail)
Started by GitHub push by williamspatrick
Revision: c7419c67e80b263872268b23cfc87410a3ada8bc
refs/remotes/origin/master

---------------------------------------------------------

Image Name: barreleye-20160909203850.all.tar 	 Size: 32.00 MB

bradbishop, 2016-09-10:
I noticed this the other day:
https://gerrit.openbmc-project.xyz/#/c/439/

closed 2016-09-10 by williamspatrick

bradbishop, 2016-09-10:
And for the rest server: 
https://gerrit.openbmc-project.xyz/#/c/460/1