openbmc/openbmc #563: [Barreleye VERSION_ID="v1.99.0-47-g194b590-dirty" ]  Basic Poweron Failed - Host not booted

gkeishin, 2016-09-12:
A basic poweron failed to boot host to HOST_BOOTED state. 

FFDC attached with journal data here --> 
[openbmc_ffdc_report.txt](https://github.com/openbmc/openbmc/files/466593/openbmc_ffdc_report.txt)

Response code:200, Content:{
  "data": "HOST_BOOTING",
  "message": "200 OK",
  "status": "ok"
}

FFDC Collection Started         : 2016-09-12 00:33:56,717780
FFDC Collection Completed       : 2016-09-12 00:34:04,997769
| FAIL |
Keyword 'Is Host Booted' failed after retrying for 15 minutes. The last error was: 'HOST_BOOTING' does not contain 'HOST_BOOTED'
------------------------------------------------------------------------------
Tests.Test Basic Poweron :: This module will test basic power on u... | FAIL |
1 critical test, 0 passed, 1 failed
1 test total, 0 passed, 1 failed

root@barreleye:~# cat /etc/version
20160912025124
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-47"
VERSION_ID="v1.99.0-47-g194b590-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-47"
BUILD_ID="v1.99.0"
root@barreleye:~# obmcutil state
 = HOST_BOOTING
root@barreleye:~# uptime
 21:35:46 up 14 min,  load average: 0.81, 0.70, 0.72
root@barreleye:~#

-- Other build info ---

Build Name: #226-194b590eb913cc 	 Build Number: 226

---------------------------------------------------------
-----------------
Build Info:
-----------------
Build #226-194b590eb913cc (Sep 12, 2016 2:48:05 AM) 

-----------------
Build Change info:
-----------------

Changes
Git (git https://github.com/openbmc/openbmc.git)
skeleton-rev.bbclass: bump version (commit: 194b590) (detail)
Started by GitHub push by williamspatrick
Revision: 194b590eb913cc8463c2c7774865d6bc8b07aa17
refs/remotes/origin/master

---------------------------------------------------------

williamspatrick, 2016-09-12:
I don't understand how this is related to #564, can you elaborate?  Is there a journal output?  Or is the underlying problem that the network went down?  In that case, JSON queries would also fail, so did you check the system state through the console?

gkeishin, 2016-09-12:
When i tested i see both of them.. So i just split this into two different issues.. Not related but just same driver test.. so mentioned just in case if there could be related issues... 

The FFDC is attached there in the above opening statment with journal info ..

gkeishin, 2016-09-13:
The newer build also failed to power on host to booted stated

root@barreleye:~# cat /etc/version
20160913033422
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-60"
VERSION_ID="v1.99.0-60-g2b717d8-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-60"
BUILD_ID="v1.99.0"
root@barreleye:~# uptime
 21:38:25 up 17 min,  load average: 0.17, 0.45, 0.63
root@barreleye:~# obmcutil state
 = HOST_POWERED_ON
root@barreleye:~# obmcutil state
 = HOST_BOOTING
root@barreleye:~#

Response code:200, Content:{
  "data": "HOST_BOOTING",
  "message": "200 OK",
  "status": "ok"
}

FFDC Collection Started         : 2016-09-12 23:51:01,471417
FFDC Collection Completed       : 2016-09-12 23:51:10,257658
| FAIL |
Keyword 'Is Host Booted' failed after retrying for 15 minutes. The last error was: 'HOST_BOOTING' does not contain 'HOST_BOOTED'
------------------------------------------------------------------------------
Tests.Test Basic Poweron :: This module will test basic power on u... | FAIL |
1 critical test, 0 passed, 1 failed
1 test total, 0 passed, 1 failed

--- Driver Build info ---
Build Name: #235-2b717d8489c1f5 	 Build Number: 235

---------------------------------------------------------
-----------------
Build Info:
-----------------
Build #235-2b717d8489c1f5 (Sep 13, 2016 3:31:14 AM) 

-----------------
Build Change info:
-----------------

Changes
Git (git https://github.com/openbmc/openbmc.git)
obmc-mapper.bb: bump version (commit: 2b717d8) (detail)
Started by GitHub push by williamspatrick
Revision: 2b717d8489c1f52464974bb72bda1f66aac24332
refs/remotes/origin/master

gkeishin, 2016-09-14:
This problem of not able to boot the host is going on for few driver as we can see in above post as well.

In today's build, post code update, we were able to see once to HOST_BOOTED state, post that once powered off and power on it never came back up to HOST_BOOTED state.

There are FFDC logs journal, SOL console log attached bellow..   

@williamspatrick  anything we would need to follow up on this ?

------------------------------------------------------------------------
APPLICATION DATA : BMC state    Executed : /usr/sbin/obmcutil  state
------------------------------------------------------------------------
 = HOST_BOOTING

------------------------------------------------------------------------
DRIVER INFO : FW Level  Executed : cat /etc/os-release
------------------------------------------------------------------------
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-75"
VERSION_ID="v1.99.0-75-gc36e20e-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-75"
BUILD_ID="v1.99.0"

------------------------------------------------------------------------
DRIVER INFO : Build Info    Executed : cat /etc/version
------------------------------------------------------------------------
20160913205350

**-- More info on builds --**
Build Name: #239-c36e20e583d937      Build Number: 239
---------------------------------------------------------
-----------------
Build Info:
-----------------
Build #239-c36e20e583d937 (Sep 13, 2016 8:48:46 PM) 
-----------------
Build Change info:
-----------------
Changes
Git (git https://github.com/openbmc/openbmc.git)
network: add missing systemd dep (commit: c36e20e) (detail)
Started by GitHub push by williamspatrick
Revision: c36e20e583d937052a0564794fa7f8e6be3264a7
refs/remotes/origin/master
---------------------------------------------------------

**FFDC Logs** 
[20160913232108674226_BMC_journalctl.txt](https://github.com/openbmc/openbmc/files/471363/20160913232108674226_BMC_journalctl.txt)

-bash-4.1$ tree 20160913232108674226_TestsTestBasicPoweron/
20160913232108674226_TestsTestBasicPoweron/
└── 20160913232108674226_powerontest
    ├── 20160913232108674226_BMC_journalctl.log
    ├── 20160913232108674226_BMC_proc_list
    └── 20160913232108674226_ffdc_report.txt

1 directory, 3 files

**CONSOLE LOG:**
[Console_log.txt](https://github.com/openbmc/openbmc/files/471360/Console_log.txt)

gkeishin, 2016-09-20:
Newer build 251 is also taking the host to HOST_BOOTED state
Attached SOL log here --> 
[CONSOLE_BARRELEYE_LOG.txt](https://github.com/openbmc/openbmc/files/481766/CONSOLE_BARRELEYE_LOG.txt)
 

root@barreleye:~# cat /etc/version
20160917130947
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-108"
VERSION_ID="v1.99.0-108-g5d5f8df-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-108"
BUILD_ID="v1.99.0"

williamspatrick, 2016-09-21:
We found an issue where the IPMI server is crashing due to a rename of some of the inventory objects.  @spinler will be fixing that, which should resolve this problem.  (With the IPMI server crashed, it was not getting sensor updates from hostboot / skiboot which indicate the boot is "done", thus we never move to HOST_BOOTED state.)

closed 2016-09-21 by williamspatrick