openbmc/openbmc #568: Configure GPIOs in userspace

shenki, 2016-09-13:
The BMC needs to set the GPIO state so the host can power on.

It also needs to recognise when the BMC has been rebooted while the host is powered on.

shenki, 2016-09-13:
1. Andrew (@amboar) will send his GPIO-hog dt patch for inclusion in the openbmc repository as a patch to the device tree
2. Joel will boot test this on the two Palmettos he has access to, including booting the host, rebooting the host, and rebooting the BMC while the host is powered on. Results will be included in the commit message.
3. The patch will be merged into the tree

This will hard-code the GPIOs used for APSS, FSI and other functions to a set state, and they will not be available to userspace. This may affect using Cronus.

The list of pins on Palmetto are:

| Net name              | GPIO   |
| --------------------- | ------ |
|BMC_FAN_RESERVED_N     | GPIO A0|
|APSS_WDT_N             | GPIO A1|
|APSS_BOOT_MODE         | GPIO B1|
|APSS_RESET_N           | GPIO B2|
|SPIVID_STBY_RESET_N    | GPIO B7|
|BMC_POWER_UP           | GPIO D1|
|BMC_BATTERY_TEST       | GPIO F1|
|AST_HW_FAULT_N         | GPIO F4|
|AST_SYS_FAULT_N        | GPIO F5|
|BMC_FULL_SPEED_N       | GPIO F7|
|BMC_FAN_ERROR_N        | GPIO G3|
|BMC_WDT_RST1_P         | GPIO G4|
|BMC_WDT_RST2_P         | GPIO G5|
|PE_SLOT_TEST_EN_N      | GPIO H0|
|BMC_RTCRST_N           | GPIO H1|
|SYS_POWEROK_BMC        | GPIO H2|
|SCM1_FSI0_DATA_EN      | GPIO H6|
|BMC_TMP_INT_N          | GPIO H7|

amboar, 2016-09-15:
> Andrew (@amboar) will send his GPIO-hog dt patch for inclusion in the openbmc repository as a patch to the device tree

Pushed: https://gerrit.openbmc-project.xyz/509

geissonator, 2016-09-19:
I picked up https://gerrit.openbmc-project.xyz/#/c/510/ against openbmc master.  This seems to have regressed me on my palm205 system though.  We fail to boot into hostboot (hit watchdog)

When I power on, I get this error:

> root@palmetto:~# obmcutil poweron
>  = None
> root@palmetto:~# ftgmac100 1e660000.ethernet eth0: NCSI interface down
> ftgmac100 1e660000.ethernet eth0: NCSI interface up
> aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: pin AB1 already requested by 1e6e2000.syscon:pinctrl@1e6e2000; cannot claim for 1e780000.gpio:427
> aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: pin-107 (1e780000.gpio:427) status -22
> aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: pin W5 already requested by 1e6e2000.syscon:pinctrl@1e6e2000; cannot claim for 1e780000.gpio:428
> aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: pin-108 (1e780000.gpio:428) status -22

Here's the relevant messages from startup:

> GPIO line 340 (func_mode0) hogged as output/low
> GPIO line 341 (func_mode1) hogged as output/low
> GPIO line 342 (func_mode2) hogged as output/low

root@palmetto:~# uname -a
Linux palmetto 4.7.3-be2001133835be80ba4657232b823ba2daaac0d6 #1 Mon Sep 19 14:36:58 UTC 2016 armv5tejl GNU/Linux

I assume https://gerrit.openbmc-project.xyz/#/c/510/ has the patch from https://gerrit.openbmc-project.xyz/#/c/509/ within it or was there something else I needed to pick up?

geissonator, 2016-09-19:
Actually, I went to just "master" and got the same fails so apparently not related to this patch but something else that's gone in since I last tried master on this system.

[palmetto]# uname -a
Linux palmetto 4.7.1-6d93f625315a1e9b51d017a78f8c0938a097ea85 #1 Thu Aug 25 12:51:31 UTC 2016 armv5tejl GNU/Linux

Some additional data from the fail in the journal:

> Sep 19 14:33:42 palmetto system_manager.py[738]: Running System State: HOST_POWERED_ON
> Sep 19 14:33:42 palmetto systemd[1]: Started Wait for Power0 to turn on.
> Sep 19 14:33:42 palmetto systemd[1]: Reached target Power0 On.
> Sep 19 14:33:42 palmetto systemd[1]: Reached target Power0 (On).
> Sep 19 14:33:42 palmetto systemd[1]: Starting PCIE Slot Presence Detection For Chassis0...
> Sep 19 14:33:42 palmetto kernel: ftgmac100 1e660000.ethernet eth0: NCSI interface up
> Sep 19 14:33:42 palmetto kernel: random: nonblocking pool is initialized
> Sep 19 14:33:42 palmetto kernel[694]: ftgmac100 1e660000.ethernet eth0: NCSI interface up
> Sep 19 14:33:42 palmetto kernel[694]: random: nonblocking pool is initialized
> Sep 19 14:33:43 palmetto pcie_slot_present.exe[832]: Checking Presence: SLOT0_RISER_PRESENT
> Sep 19 14:33:43 palmetto system_manager.py[738]: ERROR SystemManager: dbus.String(u'SLOT0_RISER_PRESENT') not found in lookup
> Sep 19 14:33:43 palmetto pcie_slot_present.exe[832]: Checking Presence: SLOT1_RISER_PRESENT
> Sep 19 14:33:43 palmetto system_manager.py[738]: ERROR SystemManager: dbus.String(u'SLOT1_RISER_PRESENT') not found in lookup
> Sep 19 14:33:43 palmetto pcie_slot_present.exe[832]: Checking Presence: SLOT2_RISER_PRESENT
> Sep 19 14:33:43 palmetto system_manager.py[738]: ERROR SystemManager: dbus.String(u'SLOT2_RISER_PRESENT') not found in lookup
> Sep 19 14:33:43 palmetto pcie_slot_present.exe[832]: Checking Presence: SLOT0_PRESENT
> Sep 19 14:33:43 palmetto hwmon.py[760]: WARNING - hwmon: Unhandled hwmon: /sys/class/hwmon/hwmon0/
> Sep 19 14:33:44 palmetto pcie_slot_present.exe[832]: GPIO Lookup:  SLOT0_PRESENT = 427,in
> Sep 19 14:33:44 palmetto inventory_items.py[689]: Caching: /org/openbmc/inventory/system/chassis/io_board/pcie_slot0
> Sep 19 14:33:44 palmetto pcie_slot_present.exe[832]: ERROR pcie_slot_present: GPIO error SLOT0_PRESENT (rc=16)
> Sep 19 14:33:44 palmetto pcie_slot_present.exe[832]: Checking Presence: SLOT1_PRESENT
> Sep 19 14:33:44 palmetto kernel: aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: pin AB1 already requested by 1e6e2000.syscon:pinctrl@1e6e2000; cannot claim for 1e780000.gpio:427
> Sep 19 14:33:44 palmetto kernel: aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: pin-107 (1e780000.gpio:427) status -22
> Sep 19 14:33:44 palmetto kernel[694]: aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: pin AB1 already requested by 1e6e2000.syscon:pinctrl@1e6e2000; cannot claim for 1e780000.gpio:427
> Sep 19 14:33:44 palmetto kernel[694]: aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: pin-107 (1e780000.gpio:427) status -22
> Sep 19 14:33:44 palmetto inventory_upload.py[827]: Uploading inventory to PNOR in dtb format...
> Sep 19 14:33:44 palmetto pcie_slot_present.exe[832]: GPIO Lookup:  SLOT1_PRESENT = 428,in
> Sep 19 14:33:44 palmetto inventory_items.py[689]: Caching: /org/openbmc/inventory/system/chassis/io_board/pcie_slot1
> Sep 19 14:33:44 palmetto pcie_slot_present.exe[832]: ERROR pcie_slot_present: GPIO error SLOT1_PRESENT (rc=16)
> Sep 19 14:33:44 palmetto pcie_slot_present.exe[832]: Checking Presence: SLOT2_PRESENT
> Sep 19 14:33:44 palmetto system_manager.py[738]: ERROR SystemManager: dbus.String(u'SLOT2_PRESENT') not found in lookup
> Sep 19 14:33:45 palmetto pcie_slot_present.exe[832]: Checking Presence: MEZZ0_PRESENT
> Sep 19 14:33:45 palmetto kernel: aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: pin W5 already requested by 1e6e2000.syscon:pinctrl@1e6e2000; cannot claim for 1e780000.gpio:428
> Sep 19 14:33:45 palmetto kernel: aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: pin-108 (1e780000.gpio:428) status -22
> Sep 19 14:33:44 palmetto systemd[1]: Started PCIE Slot Presence Detection For Chassis0.
> Sep 19 14:33:45 palmetto system_manager.py[738]: ERROR SystemManager: dbus.String(u'MEZZ0_PRESENT') not found in lookup
> Sep 19 14:33:45 palmetto system_manager.py[738]: ERROR SystemManager: dbus.String(u'MEZZ1_PRESENT') not found in lookup
> Sep 19 14:33:44 palmetto kernel[694]: aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: pin W5 already requested by 1e6e2000.syscon:pinctrl@1e6e2000; cannot claim for 1e780000.gpio:428
> Sep 19 14:33:45 palmetto pcie_slot_present.exe[832]: Checking Presence: MEZZ1_PRESENT
> Sep 19 14:33:44 palmetto kernel[694]: aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: pin-108 (1e780000.gpio:428) status -22
> Sep 19 14:33:45 palmetto inventory_upload.py[827]: Partition 'BMC_INV' not found !
> Sep 19 14:33:45 palmetto systemd[1]: Started Inventory Upload.

amboar, 2016-09-20:
> I assume https://gerrit.openbmc-project.xyz/#/c/510/ has the patch from https://gerrit.openbmc-project.xyz/#/c/509/ within it or was there something else I needed to pick up?

@geissonator you need to apply both patches from gerrit; first 510 and then 509. Alternatively, from the Download menu on 509, copy and run the checkout line:

    git fetch https://gerrit.openbmc-project.xyz/openbmc/openbmc refs/changes/09/509/2 && git checkout FETCH_HEAD

I'll look into the `*_PRESENT` errors.

geissonator, 2016-09-20:
Per our call, we'll rebase the patches so we'll get a jenkins build that has both fixes and then we'll all use that jenkins image for testing across all Palmetto's.  Please post jenkins link when available, thanks!

amboar, 2016-09-20:
@geissonator it just occurred to me that the comments above should have been made against #513 where the rest of the conversation has taken place. As per the title this issue is for the long-term approach of doing GPIOs from userspace, not the short-term workaround with devicetree hogs.

Regardless, your build is available from:

https://openpower.xyz/job/openbmc-build-gerrit-trigger-multi/distro=ubuntu,target=palmetto/665/

Please continue the conversation in #513.

closed 2019-04-27 by stale[bot]