openbmc/openbmc #569: Firestone obmc kernel failed to boot

adamliyi, 2016-09-13:
OBMC Version: v1.99.0
kernel: cuImage--4.7+git0+6d93f62531-r0-firestone-20160909065156.bin

Netboot on firestone, obmc hangs at kernel booting:

```
boot# bootm 0x40000000 0x40800000
## Booting kernel from Legacy Image at 40000000 ...
   Image Name:   Phosphor OpenBMC (Phosphor OpenB
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    1555652 Bytes = 1.5 MiB
   Load Address: 40008000
   Entry Point:  40008000
## Loading init Ramdisk from Legacy Image at 40800000 ...
   Image Name:   obmc-phosphor-image
   Image Type:   ARM Linux RAMDisk Image (uncompressed)
   Data Size:    11304530 Bytes = 10.8 MiB
   Load Address: 40800000
   Entry Point:  40800000
   Loading Kernel Image ... OK

Starting kernel ...

Uncompressing Linux... done, booting the kernel.
```

shenki, 2016-09-13:
What are your `bootargs`?

Please make sure they contain `earlyprintk`.

williamspatrick, 2016-09-13:
@madscientist159 also had this problem at one point on his Firestone.  I think there is something real here.

madscientist159, 2016-09-13:
@williamspatrick I have a (WIP) tree here with various fixes for Firestone.  This branch reenables the serial console (among other things) but still has a ways to go before I am comfortable issuing a pull request:

https://github.com/madscientist159/linux-openbmc/tree/dev-4.7-firestone-fixes

adamliyi, 2016-09-14:
@shenki, here is the log with earlyprintk:

```
ast# bootm 0x40000000 0x40800000
## Booting kernel from Legacy Image at 40000000 ...
   Image Name:   Phosphor OpenBMC (Phosphor OpenB
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    1555652 Bytes = 1.5 MiB
   Load Address: 40008000
   Entry Point:  40008000
## Loading init Ramdisk from Legacy Image at 40800000 ...
   Image Name:   obmc-phosphor-image
   Image Type:   ARM Linux RAMDisk Image (uncompressed)
   Data Size:    11304530 Bytes = 10.8 MiB
   Load Address: 40800000
   Entry Point:  40800000
   Loading Kernel Image ... OK

Starting kernel ...

Uncompressing Linux... done, booting the kernel.
Booting Linux on physical CPU 0x0
Linux version 4.7.1-6d93f625315a1e9b51d017a78f8c0938a097ea85 (adam@ubuntu) (gcc version 5.3.0 (GCC) ) #1 Fri Sep 9 15:22:36 CST 2016
CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=0005317f
CPU: VIVT data cache, VIVT instruction cache
Machine model: Firestone BMC
bootconsole [earlycon0] enabled
Memory policy: Data cache writeback
SOC Rev: 02010303
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 125984
Kernel command line: console=ttyS4,38400n8 root=/dev/ram rw earlyprintk=serial,ttyS4,38400n8
PID hash table entries: 2048 (order: 1, 8192 bytes)
Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
Memory: 486996K/507904K available (3175K kernel code, 130K rwdata, 728K rodata, 1024K init, 110K bss, 20908K reserved, 0K cma-reserved)
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xdf800000 - 0xff800000   ( 512 MB)
    lowmem  : 0xc0000000 - 0xdf000000   ( 496 MB)
      .text : 0xc0008000 - 0xc04cff28   (4896 kB)
      .init : 0xc0500000 - 0xc0600000   (1024 kB)
      .data : 0xc0600000 - 0xc0620970   ( 131 kB)
       .bss : 0xc0620970 - 0xc063c288   ( 111 kB)
NR_IRQS:16 nr_irqs:16 16
clocksource: moxart_timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 39817925974 ns
sched_clock: 32 bits at 100 Hz, resolution 10000000ns, wraps every 21474836475000000ns
Calibrating delay loop... 190.87 BogoMIPS (lpj=954368)
pid_max: default: 32768 minimum: 301
Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
CPU: Testing write buffer coherency: ok
Setting up static identity map for 0x40100000 - 0x4010003c
devtmpfs: initialized
clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
pinctrl core: initialized pinctrl subsystem
NET: Registered protocol family 16
DMA: preallocated 256 KiB pool for atomic coherent allocations
clocksource: Switched to clocksource moxart_timer
NET: Registered protocol family 2
TCP established hash table entries: 4096 (order: 2, 16384 bytes)
TCP bind hash table entries: 4096 (order: 2, 16384 bytes)
TCP: Hash tables configured (established 4096 bind 4096)
UDP hash table entries: 256 (order: 0, 4096 bytes)
UDP-Lite hash table entries: 256 (order: 0, 4096 bytes)
NET: Registered protocol family 1
RPC: Registered named UNIX socket transport module.
RPC: Registered udp transport module.
RPC: Registered tcp transport module.
RPC: Registered tcp NFSv4.1 backchannel transport module.
[... 64 lines cut ...]
systemd[1]: Mounting Configuration File System...
systemd[1]: Starting Apply Kernel Variables...
systemd[1]: Created slice system-serial\x2dgetty.slice.
systemd[1]: Starting Remount Root and Kernel File Systems...
systemd[1]: Starting Journal Service...
systemd[1]: Reached target Paths.
systemd[1]: Listening on Network Service Netlink Socket.
systemd[1]: Starting udev Coldplug all Devices...
systemd[1]: Created slice system-getty.slice.
systemd[1]: Mounted Configuration File System.
systemd[1]: Mounted Debug File System.
systemd[1]: Mounted Temporary Directory.
systemd[1]: Started Apply Kernel Variables.
systemd[1]: Started Remount Root and Kernel File Systems.
systemd[1]: Starting Create System Users...
systemd[1]: Started udev Coldplug all Devices.
systemd[1]: Started Create System Users.
systemd[1]: Started Journal Service.
systemd-journald[316]: Received request to flush runtime journal from PID 1
aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: request() failed for pin 56
aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: pin-56 (1e780000.gpio:376) status -1
random: nonblocking pool is initialized
<hangs>
```
Looks like the pinctl issue.

shenki, 2016-09-14:
Is the output you provided with the fixes mentioned by @madscientist159?

You do have a pinctrl issue in that the pin you're requesting is not defined. I have opened an issue #575 to track this.

However, the hang you see is not a kernel failing to boot. As you can see, systemd is running and therefore you have booted to userspace.

I suspect your issue is that a getty has not spawned on the serial you're watching. I have opened an issue to track that failure #576.

Does your image request a DHCP lease? If so, are you able to SSH in?

madscientist159, 2016-09-14:
I just checked again, with the patches in my kernel tree OpenBMC definitely boots with both kernel messages and functional console over UART5.  In addition, the network is functional.

Firestone stubbornly refuses to IPL despite my best efforts however.

adamliyi, 2016-09-16:
@shenki, @madscientist159 , I tested on firestone using the V1.99.0 obmc, so there is no madscientist159's fix.  I can verify @madscientist159's fix if it has been merged to obmc tree.

madscientist159, 2016-09-16:
@adamliyi I'm in the process of submitting V2 of the patches to the mailing list.  It should be merged shortly, or you could just pull in my development branch for testing: https://github.com/madscientist159/linux-openbmc/commits/dev-4.7-firestone-fixes

shenki, 2016-10-24:
This has been resolved.

closed 2016-10-24 by shenki