openbmc/openbmc #571: [master] Phosphor Fan Control on Firestone fails to start (loops constantly)

madscientist159, 2016-09-13:
On Firestone with OpenBMC master, the "Phosphor Fan Control" service fails to start, continually looping with the following output in journalctl:

```
Sep 12 18:26:19 firestone systemd[1]: Starting Phosphor Fan Control...
Sep 12 18:26:22 firestone fan_control.py[812]: Initializing fan: /org/openbmc/sensors/speed/fan0
Sep 12 18:26:22 firestone fan_control.py[812]: Initializing fan: /org/openbmc/sensors/speed/fan1
Sep 12 18:26:22 firestone fan_control.py[812]: Initializing fan: /org/openbmc/sensors/speed/fan2
Sep 12 18:26:22 firestone fan_control.py[812]: Initializing fan: /org/openbmc/sensors/speed/fan3
Sep 12 18:26:22 firestone fan_control.py[812]: Initializing fan: /org/openbmc/sensors/speed/fan4
Sep 12 18:26:22 firestone fan_control.py[812]: Initializing fan: /org/openbmc/sensors/speed/fan5
Sep 12 18:26:22 firestone fan_control.py[812]: Starting fan control
Sep 12 18:26:22 firestone fan_control.py[812]: Setting fans to max
Sep 12 18:26:22 firestone fan_control.py[812]: Traceback (most recent call last):
Sep 12 18:26:22 firestone fan_control.py[812]:   File "/usr/sbin/fan_control.py", line 53, in <module>
Sep 12 18:26:22 firestone fan_control.py[812]:     #fan_control.setMax()
Sep 12 18:26:22 firestone fan_control.py[812]:   File "/usr/sbin/fan_control.py", line 43, in setMax
Sep 12 18:26:22 firestone fan_control.py[812]:     intf.setValue(dbus.UInt32(255))
Sep 12 18:26:22 firestone fan_control.py[812]:   File "/usr/lib/python2.7/site-packages/dbus/proxies.py", line 145, in __call__
Sep 12 18:26:22 firestone fan_control.py[812]:     **keywords)
Sep 12 18:26:22 firestone fan_control.py[812]:   File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 651, in call_blocking
Sep 12 18:26:22 firestone fan_control.py[812]:     message, timeout)
Sep 12 18:26:22 firestone fan_control.py[812]: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.UnknownMethod: Method "setValue" with signature "u" on interface "org.openbmc.SensorValue" doesn't exist
Sep 12 18:26:22 firestone systemd[1]: org.openbmc.control.Fans.service: Main process exited, code=exited, status=1/FAILURE
Sep 12 18:26:22 firestone systemd[1]: Failed to start Phosphor Fan Control.
Sep 12 18:26:22 firestone systemd[1]: org.openbmc.control.Fans.service: Unit entered failed state.
Sep 12 18:26:22 firestone systemd[1]: org.openbmc.control.Fans.service: Failed with result 'exit-code'.
Sep 12 18:26:22 firestone systemd[1]: org.openbmc.control.Fans.service: Service hold-off time over, scheduling restart.
```

madscientist159, 2016-09-20:
The existing pyfanctl application requires exactly six fans to be exposed to the system via hwmon.  If any of these six fans are not present the service will enter a startup loop.  Applying this patch resolves this bug:
https://gerrit.openbmc-project.xyz/#/c/555/

However, there is still no automatic fan control pending merge of https://gerrit.openbmc-project.xyz/#/c/120/

gtmills, 2018-05-31:
@madscientist159 This issue hasn't been updated in over a year. Still, a problem or can this be closed?

gtmills, 2018-06-01:
firestone was removed here: https://gerrit.openbmc-project.xyz/#/c/10957/

madscientist159, 2018-06-01:
Yeah, Firestone never worked right.

closed 2018-06-01 by madscientist159