openbmc/openbmc #572: dbus error against to  send ipmi command with ipmitool

williamli80, 2016-09-13:
on palm309(9.3.164.177),
root@palmetto:~#  /tmp/ipmitool -I dbus chassis
root@palmetto:~#  /tmp/ipmitool -I dbus chassis status
Then do "journalctl --no-page|grep chassis",  I got lots of error:

Sep 13 16:47:09 palmetto dbus[677]: [system] Rejected send message, 0 matched rules; type="error", sender=":1.44" (uid=0 pid=1016 comm="/tmp/ipmitool -I dbus chassis ") interface="(unset)" member="(unset)" error name="org.freedesktop.DBus.Error.UnknownMethod" requested_reply="0" destination=":1.43" (uid=0 pid=1010 comm="/usr/sbin/ipmid -d 0xff ")
Sep 13 16:47:09 palmetto dbus[677]: [system] Rejected send message, 0 matched rules; type="error", sender=":1.44" (uid=0 pid=1016 comm="/tmp/ipmitool -I dbus chassis ") interface="(unset)" member="(unset)" error name="org.freedesktop.DBus.Error.UnknownMethod" requested_reply="0" destination=":1.43" (uid=0 pid=1010 comm="/usr/sbin/ipmid -d 0xff ")
Sep 13 16:47:30 palmetto dbus[677]: [system] Rejected send message, 0 matched rules; type="error", sender=":1.45" (uid=0 pid=1019 comm="/tmp/ipmitool -I dbus chassis status ") interface="(unset)" member="(unset)" error name="org.freedesktop.DBus.Error.UnknownMethod" requested_reply="0" destination=":1.43" (uid=0 pid=1010 comm="/usr/sbin/ipmid -d 0xff ")
Sep 13 16:47:30 palmetto dbus[677]: [system] Rejected send message, 0 matched rules; type="error", sender=":1.45" (uid=0 pid=1019 comm="/tmp/ipmitool -I dbus chassis status ") interface="(unset)" member="(unset)" error name="org.freedesktop.DBus.Error.UnknownMethod" requested_reply="0" destination=":1.43" (uid=0 pid=1010 comm="/usr/sbin/ipmid -d 0xff ")
Sep 13 16:47:30 palmetto dbus[677]: [system] Rejected send message, 0 matched rules; type="error", sender=":1.45" (uid=0 pid=1019 comm="/tmp/ipmitool -I dbus chassis status ") interface="(unset)" member="(unset)" error name="org.freedesktop.DBus.Error.UnknownMethod" requested_reply="0" destination=":1.43" (uid=0 pid=1010 comm="/usr/sbin/ipmid -d 0xff ")

root@palmetto:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.0-163"
VERSION_ID="v1.0-163-g07d6a9f-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.0-163"
BUILD_ID="v1.0"

williamspatrick, 2016-09-13:
```
VERSION_ID="v1.0-163-g07d6a9f-dirty"

$ git rev-list origin/master ^07d6a9f | wc     74      74    3034
```
You are running code that is 74 commits back.  Can you reproduce this on a recent build?

williamli80, 2016-09-13:
shanghahi palmetto IPMItool command error log:

root@palmetto:~# /tmp/ipmitool -I dbus chassis
Chassis Commands:  status, power, identify, policy, restart_cause, poh, bootdev, bootparam, selftest
root@palmetto:~# /tmp/ipmitool -I dbus chassis status
Error sending Chassis Status command: Insufficient privilege level
root@palmetto:~# /tmp/ipmitool -I dbus sel
SEL Information
Version          : 1.5 (v1.5, v2 compliant)
Entries          : 0
Free Space       : 65535 bytes or more
Percent Used     : unknown
Last Add Time    : Not Available
Last Del Time    : Not Available
Overflow         : false
Supported Cmds   : 'Partial Add' 'Reserve'
root@palmetto:~# journalctl --no-page|grep ipmitool
Sep 13 22:53:36 palmetto dbus[679]: [system] Rejected send message, 0 matched rules; type="error", sender=":1.38" (uid=0 pid=812 comm="/tmp/ipmitool -I dbus chassis ") interface="(unset)" member=")
Sep 13 22:53:36 palmetto dbus[679]: [system] Rejected send message, 0 matched rules; type="error", sender=":1.38" (uid=0 pid=812 comm="/tmp/ipmitool -I dbus chassis ") interface="(unset)" member=")
Sep 13 22:53:41 palmetto dbus[679]: [system] Rejected send message, 0 matched rules; type="error", sender=":1.39" (uid=0 pid=813 comm="/tmp/ipmitool -I dbus chassis status ") interface="(unset)" m)
Sep 13 22:53:41 palmetto dbus[679]: [system] Rejected send message, 0 matched rules; type="error", sender=":1.39" (uid=0 pid=813 comm="/tmp/ipmitool -I dbus chassis status ") interface="(unset)" m)
Sep 13 22:53:41 palmetto dbus[679]: [system] Rejected send message, 0 matched rules; type="error", sender=":1.39" (uid=0 pid=813 comm="/tmp/ipmitool -I dbus chassis status ") interface="(unset)" m)
Sep 13 22:54:10 palmetto dbus[679]: [system] Rejected send message, 0 matched rules; type="error", sender=":1.40" (uid=0 pid=814 comm="/tmp/ipmitool -I dbus sel ") interface="(unset)" member="(uns)
Sep 13 22:54:10 palmetto dbus[679]: [system] Rejected send message, 0 matched rules; type="error", sender=":1.40" (uid=0 pid=814 comm="/tmp/ipmitool -I dbus sel ") interface="(unset)" member="(uns)
Sep 13 22:54:10 palmetto dbus[679]: [system] Rejected send message, 0 matched rules; type="error", sender=":1.40" (uid=0 pid=814 comm="/tmp/ipmitool -I dbus sel ") interface="(unset)" member="(uns)
root@palmetto:~#

root@palmetto:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-60"
VERSION_ID="v1.99.0-60-g2b717d8-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-60"
BUILD_ID="v1.99.0"
root@palmetto:~# random: nonblocking pool is initialized

williamspatrick, 2016-09-13:
@tomjoseph83 - Please look into this.  It seems like restricted mode is somehow becoming the default, or the ipmi code isn't quite working to read it.

williamli80, 2016-09-13:
william@opdt01-ThinkCentre-M58p:~$ curl -b cjar -k https://192.168.0.60/org/openbmc/settings/host0
{
  "data": {
    "boot_flags": "default",
    "boot_policy": "ONETIME",
    "network_config": "ipaddress=,prefix=,gateway=,mac=,addr_type=",
    "power_cap": 0,
    "power_policy": "RESTORE_LAST_STATE",
    "restricted_mode": 0,
    "system_state": "",
    "time_mode": "NTP",
    "time_owner": "BMC",
    "use_dhcp_ntp": "yes"
  },
  "message": "200 OK",
  "status": "ok"

It is not in restricted mode.

tomjoseph83, 2016-09-14:
@williamli80 observed with the latest patch that running Chassis status command is successful when the restricted_mode is turned off. Additionally Chassis status command is a candidate to be added to the Whitelisted commands so that  even in restricted mode it can be run.

williamli80, 2016-09-21:
@tomjoseph83 I miss the message.

tomjoseph83, 2016-12-12:
@williamli80 I think that this issue can be closed. Please update if the issue still persist.

geissonator, 2016-12-14:
See comment above from Tom

closed 2016-12-14 by geissonator