openbmc/openbmc #577: Issues when running "runqemu qemuarm"

ruchi393, 2016-09-14:
Hi,

I followed the following steps as mentioned in teh README

bitbake obmc-phosphor-image
runqemu qemuarm

Qemu console comes up, I reach upto qemuarm login.

However on further trying to enter any character on this console, I observe strange behavior. The keyboard input the console takes and prints doesn't match with the characters I enter. 

Am I following the right steps ?

Regards,
Ruchika

causten, 2016-09-14:
I just tried and I did not see anything wrong.  Here are my steps...

git clone ssh://openbmc.gerrit/openbmc/openbmc
cd openbmc/
. oe-init-build-env
bitbake obmc-phosphor-image
runqemu qemuarm

if that doesn't work for you does your output from launch qemu looking like...
/home/causten/openbmc/build/tmp/sysroots/x86_64-linux/usr/bin/qemu-system-arm -kernel /home/causten/openbmc/build/tmp/deploy/images/qemuarm/zImage-qemuarm.bin -net nic,model=virtio -net tap,vlan=0,ifname=tap0,script=no,downscript=no -M versatilepb -drive file=/home/causten/openbmc/build/tmp/deploy/images/qemuarm/obmc-phosphor-image-qemuarm-20160914180936.rootfs.ext4,if=virtio,format=raw -no-reboot -show-cursor -usb -usbdevice tablet -no-reboot -m 128 -serial mon:vc -serial null -append "root=/dev/vda rw console=ttyAMA0,115200 console=tty ip=192.168.7.2::192.168.7.1:255.255.255.0 mem=128M highres=off rootfstype=ext4 "

ruchi393, 2016-09-14:
My output from qemu is similar - 

/home/r66431/openbmc/build/build/tmp/sysroots/x86_64-linux/usr/bin/qemu-system-arm -kernel /home/r66431/openbmc/build/build/tmp/deploy/images/qemuarm/zImage-qemuarm.bin -net nic,model=virtio -net tap,vlan=0,ifname=tap0,script=no,downscript=no -M versatilepb -drive file=/home/r66431/openbmc/build/build/tmp/deploy/images/qemuarm/obmc-phosphor-image-qemuarm-20160914065004.rootfs.ext4,if=virtio,format=raw -no-reboot -show-cursor -usb -usbdevice tablet -no-reboot -m 128 -serial mon:vc -serial null -append "root=/dev/vda rw console=ttyAMA0,115200 console=tty ip=192.168.7.2::192.168.7.1:255.255.255.0 mem=128M highres=off rootfstype=ext4

After this I get the "qemu console", linux boots up and I reach 
The strange behavior which I observe is that "qemuarm login". However , when I try to login in the system, on typing characters I observe that input is not received correctly by the console. On typing "v" from keyboard,, console shows character "r" , "enter" key results in a space etc.

I also tried to execute the following:
curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST https://192.168.7.2/login -d "{\"data\": [ \"root\", \"0penBmc \" ] }"

This gives me error 
"curl: (7) Failed to connect to 192.168.7.2 port 443: No route to host"

Looks like the "ip address" assignment to the machine was not successful.

amboar, 2016-09-15:
@ruchi393 there may be something different with your environment, such as locale or keymap.

If you're using systemd on your host:

```
$ localectl status
   System Locale: LANG=en_AU.UTF-8
                  LANGUAGE=en_AU:en
       VC Keymap: n/a
      X11 Layout: us
       X11 Model: pc105
```
Otherwise `locale` might give us some hints:

```
$ locale
LANG=en_AU.UTF-8
LANGUAGE=en_AU:en
LC_CTYPE="en_AU.UTF-8"
LC_NUMERIC="en_AU.UTF-8"
LC_TIME="en_AU.UTF-8"
LC_COLLATE="en_AU.UTF-8"
LC_MONETARY="en_AU.UTF-8"
LC_MESSAGES="en_AU.UTF-8"
LC_PAPER="en_AU.UTF-8"
LC_NAME="en_AU.UTF-8"
LC_ADDRESS="en_AU.UTF-8"
LC_TELEPHONE="en_AU.UTF-8"
LC_MEASUREMENT="en_AU.UTF-8"
LC_IDENTIFICATION="en_AU.UTF-8"
LC_ALL=
```

ruchi393, 2016-09-15:
Given below is the output I get 

$ localectl status
   System Locale: LANG=en_IN
                  LANGUAGE=en_IN:en
       VC Keymap: n/a
      X11 Layout: us
       X11 Model: pc105

$ locale
LANG=en_IN
LANGUAGE=en_IN:en
LC_CTYPE="en_IN"
LC_NUMERIC="en_IN"
LC_TIME="en_IN"
LC_COLLATE="en_IN"
LC_MONETARY="en_IN"
LC_MESSAGES="en_IN"
LC_PAPER="en_IN"
LC_NAME="en_IN"
LC_ADDRESS="en_IN"
LC_TELEPHONE="en_IN"
LC_MEASUREMENT="en_IN"
LC_IDENTIFICATION="en_IN"
LC_ALL=

williamspatrick, 2016-09-16:
After I did some investigation for another problem, I suspect this is a duplicate of #584 .  What is happening is that two different processes are both claiming /dev/ttyS0 when we run qemu{arm,x86-64}.  I have a fix sequence I am testing out now.

williamspatrick, 2016-09-16:
https://gerrit.openbmc-project.xyz/#/c/537/  -- This code sequence may resolve your issues.

amboar, 2016-09-19:
> code sequence

"Down-level", "IPL" ... this all makes me think we need an OpenBMC jargon file.

ruchi393, 2016-09-19:
https://gerrit.openbmc-project.xyz/#/c/537/ doesn't seem to fix the issue I am facing..the key I type and the output which appears on console doesn't match

ruchi393, 2016-09-21:
Any more suggestions on how to try this out / fix of the issue

williamspatrick, 2016-09-21:
@ruchi393 - The only theory we have is related to keyboard bindings due to your locale.

```
$ localectl status
   System Locale: LANG=en_US.utf8
       VC Keymap: us
      X11 Layout: us
```

Can you try one of en_AU or en_US and see if that resolves anything?  If so it is likely an upstream QEMU issue.  If not, I don't have any further ideas and suggest we wait for change 537 to be merged and try again.

williamspatrick, 2016-10-05:
Is this resolved with the latest master?

closed 2017-05-31 by williamspatrick