openbmc/openbmc #578: [master] Firestone uses random MAC address for BMC Ethernet

madscientist159, 2016-09-15:
The Firestone BMC (re)boots every time with a different, random MAC address on the BMC Ethernet port.  Determine if there is nonvolatile storage for the MAC outside of the main Flash ROM and add the needed glue to set this MAC address in u-boot.

Alternatively, fix u-boot to preprogram the value stored in the "ethaddr" u-boot environment variable into the MAC before booting Linux.

williamspatrick, 2016-09-15:
The Barreleye system had a defined field from manufacturing that would have the MAC address in the backplane VPD.  That wasn't done for some of the other open-power servers, and we're still defining where to place it for Witherspoon.  We might be able to hack something together to save MAC addresses into an unused space in the planar VPD for existing systems.

williamspatrick, 2016-09-15:
@anoo1 - Assigning to you so you can comment in more detail.  You've kind of driven the MAC saving code for now.

madscientist159, 2016-09-15:
@williamspatrick What about simply using the u-boot environment storage for these machines?  Once u-boot programs the MAC into the hardware (on each boot), Linux doesn't change it afterward.

anoo1, 2016-09-15:
That's the current 'workaround' (to manually set the u-boot variable), but saving the MAC in vpd (like in barreleye) allows the u-boot variable to be automatically re-populated in case the BMC gets replaced or the u-boot variable is cleared (like with a complete flash update).

mdmillerii, 2016-09-15:
@madscientist159 Once the mac is stored in `ethaddr` it will be programmed by `u-boot` and picked up by the kernel.  The code added to barreleye was to read an external vpd and write it to the `u-boot` environment.

I don't know if the firestone vpd contains the MAC.

madscientist159, 2016-09-15:
@anoo1 On master at least this workaround is broken.  Setting the ethaddr variable requires that "dhcp" or similar be run on every startup before "boot", otherwise u-boot does not set up the MAC address in the hardware.

mdmillerii, 2016-09-15:
@madscientist159 that needs to be opened as a `u-boot` issue if it is skipping that part of the init.  Either it needs to write the register or update the device tree (which `u-boot` does not yet see).  That may involve releasing the reset.

I am guessing it was skipped to save the 2 seconds startup delay.  That is only needed as the timeout for NC-SI link providers to start responding to packets, the hardware is available after pll delay (10ms ?).

madscientist159, 2016-09-15:
@mdmillerii OK, I wasn't sure where to file this due to the split development structure.

I'll leave this particular report open to track the VPD persistent MAC storage feature.

mdmillerii, 2016-09-15:
@madscientist159 also I would expect any open of the ethernet to program the device even if it did not write a dhcp packet.

madscientist159, 2016-09-15:
@mdmillerii Agreed.  "dhcp" just happend to be the fastest thing to type to force network init.

u-boot issue has been filed.

kc8apf, 2016-09-15:
@maxims from Google is working on adding i2c drivers to u-boot so it can read MAC addresses from an EEPROM.

williamspatrick, 2016-09-15:
@kc8apf - "so it can read MAC addresses from an EEPROM" ... is that the reason why he is doing this or just a side-effect?  There isn't a common EEPROM format for storing the MAC address now, so we'd also need to add parsers into u-boot.  There is IPMI format and Power-VPD format that we'll need to support, at least, unfortunately.  We already have a userspace parser for the IPMI format and we plan to port some code from elsewhere for the Power-VPD format.  

Is it really advantageous to also have this support in u-boot?  It seems like the u-boot environment + userspace sync is more generic.  Typically the MAC would be programmed on the manufacturing line (and the userspace sync would even run then) so the only time when you have a random MAC is when someone does a raw rewrite of the NOR module, which also should not happen in production.

kc8apf, 2016-09-15:
The initial focus on porting i2c is because of wanting to read the MAC address from an EEPROM in u-boot.  Moving to device tree in u-boot and porting drivers from Linux to u-boot is something he'll be working on either way.  I'll need to check with some people about whether relying on u-boot environment + userspace sync works for us.

kc8apf, 2016-09-15:
Zaius will be storing the MAC address in an IPMI FRU.  There is a strong preference for us to have u-boot  honor an 'ethaddr' setting in the u-boot environment, fall back to the MAC programmed into the FRU if available, and use a random MAC as a last resort.

There is concern that we will see corrupted u-boot environments occasionally at our scale.  While we'll need to deal with that anyway, that causing random MACs to appear on the network will cause additional problems.

williamspatrick, 2016-09-16:
Agree on the u-boot honoring ethaddr setting.

What would be the mechanism to feed the chip location and parser type into u-boot?  I still wonder if there is something we can do to avoid the random MAC address and defer up to userspace in the case of ethaddr being missing.

kc8apf, 2016-09-16:
http://git.denx.de/?p=u-boot.git;a=blob;f=doc/README.enetaddr;h=50e4899787792246face19fcb4561afe0eafca1e;hb=HEAD

Sounds like it would be up to the board init function.  If u-boot is moving to a device tree as well, it might be possible to encode this in the dts somehow instead.

gtmills, 2018-06-01:
@madscientist159 @kc8apf Firestone was removed here: https://gerrit.openbmc-project.xyz/#/c/10957/. Can this be closed?

madscientist159, 2018-06-01:
Yeah, Firestone never worked right.

closed 2018-06-01 by madscientist159

jk-ozlabs, 2020-06-07:
> Zaius will be storing the MAC address in an IPMI FRU.

@kc8apf , did you end up implementing this? Can you share details about the FRU format you used to store the MAC address(es)?