openbmc/openbmc #579: Strap register needs to change for PNOR SPI FLASH controller

shenki, 2016-09-16:
On the ast2500 the PNOR SPI FLASH controller has changed a bit. One issue is that it is not strapped to be enabled as a SPI master by default, at least on the witherspoon and ast2500-evb.

Pinmux in the kernel treats the strap as read only. This means we have a few options to consider:

1. Write to the strapping register in u-boot
2. Write to the strapping register in Kernel early init, such as in the board file
3. Change the strapping of the processor (resistors? flash?)

This needs to be resolved before https://github.com/open-power/skiboot/issues/38 can be resolved.

mdmillerii, 2016-09-16:
or option 4.
4. configure the 2500 to look for a update table and write the straps in the u-boot image.

GHF, 2016-10-11:
@yuennancy Our Zaius EVT1 boards only _sort of_ have this issue (our resistor straps ended up as "10: Reserved, enable SPI Master and SPI Slave to AHB Bridge (debug mode)") which we're working around using option 2. EVT2 boards will have the correct straps (option 3) so no Zaius-specific software strap overriding should be necessary.

closed 2019-04-27 by stale[bot]

chnguyen-ampere, 2020-12-05:
Hi all,
current, I'm using ast2500-evb and it is not strapped to be enabled as a SPI master by default.

Anyone help me point the commit which fixes it on ast2500-evb? And what's best solution in that case 1 or 2 or 3 ?

Thank you !