openbmc/openbmc #581: [Barreleye/Palmetto]  Phosphor user add buggy

gkeishin, 2016-09-16:
The code to support useradd in REST obmc-phosphor-userd is buggy.

By default it should have picked up the default SHELL from /etc/default/useradd

root@barreleye:~#cat /etc/default/useradd
# useradd defaults file
GROUP=100
HOME=/home
INACTIVE=-1
EXPIRE=
SHELL=/bin/sh
SKEL=/etc/skel
#CREATE_MAIL_SPOOL=yes

But it doesnt and fills it with junk chars when a useradd comes in

**Example** 
xCuuRgkf:x:1039:1039:1cLU5gDG:/home/xCuuRgkf:eX

Where the correct should be something like this

WaEGbQEE:x:1059:1059:wxEu4Gko:/home/WaEGbQEE:/bin/sh

Looking at the code, 

        if groupname:
            cmd = "adduser "  + opts + " " + " -G " + groupname + " " + username
        else:
            cmd = "adduser "  + opts + " " + username

Either this has to be something like this

        if groupname:
            cmd = "adduser "  + opts + " " + " -G " + groupname + " " + "-s /bin/sh" + " "  + username
        else:
            cmd = "adduser "  + opts + " " + "-s /bin/sh" + " " + username

Or look at why it isn't really picking up the default shell...

gkeishin, 2016-09-16:
There is a problem as well with the password change. Will analyze this through what else changes needed to be done...

gkeishin, 2016-09-16:
There are couple of bugs due to pexpect
1. useradd - pexecpect garble up the shell and set what the prompt it user default is set with garbage
2. passwd - pexpect needs extra sendline for 'Re-enter new password: '

Code now working good to create and change password through REST interface.

gkeishin, 2016-09-16:
Patched the system by changing the /usr/sbin/obmc-phosphor-userd code

Ran the test, now looks working good.. will push the code for review

------------------------------------------------------------------------------
Test User :: This suite is for testing Open BMC user account manag... | PASS |
12 critical tests, 12 passed, 0 failed
12 tests total, 12 passed, 0 failed
==============================================================================

gkeishin, 2016-09-17:
This is a quick fix. I see there could be problem with the way how the existing code is in place.. however that I'll not be taking up or addressing for this minor fixes..

By default the shell should have been taken from default

root@barreleye:~# useradd -D
GROUP=100
HOME=/home
INACTIVE=-1
EXPIRE=
SHELL=/bin/sh
SKEL=/etc/skel
CREATE_MAIL_SPOOL=no
root@barreleye:~#

However the pexpect garble up and populate with junk char data resulting in user adding success but not valid to be use for accessing it. So I had to enforce the SHELL when adding the user with -s option.

Second fix was for changing password, there are 3 new line we need to key in from pexpect, added the missing third one and the behavior to key in to passwd is same as that of adduser for changing password.

anoo1, 2016-09-20:
Thanks George for fixing this up. Replacing the pexpect call is being tracked with https://github.com/openbmc/phosphor-networkd/issues/9 so as you mentioned there's no need to debug the pexpect 'buggyness' now that function works correctly.

gkeishin, 2016-09-20:
@anoo1 Sure np.. anything to help close down the current defect list.

closed 2016-09-20 by williamspatrick