openbmc/openbmc #583: phosphor-gevent.service fills journal with SSL messages.

williamspatrick, 2016-09-16:
If someone uses curl, with no options, to connect to the REST server we end up with content going into the journal.

On external system --- expected:
```
$ curl https://localhost:5443curl: (60) Peer's certificate issuer has been marked as not trusted by the user.
More details here: https://curl.haxx.se/docs/sslcerts.html

curl performs SSL certificate verification by default, using a "bundle"
 of Certificate Authority (CA) public keys (CA certs). If the default
 bundle file isn't adequate, you can specify an alternate file
 using the --cacert option.
If this HTTPS server uses a certificate signed by a CA represented in
 the bundle, the certificate verification probably failed due to a
 problem with the certificate (it might be expired, or the name might
 not match the domain name in the URL).
If you'd like to turn off curl's verification of the certificate, use
 the -k (or --insecure) option.
```

On BMC --- unexpected:
```
 phosphor-gevent.service - Phosphor Webserver
   Loaded: loaded (/lib/systemd/system/phosphor-gevent.service; static; vendor preset: enabled)
   Active: active (running) since Fri 2016-09-16 21:18:16 UTC; 4min 9s ago
 Main PID: 217 (python)
   CGroup: /system.slice/phosphor-gevent.service
           └─217 python /usr/sbin/phosphor-gevent rest_dbus

Sep 16 21:21:44 qemux86-64 phosphor-gevent[217]:   File "/usr/lib/python2.7/site-packages/gevent/server.py", line 102, in wrap_socket_and_handle
Sep 16 21:21:44 qemux86-64 phosphor-gevent[217]:     ssl_socket = self.wrap_socket(client_socket, **self.ssl_args)
Sep 16 21:21:44 qemux86-64 phosphor-gevent[217]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 402, in wrap_socket
Sep 16 21:21:44 qemux86-64 phosphor-gevent[217]:     ciphers=ciphers)
Sep 16 21:21:44 qemux86-64 phosphor-gevent[217]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 113, in __init__
Sep 16 21:21:44 qemux86-64 phosphor-gevent[217]:     self.do_handshake()
Sep 16 21:21:44 qemux86-64 phosphor-gevent[217]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 324, in do_handshake
Sep 16 21:21:44 qemux86-64 phosphor-gevent[217]:     return self._sslobj.do_handshake()
Sep 16 21:21:44 qemux86-64 phosphor-gevent[217]: SSLError: [SSL: TLSV1_ALERT_UNKNOWN_CA] unknown error (_ssl.c:590)
Sep 16 21:21:44 qemux86-64 phosphor-gevent[217]: <Greenlet at 0x7facfdeaa050: <bound method WSGIServer.wrap_socket_and_handle of <WSGIServer at 0x7facfe33fe10 fileno=5 address=('::%1100240', 443, 0, 1100240)>>(<socket at 0x7facfde9d9d0 fileno=9>, ('::200:0:0:0%1770665635', 52808, 0, 1770665635))> failed with SSLError
```

We should clean this up to prevent any kind of DoS by flooding the journal.

williamspatrick, 2016-09-16:
```
# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-104"
VERSION_ID="v1.99.0-104-g8dc90eb-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-104"
BUILD_ID="v1.99.0"
```

My development tree looks like this:
```
8dc90eb obmc-console: Use ttyVUART0 instead of ttyS0
1de9af1 meta-aspeed: Enable vuart rules for AST2[45]00
3852737 meta-aspeed: udev rules for vuart
21fb4de meta-aspeed: Enable common recipes
4d2c595 skiboot: Race condition in 'gard' build
77bf115 bottle: fix python-bottle-app package   <--- current master
```

This could easily be recreated on 77bf115.

JoshDKing, 2016-10-07:
- [ ] Task One * Set up structure and documentation for implementation in server.py and ssl.py
- [ ] Task Two * Implement the design in server.py
- [ ] Task Three * Implement the design in ssl.py

geissonator, 2016-11-23:
Code not merged, moving to next sprint.

closed 2016-12-05 by williamspatrick