openbmc/openbmc #585: obmc-console-ssh.socket restart policy incorrect?

williamspatrick, 2016-09-17:
This concern applies to all applications that open sockets...

If the obmc-console-ssh.socket application crashes for any reason, systemd will restart it immediately and it will then fail with a bind error for "address in use".  Is there something we need to do differently either for the application or for systemd unit file restart policies to ensure the address is freed up again for binding?

adamliyi, 2016-10-20:
I tried to restart obmc-console-ssh.socket by doing:
```
root@palmetto:~# systemctl stop obmc-console-ssh.socket
root@palmetto:~# systemctl start obmc-console-ssh.socket
Failed to start obmc-console-ssh.socket: Unit obmc-console.service not found. 
```
Changing obmc-console-ssh.socket to:

```
[Unit]
Description=Phosphor Host Console SSH Per-Connection socket
Conflicts=obmc-console-ssh.service
#Requires=obmc-console.service
Requires=obmc-console@ttyVUART0.service

[Socket]
ListenStream=2200
Accept=yes

[Install]
WantedBy=sockets.target
```
Can solve the above `Unit obmc-console.service not found. `error - is that an issue?

adamliyi, 2016-10-21:
Committed a patch here: https://gerrit.openbmc-project.xyz/#/c/874/1

adamliyi, 2016-10-25:
In current `obmc-console-ssh.socket` configure:
```
[Unit]
Description=Phosphor Host Console SSH Per-Connection socket
Conflicts=obmc-console-ssh.service
Requires=obmc-console.service
```
`obmc-console-ssh.socket` has **Requires** dependency on `obmc-console` service. As a result, killing `obmc-console` service will restarting `obmc-console-ssh.socket`. 
```
Oct 25 06:46:11 palmetto systemd[1]: obmc-console@ttyVUART0.service: Service hold-off time over, scheduling restart.
Oct 25 06:46:11 palmetto systemd[1]: Stopped Phosphor Console Muxer listening on device /dev/ttyVUART0.
Oct 25 06:46:11 palmetto systemd[1]: Started Phosphor Console Muxer listening on device /dev/ttyVUART0.
Oct 25 06:46:11 palmetto systemd[1]: Closed Phosphor Host Console SSH Per-Connection socket.
Oct 25 06:46:11 palmetto systemd[1]: Stopping Phosphor Host Console SSH Per-Connection socket.
Oct 25 06:46:11 palmetto systemd[1]: Listening on Phosphor Host Console SSH Per-Connection socket.
```
If we remove this dependency, how can systemd restarts a socket?
I searched here: https://www.freedesktop.org/software/systemd/man/systemd.socket.html
Looks systemd does not define restart policy for socket.

So after fix issue: https://github.com/openbmc/obmc-console/issues/7
Can we mark this issue#585 as invalid?

geissonator, 2016-11-10:
Gotta either move this to next sprint or close it out.  Need some feedback from @williamspatrick I guess?

williamspatrick, 2016-11-10:
I'm going to just move it.  It was an issue so it isn't affecting our burndown rate.

adamliyi, 2016-11-17:
@williamspatrick , @geissonator , can we close this issue. This issue is related with: openbmc/obmc-console#7, which has been close. And Andrew mentioned he did not observe this bug in auto test any more.

geissonator, 2016-11-28:
@adamliyi - This issue was specifically to track the fact that if an application is automatically restarted after it crashes, which uses the same socket address, it will hit a failure saying the address is still in use.  Has that problem been resolved with openbmc/obmc-console#7 ?

geissonator, 2016-12-14:
Moving again, no response from @adamliyi

adamliyi, 2016-12-23:
@geissonator, @williamspatrick , I committed a patch here: https://gerrit.openbmc-project.xyz/#/c/1587/ to add a 'ReusePort' option (https://www.freedesktop.org/software/systemd/man/systemd.socket.html#ReusePort=).

I tested with obmc-console-ssh.socket, this option can make a difference and fix "Address already in use" error.

But still, we need to avoid unnecessary dependency which unexpectedly triggering socket restarting, as in openbmc/obmc-console#7. As the systemd *socket* is designed to activate services "listening" on the socket, it should keep alive, instead of being restarted by other services unexpectedly.

To summarize - to fix "Address already in use" issue:
* Avoid unnecessary dependency from socket to other services
* Enable *ReusePort* option if this socket is expected to be restarted

bradbishop, 2017-01-03:
Is there any value in the ReusePort option?  I sort of agree with @adamliyi statement on nov 7 that this should be closed as resolved.

geissonator, 2017-01-06:
@adamliyi please bring up in design meeting so we can close out.

adamliyi, 2017-01-16:
@williamspatrick , I abandoned the patch https://gerrit.openbmc-project.xyz/#/c/1587/. Will you close this issue?

closed 2017-01-16 by williamspatrick