openbmc/openbmc #587: Build Error while userland do_fetch

Arjun-Padmanabhan, 2016-09-19:
I am compiling openbmc for rpi-test-image target. I am getting build error at userland do_fetch.
Console shows
"ERROR: Command Error: 'quilt --quiltrc /root/Desktop/openbmc/openbmc/rpi-build/tmp/sysroots/x86_64-linux/etc/quiltrc push' exited with 1"

Build Configuration:
BB_VERSION        = "1.30.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "universal"
TARGET_SYS        = "arm-openbmc-linux-gnueabi"
MACHINE           = "raspberrypi3"
DISTRO            = "openbmc-phosphor"
DISTRO_VERSION    = "0.1.0"
TUNE_FEATURES     = "arm armv7ve vfp thumb neon vfpv4 callconvention-hard cortexa7"
TARGET_FPU        = "hard"
meta
meta-poky
meta-oe
meta-multimedia
meta-networking
meta-python
meta-virtualization
meta-phosphor
meta-raspberrypi  = "master:5d5f8df3ecf64104ba4734d09659eae997a7a722"

Attaching log file with this issue. 
[log.do_patch.zip](https://github.com/openbmc/openbmc/files/479836/log.do_patch.zip)

Arjun-Padmanabhan, 2016-09-19:
Console log output:
"
ERROR: userland-git-r5 do_patch: Command Error: 'quilt --quiltrc /root/Desktop/openbmc/openbmc/rpi-build/tmp/sysroots/x86_64-linux/etc/quiltrc push' exited with 1  Output:
Applying patch 0002-wayland-Add-support-for-the-Wayland-winsys.patch
patching file .gitignore
patching file CMakeLists.txt
patching file README.md
patching file buildme
patching file interface/khronos/CMakeLists.txt
patching file interface/khronos/common/khrn_client.c
patching file interface/khronos/common/khrn_client.h
patching file interface/khronos/common/khrn_client_mangle.h
patching file interface/khronos/common/khrn_client_platform.h
patching file interface/khronos/common/khrn_client_unmangle.h
patching file interface/khronos/common/linux/khrn_client_platform_linux.c
patching file interface/khronos/common/linux/khrn_wayland.c
patching file interface/vmcs_host/vc_vchi_dispmanx.h
patching file interface/khronos/egl/egl_client.c
patching file interface/khronos/egl/egl_client_get_proc.c
patching file interface/khronos/egl/egl_client_surface.c
patching file interface/khronos/egl/egl_client_surface.h
patching file interface/khronos/egl/egl_int_impl.h
patching file interface/khronos/ext/egl_wayland.c
patching file interface/khronos/include/EGL/eglext.h
patching file interface/khronos/wayland-egl/wayland-egl-priv.h
patching file interface/khronos/wayland-egl/wayland-egl.c
patching file interface/khronos/wayland-egl/wayland-egl.pc.in
patching file interface/vmcs_host/CMakeLists.txt
patching file interface/vmcs_host/vc_dispmanx.h
patching file interface/vmcs_host/vc_vchi_dispmanx.c
patching file interface/vmcs_host/vc_vchi_dispmanx.h
Hunk #1 FAILED at 66.
1 out of 1 hunk FAILED -- rejects in file interface/vmcs_host/vc_vchi_dispmanx.h
patching file interface/wayland/dispmanx.xml
patching file makefiles/cmake/Wayland.cmake
Patch 0002-wayland-Add-support-for-the-Wayland-winsys.patch does not apply (enforce with -f)
ERROR: userland-git-r5 do_patch: Function failed: patch_do_patch
ERROR: Logfile of failure stored in: /root/Desktop/openbmc/openbmc/rpi-build/tmp/work/raspberrypi3-openbmc-linux-gnueabi/userland/git-r5/temp/log.do_patch.15133
ERROR: Task 1 (/root/Desktop/openbmc/openbmc/import-layers/meta-raspberrypi/recipes-graphics/userland/userland_git.bb, do_patch) failed with exit code '1'
NOTE: Tasks Summary: Attempted 10 tasks of which 9 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /root/Desktop/openbmc/openbmc/import-layers/meta-raspberrypi/recipes-graphics/userland/userland_git.bb, do_patch
Summary: There were 2 WARNING messages shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
"

mdmillerii, 2016-09-19:
This error is frequenly encontered when some pruning process deletes old files, resulting in the `bitbake` state cache not matching the file system and bitbake trying to apply patches that are already applied.  If your build tree is in `/tmp` or similar please try removing and recreating your build tree.

williamspatrick, 2016-09-19:
We don't entirely support RaspPi yet.  @adamliyi is working on openbmc/openbmc#399 , which is suppose to finish adding support.

The error you are encountering appears to be building some X or Wayland-based applications.  We don't support a GUI for sure, so there must be some MACHINE_FEATURE or DISTRO_FEATURE options we need to disable for RaspPi.

closed 2016-09-19 by williamspatrick