openbmc/openbmc #589: Differences between facebook/openbmc and openbmc/openbmc

ruchi393, 2016-09-20:
For openBMC, there seem to be two yocto based projects -
facebook/openbmc
openbmc/openbmc

How similar or how different are these ? Are there any plans to merge the two efforts ? Can the utilities from openbmc/openbmc be used with this openbmc project and vica versa ?

madscientist159, 2016-09-20:
facebook/openbmc is the original release.  openbmc/openbmc appears to be the actively developed fork / continuation of that original project.

amboar, 2016-09-21:
It's worth noting that the openbmc/openbmc fork has a development team that concentrates on enabling OpenBMC on OpenPOWER machines. However, as a consequence, the team is generally supporting hardware like the AST2400 and AST2500 SoCs, implementing and upstreaming firmware/kernel drivers and infrastructure. Also, part of openbmc/openbmc's role is to be a reference BMC distribution, so contributions for non-OpenPOWER systems are of course welcome!

williamspatrick, 2016-09-21:
@ruchi393 - 

At this point, facebook/openbmc and openbmc/openbmc are similar in name and concept only.

We [IBM] had been interested in an open-source BMC for the OpenPower community but nothing was available.  Facebook released their project and it gained some interest within a few members of the OpenPower community.  A little over a year ago, Rackspace committed to an open-source BMC project and collaborated with us to create it.  When we started, there was some design decisions Facebook had made that we did not feel was scalable for the scope of what we wanted an open source BMC to do, since their main focus was as an internal project rather than a potential commercial offering, so we decided to create our own proof-of-concept of an open-source BMC.  Hence, this project is the result of our collaboration with Rackspace (and others) to create that.

Right now, we are focused heavily on development for a Power9-based commercial offering that will utilize this code base.  We are also working closely with developers at Google for another Power9-based system that they and Rackspace will utilize.  But, our intention is not to have design aspects that are Power-unique; we fully intend that this code base will be used some day on "other architectures".

We have been working with Facebook off and on to attempt to arrive at some amount of commonality.  The first hurdle there will be to have a common Linux and u-boot, which they are working towards, since we have been very active at up-streaming drivers for the AST2400/2500 chips.  We hope that as we both get a more mature implementation of our visions we will be able to converge on the best of both implementations, but converging on the userspace applications is a little farther off.

To clarify what was said earlier, openbmc/openbmc is _not_ a fork of facebook/openbmc.  We are both working on the same concept, but with entirely different code right now other than that we are both utilizing Yocto Linux as the basis.

ruchi393, 2016-09-21:
Thanks for the details. Few follow on queries. Are there any plans to support 
1. KVM over IP
2. virtual storage
3. Web GUI based interface to the BMC

kc8apf, 2016-09-21:
1. A few people have asked about it but I haven't seen any concrete plans to implement it.  The AST2500 certainly has the right hardware to act as a video card and usb hid devices.  Device drivers need to be written for those and then a userspace application to provide access over IP.  Patches are welcome.
2. Similar story to 1
3. https://github.com/openbmc/openbmc/issues/266

mdmillerii, 2016-09-21:
2. #261

agangidi53, 2016-09-21:
@ruchi393  Thanks for starting this curious and helpful discussion. For future reference and questions, there's a openBMC mailing list and the IRC channel #openbmc

gtmills, 2018-05-21:
@ruchi393  Can this issue be closed?

amboar, 2019-04-02:
Not a bug, and we now also have pieces for each of the features in https://github.com/openbmc/openbmc/issues/589#issuecomment-248515288

closed 2019-04-02 by amboar