openbmc/openbmc #593: [Barreleye] REST service keeps as BUSY

johnhcwang, 2016-09-21:
Our test team runs the system reboot test or ac cycle test on Barreleye overnight, both of them can see this issue.
When the issue occurs, BMC can't accept curl command anymore.
It seems the rest service state in busy and the Recv_Q keep as full.
The issue will gone after restart the rest service within 1m30s.

Configuration:
obmc v1.0.5
BMC IP: 192.168.1.33
RMC IP: 192.168.1.120
````
root@barreleye:~# netstat -tn
Active Internet connections (w/o servers)
Proto Recv-Q Send-Q Local Address          Foreign Address          State
tcp      305      0 192.168.1.33:443       192.168.1.120:33406      CLOSE_WAIT
tcp        0      0 192.168.1.33:443       192.168.1.120:38409      SYN_RECV
tcp        0      0 192.168.1.33:443       192.168.1.120:40990      ESTABLISHED
tcp      305      0 192.168.1.33:443       192.168.1.120:33416      CLOSE_WAIT
````

Hi @anoo1 , do you think that we meet the SYN flood issue?
Sometimes I get this message from journal log.
````
barreleye kernel: TCP: request_sock_TCP:  Possible SYN flooding on port 443. Dropping request.  Check SNMP counters.
````

anoo1, 2016-09-26:
Hi @johnhcwang , yeah the journal log is the same as we saw in phosphor-rest-server/issues#24 where we added a workaround to restart the rest server when the network config changes.

Could you describe the scenario when this issue is seen. For example, do you see the curl command failing after the power cycle or after a few commands are issued, and are single or parallel curl commands issued, etc.
If you have a journal log of a BMC that failed, it'd be helpful to attach it to the issue as well.

Looping in @williamspatrick and @bradbishop .

johnhcwang, 2016-09-28:
One curl command was issued from remote test tool and parallel rest api command were issued from RMC (5 parallel sessions in maximum).
Remote test tool successive PING the OS static IP to make sure the system has booted into the OS and then login BMC through curl.
RMC successive send the REST command and once it get the response from BMC, it will create 5 parallel sessions to retrieve the information from BMC through REST API.
Above are continuing actions during the AC cycle test.

[journal log](https://github.com/openbmc/openbmc/files/496933/rest_issue.txt)

dugroc, 2016-10-03:
Can you tell me what the value of tcp_max_syn_backlog is ? 
4 or 5 connections should not be triggering Possible SYN flooding on port 443....

dugroc, 2016-10-04:
If one beats the call to accept in the rest server, it could cause this. Since it works most of the time, it is likely a timing issue. If you put a couple of seconds delay from when the ping works, to when you send the curl requests, this might fix this.

johnhcwang, 2016-10-04:
RMC starts the connection to send the curl requests once it receives the inarp response from BMC, not depend on the ping response.
Can we still treat it as timing issue in this case?
If yes, is it possible that we can postpone the inarp service until BMC is ready to fix this issue?

gwshan, 2016-10-04:
It's possible that the backlog is specified to 5 on TCP server side when calling listen(). It's worthy to check the parameters passed to listen() on TCP server (BMC) side.

static inline int inet_csk_reqsk_queue_is_full(const struct sock *sk)
{
        return inet_csk_reqsk_queue_len(sk) >= sk->sk_max_ack_backlog;
}

int inet_listen(struct socket *sock, int backlog)
{
       ...
       sk->sk_max_ack_backlog = backlog;
       ...
}

gwshan, 2016-10-06:
Had a discussion with @johnhcwang and we found, the Rocket's maximum allowed TCP connection by default is 5 as below:

rocket/__init__.py:DEFAULT_LISTEN_QUEUE_SIZE = 5

johnhcwang, 2016-10-06:
REST server uses Rocket to handle the connections but didn't specify the queue_size.
By Rocket's definition, the operating system’s maximum or 5 passed to the listen() function.
''''
queue_size - An integer number of connections allowed to be queued before Rocket accepts them. This number is passed to the listen() function in the operating system’s socket library. It defaults to None which either uses the operating system’s maximum or 5 if the OS max is not discoverable.
''''

johnhcwang, 2017-02-16:
Hi @anoo1 , could you help on this issue? We need someone who familiar with REST to increase the receive_queue_size by @gwshan 's comment and see if it can be fixed.

anoo1, 2017-03-09:
I'd say let's have @bradbishop review the proposal about increasing the rocket queue size.

bradbishop, 2017-03-14:
Just FYI, we aren't using Rocket anymore in master.

If bumping this queue size fixes the problem then I don't have an issue with that being done.

bradbishop, 2017-03-14:
@johnhcwang Are you able to try this out to see if it fixes the problem?

gtmills, 2018-06-02:
@johnhcwang No update in a year. Can this be closed or still a problem? Thanks!

closed 2018-06-04 by anoo1