openbmc/openbmc #594: Failed to fetch BMC version info on palmetto through REST API

SrideviRamesh, 2016-09-21:
Issue:
======
Unable to fetch BMC version info on palmetto through REST API

Log Info:
===========

-bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST https://9.3.164.177/login -d "{\"data\": [ \"root\", \"0penBmc\" ] }"
{
  "data": "User 'root' logged in",
  "message": "200 OK",
  "status": "ok"
}

-bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X GET https://9.3.164.177//org/openbmc/inventory/system/bios
{
  "data": {
    "description": "org.openbmc.ObjectMapper.Error.NotFound: path or object not found: /org/openbmc/inventory/system/bios"
  },
  "message": "404 Not Found",
  "status": "error"
}

System Info:
---------------

root@palmetto:~# cat /etc/version
20160920214528
root@palmetto:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-114"
VERSION_ID="v1.99.0-114-ged52dc1-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-114"
BUILD_ID="v1.99.0"
root@palmetto:~#

please find the FFDC logs in the attachment

[20160921052752506226_ffdc_report.txt](https://github.com/openbmc/openbmc/files/484853/20160921052752506226_ffdc_report.txt)
[20160921052752506226_BMC_proc_list.txt](https://github.com/openbmc/openbmc/files/484855/20160921052752506226_BMC_proc_list.txt)
[20160921052752506226_BMC_journalctl.txt](https://github.com/openbmc/openbmc/files/484856/20160921052752506226_BMC_journalctl.txt)

williamspatrick, 2016-09-21:
The title of this issue says "BMC version" but the REST URL is "system/bios", which is the host firmware.  What are you trying to test?  If you are trying to get the BMC version, the URL is wrong.  If you are trying to get the Host version, have you booted the host yet and is IPMI functional?

sivassrr, 2016-09-21:
@SrideviRamesh Please change it to BIOS. It is BIOS version test which you had implemented. Not BMC. @williamspatrick we tried @ BMC ready state as well we tried to read BIOS version, We could able to get it. (This we did when we introduce this test).

SrideviRamesh, 2016-09-21:
Hi,
I did a mistake. Issue is to fetch BIOS version info on palmetto.

I have tested the BIOS version check when BMC HOST is at BMC_READY and HOST_BOOTED state. I could not fetch BIOS info

williamspatrick, 2016-09-21:
@SrideviRamesh , @sivassrr - Why you add log items to github can you please escape it with 3 backticks?  That way github doesn't try to format them.

williamspatrick, 2016-09-21:
@spinler - This works on Barreleye but does not work on Palmetto.  I suspect there is a missing inventory object for the host firmware.

spinler, 2016-09-21:
Nobody ever put
` "**'PRODUCT_0'  : '<inventory_root>/system/bios', `
into Palmetto.py.  Same with:
`    'BOARD_3'    : '<inventory_root>/system/misc',`

Honestly, I don't think either of those belong in the MRW, especially not misc.  I'll leave misc for another day, but will think about how to get this into the inventory without an MRW author having to know they need to add a 'bios' unit to their system purely to hold version information.

williamspatrick, 2016-09-22:
@spinler - understand. They must have tested this on Barreleye only before. We should discuss tomorrow how / where to represent BMC-owned virtual inventory and sensor objects. I agree that the MRW owner should not need to be concerned with these.

spinler, 2016-09-22:
We discussed this, and I'm going to add the bios & misc objects to the inventory.  Eventually this will probably be redesigned and they'll be moved somewhere else.

closed 2016-09-23 by williamspatrick

sivassrr, 2016-09-26:
@SrideviRamesh Please verify when you get chance. Or work with @rahulmah on the same.