openbmc/openbmc #596: Create location to store available / installed host firmware images.

williamspatrick, 2016-09-22:
There is currently a "virtual" inventory object `.../inventory/system/bios`, which is populated by an IPMI message from the host firmware.  We want to eliminate all "virtual" inventory objects and ensure they all represent physical information.  Need to find an alternative way to store and report the firmware images on the Host.

williamspatrick, 2017-05-31:
The host firmware inventory is now handled by the /software namespace and managed directly on the BMC.  We still need to add handling for the set/get sensor calls on that sensor.

closed 2017-06-02 by rfrandse

anoo1, 2017-06-02:
Per @dkodihal , this sensor is being acknowledged by the ipmi code, the data discarded, and success returned to the host.  No more action is needed since the bios version is stored in the Software interface.