openbmc/openbmc #597: Create location to store "misc" VPD data.

williamspatrick, 2016-09-22:
There is currently a "virtual" inventory object `.../inventory/system/misc` that we believe is populated from certain EEPROM reads.  We want to eliminate all "virtual" inventory objects and instead ensure they all represent physical devices.  So, we need to move this data either into a proper inventory object or find another way to represent the data contained within.

rfrandse, 2017-05-31:
Part of IPMI work?

dkodihal, 2017-06-01:
This is not IPMI. It could mean inventory work, but I need to check if this is already covered by our existing inventory.

dkodihal, 2017-08-07:
What comprises this 'misc' inventory object are various firmware revisions, which are now already under the software namespace.

closed 2017-08-07 by dkodihal