openbmc/openbmc #599: [Palmetto VERSION_ID="v1.99.0-123-gfc7788a-dirty" ] Internal Server Error for GET operation on cpu inventory url

rahulmah, 2016-09-23:
In Palmetto, Internal Server Error is thrown out while doing GET operation on cpu inventory url. 

Get operation is done on /org/openbmc/inventory/system/chassis/motherboard/cpu/attr/present

URI:https://xx.xx.xx.xx/org/openbmc/inventory/system/chassis/motherboard/cpu/attr/present, method:Get, args:{}
Response code:500, Content:{
  "data": {
    "description": "Internal Server Error",
    "exception": "DBusException('Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.',)",
    "traceback": [
      "Traceback (most recent call last):",
      "  File \"/usr/lib/python2.7/site-packages/bottle.py\", line 858, in _handle",
      "    route, args = self.router.match(environ)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 740, in custom_router_match",
      "    route.callback._setup(**args)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 80, in _setup",
      "    return self.setup(**kw)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 265, in setup",
      "    request.route_data['obj'] = self.find(path, prop)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 252, in find",
      "    self.app.instance_handler.setup(path)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 360, in setup",
      "    request.route_data['map'] = self.find(path, callback)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 351, in find",
      "    path=path)}",
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 97, in try_mapper_call",
      "    return f(**kw)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py\", line 37, in get_object",
      "    return self.iface.GetObject(path)",
      "  File \"/usr/lib/python2.7/site-packages/dbus/proxies.py\", line 145, in __call__",
      "    **keywords)",
      "  File \"/usr/lib/python2.7/site-packages/dbus/connection.py\", line 651, in call_blocking",
      "    message, timeout)",
      "DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken."
    ]
  },
  "message": "500 Internal Server Error",
  "status": "error"

Above issue is constantly seen with automation with recent drivers. But using curl we are not seeing problem while doing the same. 

bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST https://xx.xx.xx.xx/login -d "{\"data\": [ \"root\", \"0penBmc\" ] }"
{
  "data": "User 'root' logged in",
  "message": "200 OK",
  "status": "ok"
}bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X GET -d "{\"data\":[]}" https://xx.xx.xx.xx/org/openbmc/inventory/system/chassis/mothrboard/cpu/attr/present
{
  "data": "True",
  "message": "200 OK",
  "status": "ok"
}bash-4.1$

root@palmetto:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-123"
VERSION_ID="v1.99.0-123-gfc7788a-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-123"
BUILD_ID="v1.99.0"

root@palmetto:~# cat /etc/version
20160921185744

**FFDC attached**
[20160923062813565118_BMC_journalctl_log.txt](https://github.com/openbmc/openbmc/files/489562/20160923062813565118_BMC_journalctl_log.txt)

rahulmah, 2016-09-24:
Figured out the problem with our automation. We need to sync Plametto.py file with openbmc repo

closed 2016-09-24 by rahulmah

williamspatrick, 2016-09-24:
Keep in mind that those files are all going away. @spinler is moving things over to be generated from the MRW directly.