openbmc/openbmc #600: journal flooded with cannot read attributes in2_input

geissonator, 2016-09-23:
I've noticed in recent master drivers this one popping up.  It comes every 5s so must be the hwmon polling code defined in skeleton/configs/Barreleye.py

Not quite sure why this shows 10s poll_interval but we seem to get it every 5s.

`'in2_input' : { 'object_path' : 'voltage/P0V9_CPU0_BUF1','poll_interval' : 10000,'scale' : 1,'units' : '' },`

Log Output:

> Sep 20 13:32:50 barreleye hwmon.py[735]: Cannot read attributes: /sys/class/hwmon/hwmon4/in2_input
> Sep 20 13:32:51 barreleye hwmon.py[735]: Cannot read attributes: /sys/class/hwmon/hwmon5/in2_input
> Sep 20 13:32:55 barreleye hwmon.py[735]: Cannot read attributes: /sys/class/hwmon/hwmon4/in2_input
> Sep 20 13:32:56 barreleye hwmon.py[735]: Cannot read attributes: /sys/class/hwmon/hwmon5/in2_input
> Sep 20 13:33:00 barreleye hwmon.py[735]: Cannot read attributes: /sys/class/hwmon/hwmon4/in2_input
> Sep 20 13:33:01 barreleye hwmon.py[735]: Cannot read attributes: /sys/class/hwmon/hwmon5/in2_input
> Sep 20 13:33:05 barreleye hwmon.py[735]: Cannot read attributes: /sys/class/hwmon/hwmon4/in2_input
> Sep 20 13:33:06 barreleye hwmon.py[735]: Cannot read attributes: /sys/class/hwmon/hwmon5/in2_input
> Sep 20 13:33:10 barreleye hwmon.py[735]: Cannot read attributes: /sys/class/hwmon/hwmon4/in2_input
> Sep 20 13:33:11 barreleye hwmon.py[735]: Cannot read attributes: /sys/class/hwmon/hwmon5/in2_input

mine260309, 2016-10-12:
May I ask if this issue is for
  a) the journal flood of `Cannot read attributes: /sys/class/hwmon/hwmon4/in2_input`
or
  b) the interval of the log is not expected (10s), it's 5s instead?

williamspatrick, 2016-10-12:
The issue is for the flood of journal messages.

mine260309, 2016-10-14:
Some information:
* `hwmon4` points to `i2c-4/4-0010/hwmon/hwmon4`
* `hwmon5` points to `i2c-5/5-0010/hwmon/hwmon5`

And from `Barreleye.py`:
```python
	'4-0010' :  {
		'names' : {
			'curr1_input' : { 'object_path' : 'HSCA/Iout','poll_interval' : 5000,'adjust' : 0.25,'scale' : -3,'units' : 'A' },
			'in2_input' : { 'object_path' : 'HSCA/Vout','poll_interval' : 5000,'scale' : -3,'units' : 'V' },
		}
	},
	'5-0010' :  {
		'names' : {
			'curr1_input' : { 'object_path' : 'HSCB/Iout','poll_interval' : 5000,'adjust' : 0.25,'scale' : -3,'units' : 'A' },
			'in2_input' : { 'object_path' : 'HSCB/Vout','poll_interval' : 5000,'scale' : -3,'units' : 'V' },
		}
	},
```

Check what `hwmon4` has:
```bash
-rw-r--r--    1 root     root          4096 Sep 20 21:26 curr1_highest
-r--r--r--    1 root     root          4096 Sep 20 21:25 curr1_input
-r--r--r--    1 root     root          4096 Sep 20 21:26 curr1_label
...
-rw-r--r--    1 root     root          4096 Sep 20 21:26 in1_highest
-r--r--r--    1 root     root          4096 Sep 20 21:26 in1_input
-r--r--r--    1 root     root          4096 Sep 20 21:26 in1_label
...
-r--r--r--    1 root     root          4096 Sep 20 21:26 power1_alarm
-r--r--r--    1 root     root          4096 Sep 20 21:26 power1_input
-rw-r--r--    1 root     root          4096 Sep 20 21:26 power1_input_highest
...
```
So,
1. The interval is 5s, which is OK and consistent with the log;
2. The `in2_input` does not exist in `hwmon4` or `hwmon5`, that's why the logs shows `Cannot read attributes: /sys/class/hwmon/hwmon4/in2_input`.

adamliyi, 2016-10-14:
@shenki , looks the dev-4.7 kernel does not include my adm1278 patch
https://github.com/openbmc/linux/pull/80/commits/e337f8eef44c37d0611ff63763d7e3ff7511e9dc
 to fix issue: https://github.com/openbmc/linux/issues/71.

Will you apply that patch, or I can submit a new patch for dev-4.7.

shenki, 2016-10-14:
@adamliyi thanks for catching this. I have added the patch to dev-4.7.

We should submit it upstream too

shenki, 2016-10-14:
https://lkml.org/lkml/2016/10/14/132

mine260309, 2016-10-17:
As @adamliyi 's suggest, I tried [v2-0001-hwmon-adm1275-Enable-adm1278-VOUT-sampling.patch](https://ozlabs.org/~joel/v2-0001-hwmon-adm1275-Enable-adm1278-VOUT-sampling.patch), and it works fine: `in2_input` becomes available for both `hwmon4` and `hwmon5`.

@shenki Would you help to apply the patch to linux-obmc and close this issue?

shenki, 2016-10-17:
This patch has been submitted upstream. I will wait for upstream review before backporting it to our repository; I expect this to happen in the next day or so.

williamspatrick, 2016-10-17:
@shenki - Please add a commit to openbmc with a kernel update when you have it merged so this closes out.

shenki, 2016-10-18:
https://gerrit.openbmc-project.xyz/#/c/819/

closed 2016-10-21 by williamspatrick