openbmc/openbmc #601: Withersoon and ast2500evb failing to build

shenki, 2016-09-28:
https://openpower.xyz/job/openbmc-build/266/

```
01:34:56 Parsing of 1920 .bb files complete (0 cached, 1920 parsed). 2483 targets, 373 skipped, 0 masked, 0 errors.
01:34:57 ERROR: No recipes available for:
01:34:57   /var/lib/jenkins/workspace/openbmc-build/distro/ubuntu/target/witherspoon/openbmc/meta-openbmc-machines/meta-openpower/meta-ibm/meta-witherspoon/recipes-phosphor/skeleton/skeleton.bbappend
```

GHF, 2016-09-28:
Wasn't this fixed by https://gerrit.openbmc-project.xyz/#/c/620/ ?

shenki, 2016-09-28:
Lets see if Patrick can merge it for us

williamspatrick, 2016-09-30:
Merged a few days ago.  Resolved?

shenki, 2016-10-04:
No, as you can see here Witherspoon is still failing:

https://openpower.xyz/job/openbmc-build/

```
03:10:54 ERROR: No recipes available for:
03:10:54   /var/lib/jenkins/workspace/openbmc-build/distro/ubuntu/target/witherspoon/openbmc/meta-openbmc-machines/meta-openpower/meta-ibm/meta-witherspoon/recipes-phosphor/skeleton/skeleton.bbappend
```

GHF, 2016-10-05:
to be clear, the error message here differs from #636 because Witherspoon patches skeleton rather than obmc-op-power-control, but it's basically the same failure.

closed 2016-10-05 by williamspatrick