openbmc/openbmc #602: [Barreleye VERSION_ID="v1.99.0-131-ga81f31c-dirty"] Unknown Gard records on OS are created repeatedly even after clearing gards and boot to OS

sivassrr, 2016-09-28:
@SrideviRamesh commented on [Wed Sep 28 2016](https://github.com/openbmc/openbmc-test-automation/issues/135)

Explanation of the issue:

Unknown Gard records on OS are created repeatedly even after clearing gards and boot to OS
Expected Behavior:

Gard records should not be created after clearing gard records on HOST OS and reboot to OS
Actual Behavior:

Unknown Gard records on OS are created repeatedly even after clearing gards and boot to OS

Recreations Steps:

1. When BMC is in HOST_BOOTED state. clear gard records and reboot the OS
2. when the HOST OS is running . Then check gard records
3. Make sure no gard records are created

Log Info:

1. Gard records initially seen and clearing gard records at BMC runtime state

ibm@beye5:~$  sudo /home/ibm/skiboot/external/gard/gard list
[sudo] password for ibm:
|    ID    |   Error  | Type            |
+---------------------------------------+
| 00000001 | 9000029b | physical        |
| 00000002 | 9000029b | physical        |
+=======================================+

ibm@beye5:~$ sudo /home/ibm/skiboot/external/gard/gard clear all
Clearing the entire gard partition...done

2. Gard records observed when HOST OS is booted again

ibm@beye5:~$ sudo /home/ibm/skiboot/external/gard/gard list
[sudo] password for ibm:
|    ID    |   Error  | Type            |
+---------------------------------------+
| 00000001 | 900003a5 | physical        |
| 00000002 | 900003a5 | physical        |
+=======================================+
ibm@beye5:~$

Specifications:

root@barreleye:~# cat /etc/version
20160926091605
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-131"
VERSION_ID="v1.99.0-131-ga81f31c-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-131"
BUILD_ID="v1.99.0

FFDC attached
[OBM_HOST_GARD_ISSUE.zip](https://github.com/openbmc/openbmc-test-automation/files/497206/OBM_HOST_GARD_ISSUE.zip)
[barraleye_client_console_logs.txt](https://github.com/openbmc/openbmc-test-automation/files/497208/barraleye_client_console_logs.txt)

sivassrr, 2016-09-28:
Please take a look and it may have to go to host firmware.

williamspatrick, 2016-09-30:
The BMC does not create gard records.  

I see this in your console log:
```
> .89047|=========================================== ModuleId   0x0b unknown
>  28.89048|  ReasonCode 0xe504 unknown
>  28.89048|  UserDCPU ID: 16
>  28.89051|  Called out entity:
>  28.89052|User Data Section 2, type UD
>  28Data Section 3, type UD
>  28.89054|  Subsection type 0x33
>  28.89054|  ComponentId unknion 5, type UD
>  28.89057|  Subsection type 0x15
>  28.89057|  ComponentId hb-trace (0x3100)
>  28.89057|Usd errl (0x0100)
>  28.89059|  STRING
>  28.89060|  Hostboot Build ID: hostboot-c223637-1d3d 31.86411|ISTEP 12. 1
```

This is garbled, but it is an error message from the diagnostic software running on the Host (PRD) that is detecting an error reported by the hardware itself.  I cannot tell if it is P8 or Centaur.  You'll need to get a non-garbled recreate of that trace message and have someone from the PRD team at IBM help you identify the problem.  @zane131 would be a good contact.

closed 2016-09-30 by williamspatrick