openbmc/openbmc #603: [Barreleye VERSION_ID="v1.99.0-131-ga81f31c-dirty"] BMC did not reset when HOST OS is hung after performing checkstop sensor operation on HOST OS

sivassrr, 2016-09-28:
@SrideviRamesh commented on [Wed Sep 28 2016](https://github.com/openbmc/openbmc-test-automation/issues/133)

Explanation of the issue:
BMC did not reset when HOST OS is hung after performing checkstop sensor operation on HOST OS

Expected Behavior:

BMC resets the system if the OS is hung.

root@barreleye:~#  ps | grep host
  687 root       0:01 /usr/sbin/host_checkstop.exe
  691 root       0:05 /usr/sbin/host_watchdog.exe
  787 root       0:04 /usr/sbin/control_host.exe
 1781 root       0:02 /usr/sbin/dropbear -i -r /etc/dropbear/dropbear_rsa_host_key -B
 2037 root       0:04 /usr/sbin/dropbear -i -r /etc/dropbear/dropbear_rsa_host_key -B
 2045 root       0:04 /usr/sbin/dropbear -i -r /etc/dropbear/dropbear_rsa_host_key -B
 2159 root       0:00 grep host

Actual Behavior:

1. BMC HOST OS did not ping even after 5 minutes

2. To Bring HOST OS up i tried to power off and on the system through BMC openbmutil commands

   and then the HOST OS started running

Recreations Steps:

1. At BMC HOST_BOOTED state. Inject an unrecoverable error to the processor core from the HOST OS

sudo /home/ibm/skiboot/external/xscom-utils/putscom -c 0x0 0x1d01080D 0x0A00000000000000

2. HOS OS goes down

3. Through BMC obmc-console-client observe that BMC resets and HOST OS comes to operating

Specifications:

root@barreleye:~# cat /etc/version
20160926091605
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-131"
VERSION_ID="v1.99.0-131-ga81f31c-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-131"
BUILD_ID="v1.99.0"

FFDC attached
[client_console_logs.txt](https://github.com/openbmc/openbmc-test-automation/files/496391/client_console_logs.txt)

sivassrr, 2016-09-28:
@SrideviRamesh Issue to be open in openbmc repository. Only automation related issues/enhancement need to be opened in Openbmc-automation repository.

gkeishin, 2016-09-28:
@anoo1 Anything on this issue we observed here..

williamspatrick, 2016-09-30:
I don't understand this issue report.  You keep using the word "BMC reset" but I think you mean that the BMC is not rebooting the host?  The BMC itself should not reset for this.

What does "Through BMC obmc-console-client observe that BMC resets and HOST OS comes to operating" mean?  The real problem here is that the BMC does not identify that the host has checkstopped and we do not power off / reboot the host?  This sounds like a simple GPIO problem if that is what you are describing.  Please confirm.

SrideviRamesh, 2016-09-30:
yes as per @anoo1 BMC should go for reboot when the checkstop is issued

williamspatrick, 2016-09-30:
Please be more precise with your words. The BMC does not reboot. The BMC issues a reboot of the host.

anoo1, 2016-09-30:
@SrideviRamesh It's my understanding from your offline message that this test case rebooted the host as expected, and that you were just concerned about the gard record left on the system, (confirmed by the attached host console log that only conttains the output of gard records). You can clear the gard record with the gard command.

SrideviRamesh, 2016-10-02:
@anoo1 No this testcase did not reboot the HOST after performing SCOM operation on HOST OS. HOST OS was not pingable. In order to bring the HOST OS up i tried to power off & on using obmutil commands. 
I am not worried about the gard records. gard record has to be present after the SCOM operation on HOST OS

SrideviRamesh, 2016-10-02:
@anoo1 I would like to know the valid reason for closing this issue

anoo1, 2016-10-04:
Please provide the bmc journal log and host console from the time the putscom command is executed.

SrideviRamesh, 2016-10-05:
please find FFDC  and host console logs in the attachments
[OBMC_OS_HUNG.zip](https://github.com/openbmc/openbmc/files/510358/OBMC_OS_HUNG.zip)
[host_console_logs.txt](https://github.com/openbmc/openbmc/files/510364/host_console_logs.txt)

anoo1, 2016-10-05:
Thanks, but the host console log only shows that a "gard list" command was issued. Please collect the logs from when the putscom command to checkstop the host is issued.

sivassrr, 2016-11-08:
@SrideviRamesh working on the same to have OS install and will retry this and provide the data.

geissonator, 2016-12-14:
Not much activity here, is this still an issue?  Going to move to backlog for now.

SrideviRamesh, 2017-01-09:
i have collected journalctl & dmesg logs after injecting checkstop on HOST OS.
please find it attachments
```
root@barreleye:~# ps -aef | grep  check
  699 root       0:00 /usr/sbin/host_checkstop.exe
 1027 root       0:00 grep check
root@barreleye:~#
'''
[dmesg.txt](https://github.com/openbmc/openbmc/files/693628/dmesg.txt)
[journal.txt](https://github.com/openbmc/openbmc/files/693629/journal.txt)

SrideviRamesh, 2017-01-19:
HI @anoo1 can u please give updates regarding this issue... let me know if u require any further info

spinler, 2017-01-23:
Barreleye has: ```GPIO_CONFIG['CHECKSTOP']   	  =   { 'gpio_pin': 'P5', 'direction': 'falling' }```
P5 is GPIO 125, which would show up as /sys/class/gpio/gpio445, but I don't see it:

```root@barreleye:/sys/class/gpio# ls
export       gpio325      gpio334      gpio344      gpio353      gpio395      gpiochip320
gpio324      gpio326      gpio343      gpio352      gpio354      gpio455      unexport
```

spinler, 2017-01-23:
OK, the problem is this call from gpio.c: gpio_init() is failing:

```
	result = g_dbus_proxy_call_sync (proxy,
                                   "gpioInit",
                                   g_variant_new ("(s)", gpio->name),
                                   G_DBUS_CALL_FLAGS_NONE,
                                   -1,
                                   NULL,
                                   &error);

	if (error != NULL) {
		return GPIO_LOOKUP_ERROR;
	}
```

spinler, 2017-01-23:
The problem is the service file didn't have the dependency to wait for the system manager service to start.  I added that, and then all is well.

https://gerrit.openbmc-project.xyz/#/c/1979/

SrideviRamesh, 2017-01-24:
@spinler ok i will be observe the HOST OS hang after https://gerrit.openbmc-project.xyz/#/c/1979/ merging right

closed 2017-01-24 by williamspatrick

SrideviRamesh, 2017-01-26:
Retried the checkstop sensor operation on barreleye HOST OS system with latest BMC  "v1.99.1-31"  version &  PNOR build (barreleye #272)
Os reboots immediately after injecting an checkstop on HOST OS