openbmc/openbmc #604: [Palmetto VERSION_ID="v1.99.0-131-ga81f31c-dirty" ] Presence detect for main planar and memory buffer fails

rahulmah, 2016-09-28:
In Palmetto, presence detect for memory buffer (membuf) and main planar (motherboard) is failing. Same issue is not seen with Barreleye. This seem to have come with recent inventory changes. 

Latest, I tested with Palmetto VERSION_ID="v1.99.0-131-ga81f31c-dirty”.

}rahulmah@rackspace:~$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X GET -d "{\"data\":[xx.xx.xx.xx]}" https://9.3.164.177/org/openbmc/inventory/system/cassis/motherboard/membuf
{
 "data": {
   "fault": "False",
   "fru_type": "MEMBUF",
   "is_fru": 1,
   "present": "False",      <---- **It is coming as not present**
   "version": ""
 },
 "message": "200 OK",
 "status": "ok"
}rahulmah@rackspace:~$

rahulmah@rackspace:~$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST https://xx.xx.xx.xx/login -d "{\"data\": [ \"root\", \"0penBmc\" ] }"
{
  "data": "User 'root' logged in",
  "message": "200 OK",
  "status": "ok"
}rahulmah@rackspace:~$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X GET https://9.3.164.177/org/openbmc/inventory/system/chassis/motherboard
{
  "data": {
    "fault": "False",
    "fru_type": "MOTHERBOARD",
    "is_fru": 1,
    "present": "False",     <---- **It is coming as not present**
    "version": ""
  },
  "message": "200 OK",
  "status": "ok"
}rahulmah@rackspace:~$

**Test case run**
minimal main planar inventory                                         .F
FFDC Collection Started         : 20160928071302043667
FFDC Collection Completed       : 20160928071319876409
minimal main planar inventory                                         | FAIL |
'0>0' should be true.

Centaur0 fault                                                        ...URI:https://xx.xx.xx.xx/org/openbmc/inventory/system/chassis/motherboard/membuf/attr/fault, method:Get, args:{}
Response code:200, Content:{
  "data": "False",
  "message": "200 OK",
  "status": "ok"
}
.F
FFDC Collection Started         : 20160928072906019008
FFDC Collection Completed       : 20160928072923880580
Centaur0 fault                                                        | FAIL |
False != True

**BMC FW:** 
root@palmetto:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-131"
VERSION_ID="v1.99.0-131-ga81f31c-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-131"
BUILD_ID="v1.99.0"
root@palmetto:~# cat /etc/version
20160927091543
root@palmetto:~#

[FFDC_membuf.txt](https://github.com/openbmc/openbmc/files/498027/FFDC_membuf.txt)
[FFDC_system_planar.txt](https://github.com/openbmc/openbmc/files/498029/FFDC_system_planar.txt)

rahulmah, 2016-09-29:
@spinler : Can you please look into this issue? And provide update on the same.

spinler, 2016-09-29:
I found the problem.  has to do with a FRU_NAME mismatch.

williamspatrick, 2016-09-30:
I believe this was resolved already.  Please reopen if you see this occur again on master.  (We had a number of duplicates of this same issue)

rahulmah, 2016-09-30:
@williamspatrick : I am still seeing this problem on palmetto with below build. Please reopen this issue.

root@palmetto:~# cat /etc/version
20160929211433
root@palmetto:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-144"
VERSION_ID="v1.99.0-144-g78d9255-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-144"
BUILD_ID="v1.99.0"
root@palmetto:~#

williamspatrick, 2016-09-30:
I'm tentatively reopening this but I suspect what is going on is the following:

1. #616 - indicates that settingsd is not starting.
2. ipmid tries to read the 'restricted_mode' setting from settingsd, which is unavailable so it assumes restricted mode.
3. Inventory records are ignored from the host when in restricted mode.

williamspatrick, 2016-09-30:
I don't want any work done on this until #616 is resolved.

williamspatrick, 2016-09-30:
@bradbishop - This gave me a thought with regard to inventory.  I'm not sure we've entirely thought through the 'restricted mode' setting, but if we are dropping inventory information in restricted mode this will affect your inventory redesign.  We may need to modify the IPMI code to at least update present state even if it ignores the rest of the inventory data.

spinler, 2016-09-30:
What's going on here is the MRW gives the motherboard a MOTHERBOARD FRU type.  However, in the IPMI code in sensorhandler.cpp it has this hardcoded:
```
sensorTypemap_t g_SensorTypeMap[] = {

    {0x01, 0x6F, "Temp"},
    {0x0C, 0x6F, "DIMM"},
    {0x0C, 0x6F, "MEMORY_BUFFER"}, <--- not MEMBUF
    {0x07, 0x6F, "PROC"},
    {0x07, 0x6F, "CORE"},
    {0x07, 0x6F, "CPU"},
    {0x0F, 0x6F, "BootProgress"},
    {0xe9, 0x09, "OccStatus"},  // E9 is an internal mapping to handle sensor type code os 0x09
    {0xC3, 0x6F, "BootCount"},
    {0x1F, 0x6F, "OperatingSystemStatus"},
    {0x12, 0x6F, "SYSTEM_EVENT"},
    {0xC7, 0x03, "SYSTEM"},
    {0xC7, 0x03, "MAIN_PLANAR"}, <--------- not MOTHERBOARD
    {0xC2, 0x6F, "PowerCap"},
    {0xFF, 0x00, ""},
}; ```

so, when hostboot sends down their sensor/FRU info, the ipmi code can't find the object it needs to set is_present on, and we get stuff like:

```
 palmetto ipmid[853]: fru data size:[57], area offset:[8], area_size:[64]
palmetto ipmid[853]: IPMI SET_SENSOR [0x2e]
palmetto ipmid[853]: Failed to find Sensor Type MEMBUF 
palmetto ipmid[853]: LOOKATME: Sensor should not be reported:  Type 0x00, Offset 0x04
palmetto ipmid[853]: LOOKATME: Sensor should not be reported:  Type 0x00, Offset 0x06
palmetto ipmid[853]: IPMI SET_SENSOR [0x1e]
```

I believe the IPMI code will eventually be refactored anyway, so for now I can just fix the MRW XML for Palmetto.

closed 2016-10-04 by williamspatrick

rahulmah, 2016-11-02:
@spinler : With latest fw(v1.99.0-212-gf1c10dd/20161102043616), I still see that presence detect for motherboard is coming as false in Palmetto. For membuf now the present detect is coming correct. 

```
rahulmah@rackspace:~$ curl -c cjar -b cjar -k https://xx.xx.xx.xx/org/openbmc/inventory/system/chassis/motherboard
{
  "data": {
    "Custom Field 1": "ECID:019A0048C0186718190C0000002604C1",
    "Custom Field 2": "",
    "Custom Field 3": "",
    "Custom Field 4": "",
    "Custom Field 5": "",
    "Custom Field 6": "",
    "Custom Field 7": "",
    "Custom Field 8": "",
    "FRU File ID": "",
    "Manufacturer": "",
    "Name": "",
    "Part Number": "",
    "Serial Number": "",
    "fault": "False",
    "fru_type": "MAIN_PLANAR",
    "is_fru": 1,
    "present": "False",            <---------- Presence detect is not present
    "version": ""
  },
  "message": "200 OK",
  "status": "ok"
}rahulmah@rackspace:~$

```