openbmc/openbmc #605: how to flash Tyan Palmetto server running AMI bmc with Openbmc?

kachini, 2016-09-28:
folks, 

we've got a Palmetto server running AMI BMC and would like to flash it with openbmc. 
How can we do this ?  
Do we need to update host FW along with BMC also? 

thanks in advance,

agangidi53, 2016-09-28:
Hey @kachini  What is the size of BMC flash chip ? I remember you telling me that you have a 16MB part only. In that case you wouldn't be able to run openBMC

sivassrr, 2016-09-28:
Run the command "pflash -b -i" - That can tell the size of BMC PNOR flash chip
Flash info:
-----------
Name          = Micron N25Q256Ax    <==============
Total size    = 32MB
Erase granule = 4KB

kachini, 2016-09-28:
hey @agangidi53, in this case the flash size is not an issue. This is a different server. We got it with AMI bmc and want to move from AMI to Openbmc. We are stuck with this as don't know if there any way to flash it without pulling flash out of the box.  AMI even don't have any cli interface to run pflash.

mdmillerii, 2016-09-28:
@kachini if you can get to the u-boot prompt with the AMI firmware, you can boot the OpenBMC stack from a network attached tftp server and from there flash the chip.  However, as @agangidi53 pointed palmetto systems shipped with a 16MB chip and do not have enough space for the OpenBMC build.  If you are just wanting to do some evaluation you can set some variables and run completely from RAM.  Otherwise you can insert a 32MB flash in the PNOR socket and write the OpenBMC code there, using the PNOR socket as a programmer.  (This would retain the AMI chip for backup).

I can help with the commands to netboot and not write to the flash if you are interested in testing this.

tk-wfischer, 2016-10-24:
@kachini Could it be that you are in fact using a 2U [TYAN TN71-BP012 (codename "Habanero")](http://www.tyan.com/solutions/tyan_openpower_system.html) system? This system comes with a AMI based BMC-firmware. The "Habanero" system is similar to "Palmetto", but in details they differ so for "Habanero" a different OpenPOWER firmware must be used. For OpenBMC, I'm not sure whether or not it will run on "Habanero".

@mdmillerii I have a [TYAN TN71-BP012 (BSP012T71V14HR-4T-3)](http://www.tyan.com/Barebones_TN71-BP012_BSP012T71V14HR-4T-3) and I'm interested in testing this. Currently I'm not sure whether the BMC flash of this machine is 16MB or 32MB. It has at least two BMC flash sockets: A "Golden BMC Flash Socket (BMC2)" and a "Primary BMC Flash Socket (BMC1)" And it has one "PNOR Socket (BIOS1)". Details see [TN71-BP012_UG_v1.0f_for_Channel.pdf](http://www.tyan.com/manuals/TN71-BP012_UG_v1.0f_for_Channel.pdf) chapter "4.4 Board Parts, Jumpers and Connectors" on pages 77 and 78. Using a new flash chip when writing the firmware would perhaps be a good idea, to retain the AMI chip(s) for backup. As far as I've seen in the OpenPOWER firmware documentation [document "Boot Flow for P8"](https://github.com/open-power/docs/blob/master/hostboot/P8_IPL_Flow_OpenPower.pdf), the BMC firmware is heavily involved in booting an OpenPOWER server. So retaining the AMI chip would be a good thing in case things go wrong.
So finally I could do the testing in two steps:
1) I'd like to do evaluation by running OpenBMC completely from RAM (in this way nothing should go wrong with the machine).
2) When the first tests are successful, I'd like to continue by writing the OpenBMC firmware to flash.

@mdmillerii could you help me by answering the following questions:
1) Do you think that OpenBMC should run on my Habanero system? Should I compile the image like described [here](https://openpowerblog.wordpress.com/2016/04/23/how-to-build-openbmc-binary-for-palmetto/) for Palmetto (and just hope that it runs on Habanero, too)?
2) Could you provide some details on how I could run OpenBMC completely from RAM without flashing it?
3) And finally could you provide details on how to write a flash chip using the PNOR socket as a programmer. (I'd assume that I would have to boot the machine, then unplug the current flash chip from the PNOR socket, and insert a new 32MB flash chip in there, and then write the firmware)

Thanks in advance for any help :-)

agangidi53, 2016-10-24:
@tk-wfischer 
In your 1st step of compiling the openBMC for Firestone you would pass the corresponding config for "firestone" in the commands specified my blog post:

So:
cd openbmc
TEMPLATECONF=meta-openbmc-machines/meta-openpower/meta-ibm/meta-firestone/conf . oe-init-build-env
bitbake obmc-phosphor-image

Everything else remains the same.

williamspatrick, 2016-10-24:
We do not support Habenaro. That is a different board from Palmetto and Firestone. We are limited as a team with how many boards we have bandwidth to support considering none of those are shipped as a real OpenBMC-based product.

tk-wfischer, 2016-10-25:
@williamspatrick thank you for your clarification that you do not support Habanero. Although I'm not a developer/programmer, I've skimmed through the OpenBMC code and I've found that e.g. for Palmetto it seems that the machine-specific code comes from https://github.com/open-power/palmetto-xml
What would be the necessary steps for someone to support Habanero? Would it be enough to create the corresponding and correct XML files like they are available for Palmetto [here](https://github.com/open-power/palmetto-xml)? Or is there much more to do and would someone need direct support from the manufacturer (TYAN in this case) to get real chances to bring OpenBMC up and running?

tk-wfischer, 2016-10-31:
@mdmillerii or @agangidi53 could you maybe give me some hints on my questions, in case that @williamspatrick is currently away:
> What would be the necessary steps for someone to support Habanero? Would it be enough to create the corresponding and correct XML files like they are available for Palmetto here? Or is there much [...]

I'd be helpful for any hints, as I'd really be pleased if there'd be a way to get OpenBMC running on Habanero.

agangidi53, 2016-10-31:
@tk-wfischer  Creating an XML IF it already doesn't exist would be a good start. If I am gonna have to summarize what to do beyond that, I need some more time.

tk-wfischer, 2016-10-31:
Thank you @agangidi53 for this first information. @mdmillerii if you have any infos already, it'd be great if you could share them with us.

williamspatrick, 2016-10-31:
@tk-wfischer - In order to create support for a new machine, there are currently two major files that need to be added. 1. Kernel device tree for the BMC. 2. System.py config file. Both of these are effectively codified subsets of the board schematics, which describe what pets are in the system and how they are connected to the BMC. I would need to check what information we are allowed to make public with regards to Habenaro.  Creating these files yourself would require access to information that is currently under NDAs.

williamspatrick, 2016-10-31:
"Pets" => "Parts" in previous comment.

williamspatrick, 2016-10-31:
The current XML files on open-power are only a subset of the information. Most of that is host-centric and not bmc-centric. There is extra info not contained in those XML files that we would need. 

For P9, we have also added bmc-centric data so that we can generate everything we need out of the system-XML. The Witherspoon XML is starting to get these. I doubt anyone is planning to invest the time on habenaro right now to also update it for this info.

tk-wfischer, 2016-11-04:
Thank you @williamspatrick for the valuable information. I'll forward this to my contact at TYAN - they should have all access to the information that is under NDA. Maybe my contact sees a way to generate the necessary data for Habanero. I'll keep you updated once I have an answer there.

tk-wfischer, 2016-11-21:
Hi again, I just wanted to let you know that the necessary information is also under NDA from TYAN's side. So for me I think that OpenBMC can't be supported today for Habanero. From my side, this issue can be closed. Thank you everybody for your help.

closed 2016-11-21 by williamspatrick