openbmc/openbmc #607: /org/openbmc/enumerate

hamzy, 2016-09-28:
[hamzy@hamzy-tp-w540 ~]$ curl -b cjar -k https://9.3.185.29/org/openbmc/; echo
{
  "data": [
    "/org/openbmc/HostServices", 
    "/org/openbmc/settings", 
    "/org/openbmc/UserManager", 
    "/org/openbmc/records", 
    "/org/openbmc/examples", 
    "/org/openbmc/NetworkManager"
  ], 
  "message": "200 OK", 
  "status": "ok"
}
[hamzy@hamzy-tp-w540 ~]$ curl -b cjar -k https://9.3.185.29/org/openbmc/list; echo
{
  "data": [
    "/org/openbmc/examples/path0/PythonObj", 
    "/org/openbmc/UserManager/Group", 
    "/org/openbmc/HostServices", 
    "/org/openbmc/UserManager/Users", 
    "/org/openbmc/records/events", 
    "/org/openbmc/settings/host0", 
    "/org/openbmc/examples/path1/SDBusObj", 
    "/org/openbmc/UserManager/User", 
    "/org/openbmc/examples/path0/SDBusObj", 
    "/org/openbmc/examples/path1/PythonObj", 
    "/org/openbmc/UserManager/Groups", 
    "/org/openbmc/NetworkManager/Interface"
  ], 
  "message": "200 OK", 
  "status": "ok"
}
[hamzy@hamzy-tp-w540 ~]$ curl -b cjar -k https://9.3.185.29/org/openbmc/enumerate; echo
{
  "data": {
    "description": "Internal Server Error", 
    "exception": "DBusException('Traceback (most recent call last):\\n  File \"/usr/lib/python2.7/site-packages/dbus/service.py\", line 751, in _message_cb\\n    _method_reply_return(connection, message, method_name, signature, *retval)\\n  File \"/usr/lib/python2.7/site-packages/dbus/service.py\", line 254, in _method_reply_return\\n    reply.append(signature=signature, *retval)\\nTypeError: \\'NoneType\\' object is not iterable\\n',)", 
    "traceback": [
      "Traceback (most recent call last):", 
      "  File \"/usr/lib/python2.7/site-packages/bottle.py\", line 862, in _handle", 
      "    return route.call(**args)", 
      "  File \"/usr/sbin/obmc-rest\", line 524, in __call__", 
      "    return self.callback(*a, **kw)", 
      "  File \"/usr/sbin/obmc-rest\", line 680, in wrap", 
      "    return self.to_jsonp(callback(*a, **kw))", 
      "  File \"/usr/lib/python2.7/site-packages/bottle.py\", line 1734, in wrapper", 
      "    rv = callback(*a, **ka)", 
      "  File \"/usr/sbin/obmc-rest\", line 615, in wrap", 
      "    resp = {'data': callback(*a, **kw)}", 
      "  File \"/usr/sbin/obmc-rest\", line 93, in __call__", 
      "    return getattr(self, 'do_' + request.method.lower())(**kw)", 
      "  File \"/usr/sbin/obmc-rest\", line 186, in do_get", 
      "    mapper_data=request.route_data['map']).dataitems()}", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/mapper.py\", line 155, in enumerate_subtree", 
      "    path, owner, interfaces, match))", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/mapper.py\", line 78, in __get_properties_on_bus", 
      "    properties_iface, i))", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/mapper.py\", line 64, in __get_properties_on_iface", 
      "    properties_iface.GetAll, iface)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/mapper.py\", line 50, in __try_properties_interface", 
      "    return f(*a)", 
      "  File \"/usr/lib/python2.7/site-packages/dbus/proxies.py\", line 145, in __call__", 
      "    **keywords)", 
      "  File \"/usr/lib/python2.7/site-packages/dbus/connection.py\", line 651, in call_blocking", 
      "    message, timeout)", 
      "DBusException: org.freedesktop.DBus.Python.TypeError: Traceback (most recent call last):", 
      "  File \"/usr/lib/python2.7/site-packages/dbus/service.py\", line 751, in _message_cb", 
      "    _method_reply_return(connection, message, method_name, signature, *retval)", 
      "  File \"/usr/lib/python2.7/site-packages/dbus/service.py\", line 254, in _method_reply_return", 
      "    reply.append(signature=signature, *retval)", 
      "TypeError: 'NoneType' object is not iterable", 
      ""
    ]
  }, 
  "message": "500 Internal Server Error", 
  "status": "error"
}

hamzy, 2016-09-28:
[hamzy@hamzy-tp-w540 ~]$ curl -b cjar -k https://9.3.185.29/org/openbmc/examples/enumerate; echo

also fails.

[hamzy@hamzy-tp-w540 ~]$ curl -b cjar -k https://9.3.185.29/org/openbmc/examples/; echo
{
  "data": [
    "/org/openbmc/examples/path0", 
    "/org/openbmc/examples/path1"
  ], 
  "message": "200 OK", 
  "status": "ok"
}
[hamzy@hamzy-tp-w540 ~]$ curl -b cjar -k https://9.3.185.29/org/openbmc/examples/path0; echo
{
  "data": {
    "description": "org.openbmc.ObjectMapper.Error.NotFound: path or object not found: /org/openbmc/examples/path0"
  }, 
  "message": "404 Not Found", 
  "status": "error"
}
[hamzy@hamzy-tp-w540 ~]$ curl -b cjar -k https://9.3.185.29/org/openbmc/examples/path1; echo
{
  "data": {
    "description": "org.openbmc.ObjectMapper.Error.NotFound: path or object not found: /org/openbmc/examples/path1"
  }, 
  "message": "404 Not Found", 
  "status": "error"
}

bradbishop, 2016-09-28:
These example applications have been yanked in HEAD so the problem should not manifest there.

williamspatrick, 2016-09-29:
@hamzy - Brad is suggesting that images built off 'master' should no longer exhibit this issue.  There are a few other issues (#609, #610) that will hopefully be resolved today with master, but I don't think those would impact the type of testing you are doing.  You can either build an image off of master or wait for fixes for those two, but I suggest upgrading the machine you are working on to a newer image.

closed 2016-09-29 by williamspatrick