openbmc/openbmc #608: journal filled with hwmon warnings

shenki, 2016-09-29:
Every 30 seconds a new hwmon warning is added to the journal by hwmon.py:

```
Sep 29 00:27:57 palmetto hwmon.py[741]: WARNING - hwmon: Unhandled hwmon: /sys/class/hwmon/hwmon0/
Sep 29 00:28:27 palmetto hwmon.py[741]: WARNING - hwmon: Unhandled hwmon: /sys/class/hwmon/hwmon0/
Sep 29 00:28:57 palmetto hwmon.py[741]: WARNING - hwmon: Unhandled hwmon: /sys/class/hwmon/hwmon0/
Sep 29 00:29:27 palmetto hwmon.py[741]: WARNING - hwmon: Unhandled hwmon: /sys/class/hwmon/hwmon0/
```

The system is a Palmetto. BMC version information:
```
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-131"
VERSION_ID="v1.99.0-131-ga81f31c-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-131"
BUILD_ID="v1.99.0"
```
```
Linux palmetto 4.7.3-be2001133835be80ba4657232b823ba2daaac0d6 #1 Wed Sep 14 12:05:02 UTC 2016 armv5tejl GNU/Linux
```

williamspatrick, 2016-09-30:
This indicates that someone added a new hwmon driver that we don't have support for from a configs/Palmetto perspective.  I'm not sure how much energy we will invest in this right now when that file is going to go away.

amboar, 2016-10-04:
@shenki what machine did you see this on?

shenki, 2016-10-04:
Palm5

williamli80, 2016-10-12:
```
 [https://github.com/openbmc/skeleton/blob/master/pyhwmon/hwmon.py#L49](url)  
The hwmon.py will poll hwmon device path periodically (/sys/class/hwmon,  there are 2 symlinks if devices under this directory ) .  We can get that the number of device is 0-0068, in palm309
 If the number of the device is not defined in HWMON_CONFIG of skeleton/config/Palmetto.py, a warning will raise.
(I have no idea about what does the number 0-0068 stand for. Who can help me?)
Currently we define '2-004c' and '3-0050' in attribute of HWMON_CONFIG.
[https://github.com/openbmc/skeleton/blob/master/configs/Palmetto.py#L110](url)

My test step:
root@palmetto:/# ls /sys/class/hwmon -l
total 0
lrwxrwxrwx    1 root     root             0 Oct 10 09:16 hwmon0 -> ../../devices/platform/ahb/ahb:apb/1e78a000.i2c/i2c-0/i2c-0/0-0068/hwmon/hwmon0
lrwxrwxrwx    1 root     root             0 Oct 10 09:16 hwmon1 -> ../../devices/platform/ahb/ahb:apb/1e78a000.i2c/i2c-2/i2c-2/2-004c/hwmon/hwmon1

root@palmetto:/# ls -l /sys/devices/platform/ahb/ahb:apb/1e78a000.i2c/i2c-0/i2c-0/0-0068/hwmon/hwmon0
total 0
lrwxrwxrwx    1 root     root             0 Oct 10 09:18 device -> ../../../0-0068
-r--r--r--    1 root     root          4096 Oct 12 05:33 name
lrwxrwxrwx    1 root     root             0 Oct 12 05:33 of_node -> ../../../../../../../../../../firmware/devicetree/base/ahb/apb/i2c@1e78a000/i2c-bus@40/rtc@68
lrwxrwxrwx    1 root     root             0 Oct 10 09:16 subsystem -> ../../../../../../../../../../class/hwmon
-r--r--r--    1 root     root          4096 Oct 12 05:33 temp1_input
-rw-r--r--    1 root     root          4096 Oct 10 09:16 uevent

Test on Palm309.
BMC Version:
root@palmetto:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-161"
VERSION_ID="v1.99.0-161-ge4bfd68-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-161"
BUILD_ID="v1.99.0"
```

williamspatrick, 2016-10-12:
0 is the bus 68 is the iic address of the hwmon device. It looks like, by one of the other names, that this is the RTC device?  Maybe it has a thermal sensor on it as well.

williamli80, 2016-10-13:
@williamspatrick @shenki  Yes.  Milton said that we added a RTC device as 0-0068. (Message 4, openbmc Digest, Vol 6, Issue 43, Feb 11)
```
Message: 4
Date: Wed, 10 Feb 2016 19:30:27 -0600
From: OpenBMC Patches <openbmc-patches@stwcx.xyz>
To: openbmc@lists.ozlabs.org
Subject: [PATCH linux 1/2] i2c: aspeed: Don't rename added devices
Message-ID:
<1455154228-7217-2-git-send-email-openbmc-patches@stwcx.xyz>

From: "Milton D. Miller II" <miltonm@us.ibm.com>

Changing the name of a device with dev_set_name after device_add
is not allowed.

However, of_create_device takes a bus_id parameter that specifies
the name of the platform device being added.

This does mean the bus property has to be retrieved multiple times.
Creation is skiped if there is trouble retrieving or formatting
the name.

The documentation for kobject_set_name (non-vargs) says:
 * This sets the name of the kobject.  If you have already added the
 * kobject to the system, you must call kobject_rename() in order to
 * change the name of the kobject.

As the kernel doc for device_rename says:
 * Make up a "real" name in the driver before you register anything, or add
 * some other attributes for userspace to find the device, or use udev to add
 * symlinks -- but never rename kernel devices later, it's a complete mess. We
 * don't even want to get into that and try to implement the missing pieces in
 * the core. We really have other pieces to fix in the driver core mess. :)

before:
1e78a000.i2c/1e78a1c0.i2c-bus/i2c-6/i2c-dev/i2c-6
1e78a000.i2c/1e78a180.i2c-bus/i2c-5/i2c-dev/i2c-5
1e78a000.i2c/1e78a140.i2c-bus/i2c-4/i2c-dev/i2c-4
1e78a000.i2c/1e78a100.i2c-bus/i2c-3/i2c-dev/i2c-3
1e78a000.i2c/1e78a100.i2c-bus/i2c-3/3-0050
1e78a000.i2c/1e78a0c0.i2c-bus/i2c-2/i2c-dev/i2c-2
1e78a000.i2c/1e78a0c0.i2c-bus/i2c-2/2-004c/hwmon/hwmon0
1e78a000.i2c/1e78a080.i2c-bus/i2c-1/i2c-dev/i2c-1
1e78a000.i2c/1e78a040.i2c-bus/i2c-0/i2c-dev/i2c-0
1e78a000.i2c/1e78a040.i2c-bus/i2c-0/0-0050
1e78a000.i2c/1e78a040.i2c-bus/i2c-0/0-0068/rtc/rtc0
1e78a000.i2c/1e78a340.i2c-bus/i2c-8/i2c-dev/i2c-8

after:
1e78a000.i2c/i2c-0/i2c-0/i2c-dev/i2c-0
1e78a000.i2c/i2c-0/i2c-0/0-0050
1e78a000.i2c/i2c-0/i2c-0/0-0068/rtc/rtc0
1e78a000.i2c/i2c-1/i2c-1/i2c-dev/i2c-1
1e78a000.i2c/i2c-2/i2c-2/i2c-dev/i2c-2
1e78a000.i2c/i2c-2/i2c-2/2-004c/hwmon/hwmon0
1e78a000.i2c/i2c-3/i2c-3/i2c-dev/i2c-3
1e78a000.i2c/i2c-3/i2c-3/3-0050
1e78a000.i2c/i2c-4/i2c-4/i2c-dev/i2c-4
1e78a000.i2c/i2c-5/i2c-5/i2c-dev/i2c-5
1e78a000.i2c/i2c-6/i2c-6/i2c-dev/i2c-6
1e78a000.i2c/i2c-7/i2c-7/i2c-dev/i2c-7

1e620000.fmc                   1e78e000.serial
1e620000.fmc:flash@0           1e78f000.serial
1e630000.spi                   ahb
1e630000.spi:flash             ahb:apb
1e660000.ethernet              alarmtimer
1e6c0080.interrupt-controller  i2c-0
1e6e2000.syscon                i2c-1
1e720000.sram                  i2c-2
1e780000.gpio                  i2c-3
1e782000.timer                 i2c-4
1e783000.serial                i2c-5
1e784000.serial                i2c-6
1e785000.wdt                   i2c-7
1e787000.vuart                 i2c-8
1e789140.ibt                   leds
1e78a000.i2c                   serial8250
1e78d000.serial

```

williamli80, 2016-10-13:
And 0-0068 was mentioned here:
```
Message: 5
Date: Wed, 18 May 2016 23:36:16 +0930
From: Joel Stanley <joel@jms.id.au>
To: Cédric Le Goater <clg@kaod.org>
Cc: OpenBMC Maillist <openbmc@lists.ozlabs.org>, Andrew Jeffery
<andrew@aj.id.au>
Subject: Re: [PATCH qemu 2/3] ast2400: add a rtc device on I2C bus 0
Message-ID:
<CACPK8XdLk84OOSN_Ke_Hc6Hwc=ng2NmB7ZjvNZRYipv26Ttbfg@mail.gmail.com>
Content-Type: text/plain; charset=UTF-8
```

mine260309, 2016-10-14:
I tried to add `0-0068` in HWMON_CONFIG as rtc's temperature, and the issue is gone:
```patch
diff --git a/configs/Palmetto.py b/configs/Palmetto.py
index 8abe4fe..19037a1 100644
--- a/configs/Palmetto.py
+++ b/configs/Palmetto.py
@@ -108,6 +108,11 @@ GPIO_CONFIG['MEZZ1_PRESENT'] =         { 'gpio_pin': 'O1', 'direction': 'in' }
 GPIO_CONFIG['CHECKSTOP']      =   { 'gpio_pin': 'P5', 'direction': 'falling' }
 
 HWMON_CONFIG = {
+	'0-0068' :  {
+		'names' : {
+			'temp1_input' : { 'object_path' : 'temperature/rtc','poll_interval' : 5000,'scale' : -3,'units' : 'C' },
+		}
+	},
 	'2-004c' :  {
 		'names' : {
 			'temp1_input' : { 'object_path' : 'temperature/ambient','poll_interval' : 5000,'scale' : -3,'units' : 'C' },
```

@williamli80 @williamspatrick Could you kindly help to check if it is the solution for this bug?

shenki, 2016-10-14:
This looks okay to me. @mine260309 I suggest you submit a patch in Gerrit

mine260309, 2016-10-14:
Submitted to gerrit: https://gerrit.openbmc-project.xyz/#/c/764/

adamliyi, 2016-10-14:
Curious about the name *temperature/rtc*, what sensor is this? A temperature sensor for RTC? We might need to choose a proper name for the object.

shenki, 2016-10-14:
It's a temperature sensor that is part of the same IC as the RTC. I guess it's actually the ambient temperature?

mine260309, 2016-10-17:
@shenki Do you know where (and how) to get the information about the sensor, so that we can give it a proper name?

It's noticed that Barreleye has the same issue (`Unhandled hwmon: /sys/class/hwmon/hwmon0/`), and Barreleye has the same RTC on the same i2c address as Palmetto, so this change applies to Barreleye as well.

I will update my review to change both `Barreleye.py` and `Palmetto.py`, but let's give it the object a proper name other than  `temperature/rtc`.

shenki, 2016-10-17:
I don't know anything about our naming schemes. I think the name in the patch is okay for now.

@bradbishop do you have feedback here?

mine260309, 2016-10-18:
The review is updated to modify both Palmetto.py and Barreleye.py

closed 2016-10-20 by williamspatrick