openbmc/openbmc #609: settings_manager.py failing repeatedly on barreleye with latest master (

geissonator, 2016-09-29:
Seeing this repeatedly in latest code on BarE01.

```
cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-142"
VERSION_ID="v1.99.0-142-gace81b5-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-142"
BUILD_ID="v1.99.0"
```

```
Sep 28 14:14:13 barreleye settings_manager.py[1105]: Traceback (most recent call last):
Sep 28 14:14:13 barreleye settings_manager.py[1105]:   File "/usr/sbin/settings_manager.py", line 179, in <module>
Sep 28 14:14:13 barreleye settings_manager.py[1105]:     obj = HostSettingsObject(bus, OBJ_NAME, s.SETTINGS, "/var/lib/obmc/")
Sep 28 14:14:13 barreleye settings_manager.py[1105]:   File "/usr/sbin/settings_manager.py", line 48, in __init__
Sep 28 14:14:13 barreleye settings_manager.py[1105]:     shk['default'])
Sep 28 14:14:13 barreleye settings_manager.py[1105]:   File "/usr/sbin/settings_manager.py", line 71, in set_settings_property
Sep 28 14:14:13 barreleye settings_manager.py[1105]:     self.Set(DBUS_NAME, name, str(value))
Sep 28 14:14:13 barreleye settings_manager.py[1105]:   File "/usr/lib/python2.7/site-packages/obmc/dbuslib/bindings.py", line 82, in Set
Sep 28 14:14:13 barreleye settings_manager.py[1105]:     self.validator(interface_name, property_name, new_value)
Sep 28 14:14:13 barreleye settings_manager.py[1105]:   File "/usr/sbin/settings_manager.py", line 164, in input_validator
Sep 28 14:14:13 barreleye settings_manager.py[1105]:     self.validate_list_ignore_case(shk['allowed'], value)
Sep 28 14:14:13 barreleye settings_manager.py[1105]:   File "/usr/sbin/settings_manager.py", line 106, in validate_list_ignore_case
Sep 28 14:14:13 barreleye settings_manager.py[1105]:     raise ValueError, "Invalid input. Data not in allowed values"
Sep 28 14:14:13 barreleye settings_manager.py[1105]: ValueError: Invalid input. Data not in allowed values
```

williamspatrick, 2016-09-29:
Is this triggered via testing of some sort or just a boot to standby?  It could be that we are accurately catching a test case failure now but inaccurately reporting it?

geissonator, 2016-09-29:
Just happens on the way to standby, then keeps happening.

closed 2016-09-29 by bradbishop