openbmc/openbmc #610: [Barreleye VERSION="v1.99.0-143" ] Code update failed to boot to BMC_READY

gkeishin, 2016-09-29:
root@barreleye:~# cat /etc/version
20160929020225
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-143"
VERSION_ID="v1.99.0-143-g1ffadd9-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-143"
BUILD_ID="v1.99.0"
root@barreleye:~# uptime
 05:04:42 up 21 min,  load average: 3.03, 1.59, 1.19
root@barreleye:~# obmcutil state
 = BMC_STARTING
root@barreleye:~#

Build#275   
https://openpower.xyz/job/openbmc-build/275/distro=ubuntu,target=barreleye/artifact/images/barreleye/

Sep 29 05:05:15 barreleye sync_inventory_items.py[1657]: Traceback (most recent call last):
Sep 29 05:05:15 barreleye sync_inventory_items.py[1657]:   File "/usr/sbin/sync_inventory_items.py", line 115, in <module>
Sep 29 05:05:15 barreleye sync_inventory_items.py[1657]:     inventory = os.path.join(
Sep 29 05:05:15 barreleye sync_inventory_items.py[1657]: NameError: name 'os' is not defined
Sep 29 05:05:15 barreleye systemd[1]: obmc-sync-inventory-item@system-chassis-motherboard.service: Main process exited, code=exited, status=1/FAILURE
Sep 29 05:05:15 barreleye systemd[1]: Failed to start Sync system/chassis/motherboard.
Sep 29 05:05:15 barreleye systemd[1]: obmc-sync-inventory-item@system-chassis-motherboard.service: Unit entered failed state.
Sep 29 05:05:15 barreleye systemd[1]: obmc-sync-inventory-item@system-chassis-motherboard.service: Failed with result 'exit-code'.
Sep 29 05:05:16 barreleye system_manager.py[796]: SystemManager Goto System State: BMC_READY
Sep 29 05:05:16 barreleye system_manager.py[796]: Running System State: BMC_READY
Sep 29 05:05:17 barreleye settings_manager.py[1659]: Traceback (most recent call last):
Sep 29 05:05:17 barreleye settings_manager.py[1659]:   File "/usr/sbin/settings_manager.py", line 15, in <module>
Sep 29 05:05:17 barreleye settings_manager.py[1659]:     import settings_file as s
Sep 29 05:05:17 barreleye settings_manager.py[1659]: ImportError: No module named settings_file
Sep 29 05:05:18 barreleye systemd[1]: phosphor-settings.service: Main process exited, code=exited, status=1/FAILURE
Sep 29 05:05:18 barreleye systemd[1]: phosphor-settings.service: Unit entered failed state.
Sep 29 05:05:18 barreleye systemd[1]: phosphor-settings.service: Failed with result 'exit-code'.
Sep 29 05:05:18 barreleye systemd[1]: phosphor-settings.service: Service hold-off time over, scheduling restart.

gkeishin, 2016-09-29:
I see the BMC_READY state came back like after 30+ minutes 
root@barreleye:~# obmcutil state
 = BMC_READY
root@barreleye:~# uptime
 05:30:13 up 46 min,  load average: 1.25, 1.21, 1.27
root@barreleye:~#

**1. It is flooded with the following message on the syslog**
Sep 29 05:31:58 barreleye systemd[1]: phosphor-settings.service: Service hold-off time over, scheduling restart.
Sep 29 05:31:58 barreleye systemd[1]: Stopped Phosphor Settings Manager.
Sep 29 05:31:58 barreleye systemd[1]: Started Phosphor Settings Manager.
Sep 29 05:31:59 barreleye hwmon.py[1582]: Cannot read attributes: /sys/class/hwmon/hwmon4/in2_input
Sep 29 05:32:00 barreleye hwmon.py[1582]: Cannot read attributes: /sys/class/hwmon/hwmon5/in2_input
Sep 29 05:32:04 barreleye hwmon.py[1582]: Cannot read attributes: /sys/class/hwmon/hwmon4/in2_input
Sep 29 05:32:05 barreleye hwmon.py[1582]: WARNING - hwmon: Unhandled hwmon: /sys/class/hwmon/hwmon0/
Sep 29 05:32:05 barreleye hwmon.py[1582]: Cannot read attributes: /sys/class/hwmon/hwmon5/in2_input
Sep 29 05:32:06 barreleye settings_manager.py[2574]: Traceback (most recent call last):
Sep 29 05:32:06 barreleye settings_manager.py[2574]:   File "/usr/sbin/settings_manager.py", line 15, in <module>
Sep 29 05:32:06 barreleye settings_manager.py[2574]:     import settings_file as s
Sep 29 05:32:06 barreleye settings_manager.py[2574]: ImportError: No module named settings_file
Sep 29 05:32:06 barreleye systemd[1]: phosphor-settings.service: Main process exited, code=exited, status=1/FAILURE
Sep 29 05:32:06 barreleye systemd[1]: phosphor-settings.service: Unit entered failed state.
Sep 29 05:32:06 barreleye systemd[1]: phosphor-settings.service: Failed with result 'exit-code'.
Sep 29 05:32:06 barreleye systemd[1]: phosphor-settings.service: Service hold-off time over, scheduling restart.
Sep 29 05:32:06 barreleye systemd[1]: Stopped Phosphor Settings Manager.
Sep 29 05:32:06 barreleye systemd[1]: Started Phosphor Settings Manager.

**2. Why the BMC READY is taking longer time to reach..**

gkeishin, 2016-09-29:
I did this experiment by using latest build on 2 systems and they exhibits the same behavior as stated above of flooding error messages and BMC boots to READY state taking longer time then normal behavior booting in few couple of minutes earlier.

williamspatrick, 2016-09-29:
This might be resolved by https://gerrit.openbmc-project.xyz/#/c/648/ which I just merged?  @anoo1 , @bradbishop would like you to look at the behavior and report if you think it is just due to a back-level version of settingsd being included.

bradbishop, 2016-09-29:
this seems like the same problem as #609

closed 2016-09-30 by williamspatrick