openbmc/openbmc #611: [Palmetto VERSION_ID="v1.99.0-131-ga81f31c-dirty" ] Systemevent url does not exist in Palmetto.

rahulmah, 2016-09-29:
Systemevent url has been complete remove from Palmetto. While this exist in Barelleye system. 
This url is used for event association with virtual sensor case. 

**Palmetto Run**
rahulmah@rackspace:~$ curl -c cjar -b cjar -k https://xx.xx.xx.xx/org/openbmc/inventory/system/systemevent/
{
  "data": {
    "description": "org.freedesktop.DBus.Error.FileNotFound: path or object not found: /org/openbmc/inventory/system/systemevent"
  },
  "message": "404 Not Found",
  "status": "error"
}rahulmah@rackspace:~$

rahulmah@rackspace:~$ curl -c cjar -b cjar -k https://xx.xx.xx.xx/org/openbmc/inventory/system/
{
  "data": [
    "/org/openbmc/inventory/system/apss",
    "/org/openbmc/inventory/system/chassis",
    "/org/openbmc/inventory/system/misc",
    "/org/openbmc/inventory/system/bios"
  ],
  "message": "200 OK",
  "status": "ok"

**Barreleye run**
rahulmah@rackspace:~$ curl -c cjar -b cjar -k https://9.3.23.32/org/openbmc/inventory/system/
{
  "data": [
    "/org/openbmc/inventory/system/bios",
    "/org/openbmc/inventory/system/systemevent",
    "/org/openbmc/inventory/system/chassis",
    "/org/openbmc/inventory/system/misc"
  ],
  "message": "200 OK",
  "status": "ok"

BMC FW info: 
root@palmetto:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-142"
VERSION_ID="v1.99.0-142-gace81b5-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-142"
BUILD_ID="v1.99.0"
root@palmetto:~# cat /etc/version
20160928194416
root@palmetto:~#

Due to this below two association related test cases are failing in Palmetto.

**Test case :** 
**Association with virtual sensor**
...
..URI:https://9.3.164.177/org/openbmc/records/events/62/fru/attr/endpoints, method:Get, args:{}
Response code:404, Content:{
  "data": {
    "description": "org.freedesktop.DBus.Error.FileNotFound: path or object not found: /org/openbmc/records/events/62/fru"
  },
  "message": "404 Not Found",
  "status": "error"
}
.F
FFDC Collection Started         : 20160929073225669226
FFDC Collection Completed       : 20160929073242902558
Association with virtual sensor :: ***GOOD PATH***                    | FAIL |
'{u'description': u'org.freedesktop.DBus.Error.FileNotFound: path or object not found: /org/openbmc/records/events/62/fru'}' does not contain '/org/openbmc/inventory/system/systemevent'

[FFDC_Association_issue.txt](https://github.com/openbmc/openbmc/files/500647/FFDC_Association_issue.txt)

williamspatrick, 2016-09-30:
@spinler - I don't recall what we discussed today to do with this one.

spinler, 2016-09-30:
We were going to figure out why there's a tie between an inventory object and and a sensor value, because we don't want this entry in the physical inventory.   I haven't dug into it yet.  @causten might know off the top of his head since he wrote it.

gkeishin, 2017-05-26:
Do we have a pending work on this one @spinler

spinler, 2017-06-07:
@gkeishin The sensors are being refactored right now, but I'm not up to speed on the details.  @dkodihal maybe know.

spinler, 2018-02-20:
The inventory has been redone since this issue was opened, and also no more palmetto fixes planned. Closing.

closed 2018-02-20 by spinler