openbmc/openbmc #612: [Barreleye VERSION="v1.99.0-143" ]  Code update BMC failed with SSLEOFError

gkeishin, 2016-09-29:
root@barreleye:~# cat /etc/version
20160929020225
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-143"
VERSION_ID="v1.99.0-143-g1ffadd9-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-143"
BUILD_ID="v1.99.0"
root@barreleye:~# obmcutil state
 = BMC_READY
root@barreleye:~# uptime
 21:36:02 up 14 min,  load average: 1.95, 1.69, 1.33
root@barreleye:~#

Sep 20 21:27:23 barreleye phosphor-gevent[763]: ::c0c3:a9b6:8e91:f402%1286572360 - - [2016-09-20 21:27:23] "POST /org/openbmc/control/flash/bmc/action/update HTTP/1.1" 200 169 9.863133
Sep 20 21:27:24 barreleye phosphor-gevent[763]: Traceback (most recent call last):
Sep 20 21:27:24 barreleye phosphor-gevent[763]:   File "/usr/lib/python2.7/site-packages/gevent/greenlet.py", line 327, in run
Sep 20 21:27:24 barreleye phosphor-gevent[763]:     result = self._run(*self.args, **self.kwargs)
Sep 20 21:27:24 barreleye phosphor-gevent[763]:   File "/usr/lib/python2.7/site-packages/gevent/server.py", line 102, in wrap_socket_and_handle
Sep 20 21:27:24 barreleye phosphor-gevent[763]:     ssl_socket = self.wrap_socket(client_socket, **self.ssl_args)
Sep 20 21:27:24 barreleye phosphor-gevent[763]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 402, in wrap_socket
Sep 20 21:27:24 barreleye phosphor-gevent[763]:     ciphers=ciphers)
Sep 20 21:27:24 barreleye phosphor-gevent[763]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 113, in __init__
Sep 20 21:27:24 barreleye phosphor-gevent[763]:     self.do_handshake()
Sep 20 21:27:24 barreleye phosphor-gevent[763]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 324, in do_handshake
Sep 20 21:27:24 barreleye phosphor-gevent[763]:     return self._sslobj.do_handshake()
Sep 20 21:27:24 barreleye phosphor-gevent[763]: SSLEOFError: EOF occurred in violation of protocol (_ssl.c:590)
Sep 20 21:27:24 barreleye phosphor-gevent[763]: <Greenlet at 0xb5b536c0L: <bound method WSGIServer.wrap_socket_and_handle of <WSGIServer at 0xb6411270L fileno=5 address=('::20a4:b04c:443c:abb6%3058677520', 443, 0, 3058677520L)>>(<socket at 0xb59adcf0L fileno=20>, ('::c0c3:a9b6:8e91:f402%1286572360', 33951, 0, 128)> failed with SSLEOFError

gkeishin, 2016-09-29:
Another failure on Palmetto 

root@palmetto:~# cat /etc/version
20160929091546
root@palmetto:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-143"
VERSION_ID="v1.99.0-143-g1ffadd9-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-143"
BUILD_ID="v1.99.0"
root@palmetto:~# obmcutil state
 = BMC_READY
root@palmetto:~# 

Sep 29 14:55:03 palmetto phosphor-gevent[757]: ::c0c3:a5b6:8e91:f402%1107331400 - - [2016-09-29 14:55:03] "PUT /org/openbmc/control/flash/bmc/attr/preserve_network_settings HTTP/1.1" 403 284 0.325311
Sep 29 14:55:03 palmetto dropbear[779]: Exit (root): Exited normally
Sep 29 14:55:04 palmetto settings_manager.py[800]: Traceback (most recent call last):
Sep 29 14:55:04 palmetto settings_manager.py[800]:   File "/usr/sbin/settings_manager.py", line 15, in <module>
Sep 29 14:55:04 palmetto settings_manager.py[800]:     import settings_file as s
Sep 29 14:55:04 palmetto settings_manager.py[800]: ImportError: No module named settings_file
Sep 29 14:55:05 palmetto flash_bios.exe[801]: Starting flasher: bios,,/org/openbmc/control/flash/bios,
Sep 29 14:55:05 palmetto flash_bios.exe[801]: flasher: , BMC = 0, address = 0x0
Sep 29 14:55:05 palmetto flash_bios.exe[801]: Setting up BIOS flash
Sep 29 14:55:05 palmetto flash_bios.exe[801]: Flash tuned
Sep 29 14:55:05 palmetto phosphor-mapper[715]: Warning: Introspection failure: service `:1.35` did not reply to method call on /
Sep 29 14:55:05 palmetto systemd[1]: phosphor-settings.service: Main process exited, code=exited, status=1/FAILURE

gkeishin, 2016-09-29:
@bradbishop Anything on the above failure on Barreleye and Palmetto..

williamspatrick, 2016-09-30:
@gkeishin - Please don't leave two totally different errors in the same issue.  The palmetto issue was resolved today.  The SSL looks like the client incorrectly terminated the connection.  @bradbishop what do you think?

gkeishin, 2016-10-04:
```
Seen on current build as well for a POST request, 
we have seen this in other POST request as well.. 
just highlighting one of the syslog message bellow. 
```
root@barreleye:~# cat /etc/version
20161004031316
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-146"
VERSION_ID="v1.99.0-146-g08ba306-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-146"
BUILD_ID="v1.99.0"
root@barreleye:~#

```
Sep 20 21:42:58 barreleye phosphor-gevent[969]: ::c053:afb6:8e91:f402%1195280712 - -
[2016-09-20 21:42:58] "POST /org/openbmc/records/events/action/acceptTestMessage HTTP/1.1" 200 169 30.003469
```
Sep 20 21:42:59 barreleye phosphor-gevent[969]: Traceback (most recent call last):
Sep 20 21:42:59 barreleye phosphor-gevent[969]:   File "/usr/lib/python2.7/site-packages/gevent/greenlet.py", line 327, in run
Sep 20 21:42:59 barreleye phosphor-gevent[969]:     result = self._run(*self.args, **self.kwargs)
Sep 20 21:42:59 barreleye phosphor-gevent[969]:   File "/usr/lib/python2.7/site-packages/gevent/server.py", line 102, in wrap_socket_and_handle
Sep 20 21:43:00 barreleye phosphor-gevent[969]:     ssl_socket = self.wrap_socket(client_socket, **self.ssl_args)
Sep 20 21:43:00 barreleye phosphor-gevent[969]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 402, in wrap_socket
Sep 20 21:43:00 barreleye phosphor-gevent[969]:     ciphers=ciphers)
Sep 20 21:43:00 barreleye phosphor-gevent[969]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 113, in __init__
Sep 20 21:43:00 barreleye phosphor-gevent[969]:     self.do_handshake()
Sep 20 21:43:00 barreleye phosphor-gevent[969]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 324, in do_handshake
Sep 20 21:43:00 barreleye phosphor-gevent[969]:     return self._sslobj.do_handshake()
```
Sep 20 21:43:00 barreleye phosphor-gevent[969]: SSLEOFError: EOF occurred in violation of protocol (_ssl.c:590)
Sep 20 21:43:00 barreleye phosphor-gevent[969]: <Greenlet at 0xb5b4df80L: <bound method WSGIServer.wrap_socket_and_handle of <WSGIServer at 0xb646a270L fileno=5 address=('::20a4:3f47:44cc:b0b6%3059042064', 443, 0, 3059042064L)>>(<socket at 0xb593aaf0L fileno=9>, ('::c053:afb6:8e91:f402%1195280712', 55220, 0, 119)> failed with SSLEOFError
```

geissonator, 2016-10-06:
I'm not sure if this is what's causing the actual code update failure.  Here's a trace of the actual update attempt (this comes after the above error) (yes we may need new issue for this one)

```
Sep 20 21:27:50 barreleye bmc_update.py[902]: Preserving network settings
Sep 20 21:27:50 barreleye bmc_update.py[902]: update: --no-flash --no-save-files --no-restore-files --no-clean-saved-files
Sep 20 21:27:52 barreleye phosphor-gevent[798]: ::c063:adb6:8e91:f402%1146325320 - - [2016-09-20 21:27:52] "POST /org/openbmc/control/flash/bmc/action/update HTTP/1.1" 200 169 6.631228
Sep 20 21:27:52 barreleye bmc_update.py[902]: Download complete. Updating...
```

Nothing after this, it just seems to keep running and then you see the test case come in and retry the same command a few seconds later.  I was thinking this was another warmReset issue but don't see that called in the skeleton/pyflashbmc/bmc_update.py code path.

We had run the prepForUpdate so no flash filesystems were mounted:

```
ls -al /run/initramfs/
total 32928
drwxr-xr-x   10 root     root           360 Sep 20 21:27 .
drwxr-xr-x   12 root     root           400 Sep 20 21:22 ..
drwxr-xr-x    2 root     root          1280 Oct  5  2016 bin
drwxr-xr-x    6 root     root           280 Oct  5  2016 etc
-rw-r--r--    1 1003     1003      33554432 Oct  5  2016 image-bmc
-rw-r--r--    1 root     root        131072 Sep 20 21:22 image-u-boot-env
-rwxr-xr-x    1 root     root          9183 Aug 29 16:59 init
-rw-r--r--    1 root     root            85 Mar  1  2016 init-options
-rw-r--r--    1 root     root             0 Mar  1  2016 init-options-base
drwxr-xr-x    4 root     root           900 Oct  5  2016 lib
drwxr-xr-x   17 root     root           234 Oct  5  2016 ro
drwxr-xr-x    4 root     root            80 Mar  1  2016 rw
drwxr-xr-x    2 root     root           780 Oct  5  2016 sbin
-rwxr-xr-x    1 root     root          1621 Aug 29 16:59 shutdown
-rwxr-xr-x    1 root     root          5179 Aug 29 16:59 update
drwxr-xr-x    6 root     root           120 Sep 20 21:41 usr
drwxr-xr-x    3 root     root            60 Oct  5  2016 var
-rw-r--r--    1 root     root           168 Aug 29 16:59 whitelist
[barreleye]# mount
dev on /dev type devtmpfs (rw,relatime,size=248248k,nr_inodes=62062,mode=755)
sys on /sys type sysfs (rw,relatime)
proc on /proc type proc (rw,relatime)
tmpfs on /run type tmpfs (rw,nosuid,nodev,mode=755)
/dev/loop0 on /run/initramfs/ro type squashfs (ro,relatime)
cow on / type overlay (rw,relatime,lowerdir=run/initramfs/ro,upperdir=run/initramfs/rw/cow,workdir=run/initramfs/rw/work)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
devpts on /dev/pts type devpts (rw,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /sys/fs/cgroup type tmpfs (ro,nosuid,nodev,noexec,mode=755)
cgroup on /sys/fs/cgroup/systemd type cgroup (rw,nosuid,nodev,noexec,relatime,xattr,release_agent=/lib/systemd/systemd-cgroups-agent,name=systemd)
debugfs on /sys/kernel/debug type debugfs (rw,relatime)
configfs on /sys/kernel/config type configfs (rw,relatime)
tmpfs on /tmp type tmpfs (rw)
tmpfs on /var/volatile type tmpfs (rw,relatime)
[barreleye]# 
```

gkeishin, 2016-10-06:
Thanks @geissonator . However this particular failure is seen in test as well for POST operation .. even though initially it was seen in code update.. So there is some generic failure which is making the failure.. Looking forward for the fixes if there is any ahead...

gkeishin, 2016-10-06:
**The test runs is badly hit even for CI, all the REST post operation has failed because of this issue .**

root@barreleye:~# cat /etc/version
20161006024821
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-161"
VERSION_ID="v1.99.0-161-ge4bfd68-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-161"
BUILD_ID="v1.99.0"
root@barreleye:~#

Sep 21 00:37:12 barreleye phosphor-gevent[906]: ::c023:a2b6:8e91:f402%1312655688 - - [2016-09-21 00:37:12] "POST /org/openbmc/records/events/action/acceptTestMessage HTTP/1.1" 200 169 42.841150
Sep 21 00:37:13 barreleye phosphor-gevent[906]: Traceback (most recent call last):
Sep 21 00:37:13 barreleye phosphor-gevent[906]:   File "/usr/lib/python2.7/site-packages/gevent/greenlet.py", line 327, in run
Sep 21 00:37:13 barreleye phosphor-gevent[906]:     result = self._run(*self.args, **self.kwargs)
Sep 21 00:37:13 barreleye phosphor-gevent[906]:   File "/usr/lib/python2.7/site-packages/gevent/server.py", line 102, in wrap_socket_and_handle
Sep 21 00:37:13 barreleye phosphor-gevent[906]:     ssl_socket = self.wrap_socket(client_socket, **self.ssl_args)
Sep 21 00:37:13 barreleye phosphor-gevent[906]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 402, in wrap_socket
Sep 21 00:37:13 barreleye phosphor-gevent[906]:     ciphers=ciphers)
Sep 21 00:37:13 barreleye phosphor-gevent[906]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 113, in __init__
Sep 21 00:37:13 barreleye phosphor-gevent[906]:     self.do_handshake()
Sep 21 00:37:13 barreleye phosphor-gevent[906]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 324, in do_handshake
Sep 21 00:37:13 barreleye phosphor-gevent[906]:     return self._sslobj.do_handshake()
``` Sep 21 00:37:13 barreleye phosphor-gevent[906]: SSLEOFError: EOF occurred in violation of protocol (_ssl.c:590) ```
Sep 21 00:37:14 barreleye phosphor-gevent[906]: <Greenlet at 0xb686eb20L: <bound method WSGIServer.wrap_socket_and_handle of <WSGIServer at 0xb63bb410L fileno=5 address=('::20a4:3e4e:449c:a3b6%3058329360', 443, 0, 3058329360L)>>(<socket at 0xb5a38b90L fileno=9>, ('::c023:a2b6:8e91:f402%1312655688', 53372, 0, 131)> failed with SSLEOFError
Sep 21 00:37:14 barreleye phosphor-gevent[906]: Traceback (most recent call last):

bradbishop, 2016-10-10:
`Sep 21 00:37:14 barreleye phosphor-gevent[906]: >(, ('::c023:a2b6:8e91:f402%1312655688', 53372, 0, 131)> failed with SSLEOFError`

is this an ipv6 address?

gkeishin, 2016-10-10:
Its seen on all systems with builds which seems to have both v4 and v6 as well.. 
so i picked up one system just to check.. 

root@barreleye:~# modprobe ipv6
modprobe: FATAL: Module ipv6 not found in directory /lib/modules/4.7.6-ae3a462415c7c05a138658ef8ffa2f321262c4ec
root@barreleye:~#

looks like the kernel is not built for v6.. but having said that... I don't remember running anything specific with v6.

jazurin, 2016-10-11:
I am also seeing this error while trying to clear the event logs (POST operation).  @bradbishop do you know when we can expect a fix? Thank you.

```
Sep 20 21:28:31 barreleye phosphor-gevent[881]: ::c013:a2b6:8e91:f402%1318422856 - - [2016-09-20 21:28:31] "POST /org/openbmc/records/events//action/clear HTTP/1.1" 200 166 10.657399
Sep 20 21:28:32 barreleye phosphor-gevent[881]: Traceback (most recent call last):
Sep 20 21:28:32 barreleye phosphor-gevent[881]:   File "/usr/lib/python2.7/site-packages/gevent/greenlet.py", line 327, in run
Sep 20 21:28:32 barreleye phosphor-gevent[881]:     result = self._run(*self.args, **self.kwargs)
Sep 20 21:28:32 barreleye phosphor-gevent[881]:   File "/usr/lib/python2.7/site-packages/gevent/server.py", line 102, in wrap_socket_and_handle
Sep 20 21:28:32 barreleye phosphor-gevent[881]:     ssl_socket = self.wrap_socket(client_socket, **self.ssl_args)
Sep 20 21:28:32 barreleye phosphor-gevent[881]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 402, in wrap_socket
Sep 20 21:28:32 barreleye phosphor-gevent[881]:     ciphers=ciphers)
Sep 20 21:28:32 barreleye phosphor-gevent[881]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 113, in __init__
Sep 20 21:28:32 barreleye phosphor-gevent[881]:     self.do_handshake()
Sep 20 21:28:32 barreleye phosphor-gevent[881]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 324, in do_handshake
Sep 20 21:28:32 barreleye phosphor-gevent[881]:     return self._sslobj.do_handshake()
Sep 20 21:28:32 barreleye phosphor-gevent[881]: SSLEOFError: EOF occurred in violation of protocol (_ssl.c:590)
Sep 20 21:28:32 barreleye phosphor-gevent[881]: <Greenlet at 0xb63d86c0L: <bound method WSGIServer.wrap_socket_and_handle of <WSGIServer at 0xb63ba410L fileno=5 address=('::20a4:964e:448c:a3b6%3058325264', 443,
0, 3058325264L)>>(<socket at 0xb6313cf0L fileno=10>, ('::c013:a2b6:8e91:f402%1318422856', 53863, 0, 131)> failed with SSLEOFError
Sep 20 21:28:34 barreleye hwmon.py[776]: Cannot read attributes: /sys/class/hwmon/hwmon4/in2_input
Sep 20 21:28:34 barreleye hwmon.py[776]: Cannot read attributes: /sys/class/hwmon/hwmon5/in2_input
Sep 20 21:28:37 barreleye hwmon.py[776]: WARNING - hwmon: Unhandled hwmon: /sys/class/hwmon/hwmon0/
Sep 20 21:28:39 barreleye hwmon.py[776]: Cannot read attributes: /sys/class/hwmon/hwmon4/in2_input
Sep 20 21:28:39 barreleye hwmon.py[776]: Cannot read attributes: /sys/class/hwmon/hwmon5/in2_input
Sep 20 21:28:43 barreleye phosphor-gevent[881]: ::c013:a2b6:8e91:f402%1318422856 - - [2016-09-20 21:28:43] "POST /org/openbmc/records/events//action/clear HTTP/1.1" 200 166 10.649344
Sep 20 21:28:44 barreleye phosphor-gevent[881]: Traceback (most recent call last):
Sep 20 21:28:44 barreleye phosphor-gevent[881]:   File "/usr/lib/python2.7/site-packages/gevent/greenlet.py", line 327, in run
Sep 20 21:28:44 barreleye phosphor-gevent[881]:     result = self._run(*self.args, **self.kwargs)
Sep 20 21:28:44 barreleye phosphor-gevent[881]:   File "/usr/lib/python2.7/site-packages/gevent/server.py", line 102, in wrap_socket_and_handle
Sep 20 21:28:44 barreleye phosphor-gevent[881]:     ssl_socket = self.wrap_socket(client_socket, **self.ssl_args)
Sep 20 21:28:44 barreleye phosphor-gevent[881]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 402, in wrap_socket
Sep 20 21:28:44 barreleye phosphor-gevent[881]:     ciphers=ciphers)
Sep 20 21:28:44 barreleye phosphor-gevent[881]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 113, in __init__
Sep 20 21:28:44 barreleye phosphor-gevent[881]:     self.do_handshake()
Sep 20 21:28:44 barreleye phosphor-gevent[881]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 324, in do_handshake
Sep 20 21:28:44 barreleye phosphor-gevent[881]:     return self._sslobj.do_handshake()
Sep 20 21:28:44 barreleye phosphor-gevent[881]: SSLEOFError: EOF occurred in violation of protocol (_ssl.c:590)
Sep 20 21:28:44 barreleye phosphor-gevent[881]: <Greenlet at 0xb63d86c0L: <bound method WSGIServer.wrap_socket_and_handle of <WSGIServer at 0xb63ba410L fileno=5 address=('::20a4:964e:448c:a3b6%3058325264', 443,
0, 3058325264L)>>(<socket at 0xb6313cf0L fileno=10>, ('::c013:a2b6:8e91:f402%1318422856', 53865, 0, 131)> failed with SSLEOFError
```

```
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-167"
VERSION_ID="v1.99.0-167-g7216f42-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-167"
BUILD_ID="v1.99.0"
```

williamspatrick, 2016-10-12:
@jazurin - we believe this is actually a problem on the pool machines (SSL client) at this point. The test team ran on another machine and didn't see this problem a single time.

geissonator, 2016-10-12:
We seem to hit this intermittently in our testing so not sure if running one place or another proves too much.  I noticed a network config change before my recent fail, could that be something?  Notice how it doesn't seem to recover until after the fail.

```
Oct 12 13:06:52 barreleye systemd-networkd[930]: eth0: Gained IPv6LL
Oct 12 13:06:52 barreleye systemd-timesyncd[719]: Network configuration changed, trying to establish connection.
Oct 12 13:06:54 barreleye hwmon.py[800]: Cannot read attributes: /sys/class/hwmon/hwmon4/in2_input
Oct 12 13:06:57 barreleye hwmon.py[800]: Cannot read attributes: /sys/class/hwmon/hwmon5/in2_input
Oct 12 13:06:59 barreleye phosphor-gevent[929]: Traceback (most recent call last):
Oct 12 13:06:59 barreleye phosphor-gevent[929]:   File "/usr/lib/python2.7/site-packages/gevent/greenlet.py", line 327, in run
Oct 12 13:06:59 barreleye phosphor-gevent[929]:     result = self._run(*self.args, **self.kwargs)
Oct 12 13:06:59 barreleye phosphor-gevent[929]:   File "/usr/lib/python2.7/site-packages/gevent/server.py", line 102, in wrap_socket_and_handle
Oct 12 13:06:59 barreleye phosphor-gevent[929]:     ssl_socket = self.wrap_socket(client_socket, **self.ssl_args)
Oct 12 13:06:59 barreleye phosphor-gevent[929]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 402, in wrap_socket
Oct 12 13:06:59 barreleye phosphor-gevent[929]:     ciphers=ciphers)
Oct 12 13:06:59 barreleye phosphor-gevent[929]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 113, in __init__
Oct 12 13:06:59 barreleye phosphor-gevent[929]:     self.do_handshake()
Oct 12 13:06:59 barreleye phosphor-gevent[929]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 324, in do_handshake
Oct 12 13:06:59 barreleye phosphor-gevent[929]:     return self._sslobj.do_handshake()
Oct 12 13:06:59 barreleye phosphor-gevent[929]: SSLEOFError: EOF occurred in violation of protocol (_ssl.c:590)
Oct 12 13:06:59 barreleye phosphor-gevent[929]: <Greenlet at 0xb63c46c0L: <bound method WSGIServer.wrap_socket_and_handle of <WSGIServer at 0xb63a6410L fileno=5 address=('::20a4:6543:44cc:a3b6%3058243344', 443, 0, 3058243344L)>>(<socket at 0xb63dd410L fileno=9>, ('::c053:a2b6:8e91:f402%1130662216', 45548, 0, 113)> failed with SSLEOFError
Oct 12 13:06:59 barreleye phosphor-gevent[929]: ::c053:a2b6:8e91:f402%1130662216 - - [2016-10-12 13:06:59] "POST /org/openbmc/managers/System//action/getSystemState HTTP/1.1" 401 230 0.391016
Oct 12 13:07:00 barreleye hwmon.py[800]: Cannot read attributes: /sys/class/hwmon/hwmon4/in2_input
Oct 12 13:07:00 barreleye hwmon.py[800]: WARNING - hwmon: Unhandled hwmon: /sys/class/hwmon/hwmon0/
Oct 12 13:07:02 barreleye hwmon.py[800]: Cannot read attributes: /sys/class/hwmon/hwmon5/in2_input
Oct 12 13:07:05 barreleye hwmon.py[800]: Cannot read attributes: /sys/class/hwmon/hwmon4/in2_input
Oct 12 13:07:05 barreleye systemd-networkd[930]: eth0: Starting DHCPv6 client after NDisc timeout failed: Invalid argument
Oct 12 13:07:05 barreleye systemd-networkd[930]: eth0: Configured
```

gkeishin, 2016-12-05:
@bradbishop Anything on this .. is there something we can do about this.. or is this related to PAM module restricting the sessions ?

gkeishin, 2016-12-09:
In new release , we see this error as well even for Login POST request

${resp} = RequestsLibrary.RequestsKeywords . Post Request openbmc, /login, data=${data}, headers=${headers}
Documentation:	
Send a POST request on the session object found using the
Start / End / Elapsed:	20161209 22:36:51.472 / 20161209 22:37:00.683 / 00:00:09.211
22:37:00.682	FAIL	SSLError: EOF occurred in violation of protocol (_ssl.c:590)

rahulmah, 2017-01-31:
@bradbishop : When we can expect fix for this issue?

bradbishop, 2017-02-28:
@rahulmah is this still happening?

sivassrr, 2017-04-26:
We still see REST timeout, right. @gkeishin @rahulmah

williamspatrick, 2017-04-26:
This is a duplicate of #1367.  We've already had significant discussion there on what the fix would be.

closed 2017-04-26 by williamspatrick