openbmc/openbmc #613: kernel panics with v1.0.5 on single server

KennethWilke, 2016-09-29:
Howdy folks,

I'm getting the following kernel panic on one of my DVT Barreleye servers:

```
occ-i2c 3-0050: occ i2c driver ready: i2c addr@0x50
occ-i2c 3-0051: occ i2c driver ready: i2c addr@0x51
i2c i2c-4: i2c_ast: timeout waiting for bus free
Unable to handle kernel paging request at virtual address 54000000
pgd = c0004000
[54000000] *pgd=00000000
Internal error: Oops: 80000005 [#1] ARM
Modules linked in:
CPU: 0 PID: 1 Comm: swapper Not tainted 4.4.16-openbmc-4.4-20160804-1 #1
Hardware name: ASpeed SoC
task: cf810aa0 ti: cf826000 task.ti: cf826000
PC is at 0x54000000
LR is at 0xc02bbf
pc : [<54000000>]    lr : [<00c02bbf>]    psr: 40000013
sp : cf827e18  ip : 00000014  fp : 00000000
r10: c05319c0  r9 : 00000000  r8 : c03a2170
r7 : cfa3f800  r6 : 00c02bad  r5 : 10000000  r4 : 00c02b99
r3 : 695f7265  r2 : 00000000  r1 : cf827e54  r0 : 0000006d
Flags: nZcv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment none
Control: 0005317f  Table: 40004000  DAC: 00000053
Process swapper (pid: 1, stack limit = 0xcf826190)
Stack: (0xcf827e18 to 0xcf828000)
7e00:                                                       00c02bbf 31000000
7e20: 5f353732 76697264 695f7265 0074696e 13c0031d 48a00000 48cfdf4c e0cfdf4c
7e40: 00c02b99 10000000 00c02bad 54000000 00c02bbf 31000000 5f353732 76697264
7e60: 695f7265 0074696e 13c0031d 48a00000 48cfdf4c e0cfdf4c 00c02b99 10000000
7e80: 00c02bad 54000000 00c02bbf cfa3f820 cfa3f854 c0523598 c0522b38 c052b5c8
7ea0: c04fa838 c01e90e8 00000000 c0523598 c01e9080 c01e73d8 cf83b06c cf8f4610
7ec0: c0523598 00000000 cdc897c0 c01e83ec c0481c2a c0481c2b c0523598 c04efc34
7ee0: 00000000 c0509e20 c0509e20 c01e98cc c0522b38 c052357c c04efc34 c028eb90
7f00: cdc8d580 c00096d8 cf80bbe0 cf805d00 cf805d00 c037704c 00000000 cfffc6e4
7f20: 00000000 c002b000 c0448483 cf80bbe0 cf80bbe0 c04b7520 0000006d 00000006
7f40: 00000006 c04b7da4 0000006c c04b7da4 00000000 00000006 c04fa82c 00000006
7f60: c04fa830 0000006d c05055c0 c05319c0 c04fa838 c04d9d68 00000006 00000006
7f80: 00000000 c04d95a0 00000000 c0371c9c 00000000 00000000 00000000 00000000
7fa0: 00000000 c0371ca4 00000000 c000a330 00000000 00000000 00000000 00000000
7fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
7fe0: 00000000 00000000 00000000 00000000 00000013 00000000 d7af2c2a 5dbd38e4
Code: bad PC value
---[ end trace f5e702c01a867766 ]---
Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b

---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b
```

When I reflash this server to our `6.15` build it will boot fine. I've tried using a clean flash directly to `1.0.5` and a clean flash to `6.15` followed by an update to `1.0.5`. In both cases the `1.0.5` version hits this error.

I'm not sure why only one server is hitting this, any advise on next troubleshooting steps?

mdmillerii, 2016-09-29:
The PC is completey in the wrong address range, not sure where it came from (I see it at x7e84 and x7e4c).
The string at the top of stack starting at 7e1f is "1275_driver_init\0" .  Looking further 0x7e18..7e4f matches 0x7e50..7e87 which may just be 2 register save areas taking the exception.

Doing some digging, that would be part of the name of the `adm1275_driver` `_init` from `module_i2c_driver`, so I would check if the driver is the same source for that driver.

mdmillerii, 2016-09-29:
I just looked at the log and noticed `i2c i2c-4: i2c_ast: timeout waiting for bus free` just before your fail, so maybe the lines are stuck low or the pullup resistors are damaged?  (combined with an error path)

shenki, 2016-09-30:
@KennethWilke Can you point me to the commit that your build is made from?

Failing that, the output of `/etc/os-release` and `uname -a`

KennethWilke, 2016-09-30:
@shenki I believe this build was based on https://github.com/ken1029/openbmc/tree/v10.50.256 

`/etc/os-release`
```
ID=openbmc-phosphor
NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)
VERSION=0.1.0 (jethro)
VERSION_ID=v10.50.256-dirty
PRETTY_NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 (jethro)
BUILD_ID=v10.50.256
```

`uname -a`
```
Linux barreleye 4.4.16-openbmc-4.4-20160804-1 #1 Wed Aug 24 15:44:52 CST 2016 armv5tejl GNU/Linux
```

@mdmillerii it could be something hardware related to the specific server, we have 20 other Barreleye DVT servers running this version that are not hitting this error. Since this server is able to run some of the older OpenBMC versions I'm wondering if you have any suggestions on how I can get this one up to the current version.

gtmills, 2018-05-31:
@KennethWilke No update in over a year and a half. Can this be closed? Thanks! This is an old Kernel.

closed 2018-06-01 by shenki