openbmc/openbmc #615: [Palmetto VERSION_ID="v1.99.0-144-g78d9255-dirty" ] UserAdd REST API giving DBusException: org.freedesktop.DBus.Error.NoReply error

rahulmah, 2016-09-30:
Again seeing that REST API to add user giving DBusException: org.freedesktop.DBus.Error.NoReply error with latest OpenBMC fw v1.99.0-144-g78d9255-dirty. This issue is seen on both Palmetto and Barreleye.

03:10:39 Create and delete user without groupname :: ***GOOD PATH***           URI:https://xx.xx.xx.xx/org/openbmc/UserManager/Users/action/UserAdd, method:Post, args:{u'data': {u'data': ['6yYgHZrH', 'fpB6qXXB', '', 'a0FGXOZO']}, u'headers': {u'Content-Type': u'application/json'}}
03:11:04 Response code:500, Content:{
03:11:04   "data": {
03:11:04     "description": "Internal Server Error", 
03:11:04     "exception": "DBusException('Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.',)", 
03:11:04     "traceback": [
03:11:04       "Traceback (most recent call last):", 
03:11:04       "  File \"/usr/lib/python2.7/site-packages/bottle.py\", line 862, in _handle", 
03:11:04       "    return route.call(**args)", 
03:11:04       "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 518, in __call__", 
03:11:04       "    return self.callback(*a, **kw)", 
03:11:04       "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 674, in wrap", 
03:11:04       "    return self.to_jsonp(callback(*a, **kw))", 
03:11:04       "  File \"/usr/lib/python2.7/site-packages/bottle.py\", line 1734, in wrapper", 
03:11:04       "    rv = callback(*a, **ka)", 
03:11:04       "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 609, in wrap", 
03:11:04       "    resp = {'data': callback(*a, **kw)}", 
03:11:04       "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 572, in wrap", 
03:11:04       "    return callback(*a, **kw)", 
03:11:04       "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 597, in wrap", 
03:11:04       "    return callback(*a, **kw)", 
03:11:04       "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 87, in __call__", 
03:11:04       "    return getattr(self, 'do_' + request.method.lower())(**kw)", 
03:11:04       "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 208, in do_post", 
03:11:04       "    return request.route_data['method'](*request.parameter_list)", 
03:11:04       "  File \"/usr/lib/python2.7/site-packages/dbus/proxies.py\", line 145, in __call__", 
03:11:04       "    **keywords)", 
03:11:04       "  File \"/usr/lib/python2.7/site-packages/dbus/connection.py\", line 651, in call_blocking", 
03:11:04       "    message, timeout)", 
03:11:04       "DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken."
03:11:04     ]
03:11:04   }, 
03:11:04   "message": "500 Internal Server Error", 
03:11:04   "status": "error"
03:11:04 }
03:11:04 
03:11:04 FFDC Collection Started 	: 20160930031104873853
03:11:18 FFDC Collection Completed 	: 20160930031118275850
03:11:18 | FAIL |
03:11:18 error != ok

Cross checked and found that user is getting created and even able to login to the system. So this issue is only with respect to error seen.

rahulmah@rackspace:~$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST https://xx.xx.xx.xx/login -d "{\"data\": [ \"root\", \"0penBmc\" ] "
{
  "data": "User 'root' logged in",
  "message": "200 OK",
  "status": "ok"
}rahulmah@rackspace:~$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST-d "{\"data\":[]}" https://xx.xx.xx.xx/org/openbmc/UserManager/Users/action/UserList
{
  "data": [
    "kkQXW8Xt",
    "q5ysmGA0",
    "3uA6AENV",
    "o80FKNl7",
    "9AbqBevx",
    "4WZVTxed",
    "XkhiICgf",
    "i7MiU88K",
    "fpB6qXXB"     <----- New user is getting created. 
  ],
  "message": "200 OK",
  "status": "ok"
}rahulmah@rackspace:~$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST https://xx.xx.xx.xx/login -d "{\"data\": [ \"fpB6qXXB\", \"a0FGXOZO" ] }"
{
  "data": "User 'root' logged in",         <----- New user is able to login also
  "message": "200 OK",
  "status": "ok"
}rahulmah@rackspace:~$

User is also present in BMC
root@palmetto:~# cat /etc/passwd | grep fpB6qXXB
fpB6qXXB:x:1015:1015:6yYgHZrH:/home/fpB6qXXB:/bin/sh
root@palmetto:~#

BMC FW info:
root@palmetto:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-144"
VERSION_ID="v1.99.0-144-g78d9255-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-144"
BUILD_ID="v1.99.0"
root@palmetto:~# cat /etc/version
20160929211433
root@palmetto:~#

FFDC from system:
[20160930031104873853_BMC_journalctl.txt](https://github.com/openbmc/openbmc/files/502520/20160930031104873853_BMC_journalctl.txt)

amboar, 2016-10-04:
@rahulmah *please* use the triple-backtick (```) [markdown](https://guides.github.com/features/mastering-markdown/) when posting log or console output, otherwise github interprets your log/console text as markdown when rendering the issue and mangles the output.

rahulmah, 2016-11-29:
Checked the journal logs and found that in Palmetto also we are hitting with SSLEOFError. This issue is same as https://github.com/openbmc/openbmc/issues/612. So closing it.

closed 2016-11-29 by rahulmah